arendst/Tasmota #5583: Help!!! Setup led Maxtrix max7219

phutruong1801, 2019-04-04:
Hello Everyone

Please help me configurated the Led Maxtrix 7219. i tried all way to setup the module on configurated page. but notthing work. when i try to use command DisplayModel, it's always show up "DisplayModel":4", but the display model need to be show is DisplayModel:3

i connect the CLK pin to D5 --- SSPI SCLK
CS pin to  D4 --- SSPI CS
DIN pin to D7 --- SSPI Mosi

ascillato2, 2019-04-04:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-04-04:
Max7219 is NOT supported as already asked https://github.com/arendst/Sonoff-Tasmota/issues/4709

phutruong1801, 2019-04-04:
> 
> 
> Max7219 is NOT supported as already asked #4709
> HT16K33 8x8 Matrix is supported

thanks you. i just think the matrix 8x8 is the Max7219. Thanks for that news. i noted that.

phutruong1801, 2019-04-04:
> Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you? [...]

Sr mod. that's my first time i create the question. noted that for next time.

Jason2866, 2019-04-04:
Please close the issue. THX

closed 2019-04-05 by ascillato2