arendst/Tasmota #5585: var5#state does not function as a rule trigger

radionerd, 2019-04-05:
BUG DESCRIPTION
var5#state does not function as a rule trigger, while var4#state does.
This occurs in 6.5.0(?-display and 6.5.0.4(ce4e9b0-display)
Example trace from test rule1:
10:42:14 CMD: var1 0
10:42:14 RSL: RESULT = {"Var1":"0"}
10:42:18 CMD: var2 0
10:42:18 RSL: RESULT = {"Var2":"0"}
10:42:23 CMD: var3 0
10:42:23 RSL: RESULT = {"Var3":"0"}
10:42:27 CMD: var4 0
10:42:27 RSL: RESULT = {"Var4":"0"}
10:42:30 CMD: var5 0
10:42:30 RSL: RESULT = {"Var5":"0"}
10:42:38 CMD: rule1 on
10:42:38 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
10:43:14 CMD:  Rule1 on var4#state do event display=4 endon on var5#state do event display=5 endon on event#display do var1 %value% endon
10:43:14 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":395,"Rules":"on var4#state do event display=4 endon on var5#state do event display=5 endon on event#display do var1 %value% endon"}
10:43:27 CMD: var4 4
10:43:27 RSL: RESULT = {"Var4":"4"}
10:43:27 RUL: VAR4#STATE performs "backlog event display=4"
10:43:27 RSL: RESULT = {"Event":"Done"}
10:43:27 RUL: EVENT#DISPLAY performs "var1 4"
10:43:27 RSL: RESULT = {"Var1":"4"}
10:43:36 CMD: var1
10:43:36 RSL: RESULT = {"Var1":"4"}
10:43:45 CMD: var5 5
10:43:45 RSL: RESULT = {"Var5":"5"}
10:43:51 CMD: var1
10:43:51 RSL: RESULT = {"Var1":"4"}

### REQUESTED INFORMATION
- [✔ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [✔] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [✔] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [✔] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [✔] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [✔] Device used (i.e. Sonoff Basic) : esp12 module 
- [✔] Tasmota binary firmware version number used : 6.5.0.4(ce4e9b0-display) pre-compiled 
- [ ] Development IDE - Compiler / Upload tools used : None / esptool.py
- [✔] Provide the output of command ``status 0`` :
STATUS 0 OUTPUT HERE:
11:04:28 CMD: status 0
11:04:28 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["esp12-Breadboard","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:04:28 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:28","StartupUTC":"2019-04-05T10:04:00","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":62,"SaveAddress":"3F9000"}}
11:04:28 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0.4(ce4e9b0-display)","BuildDateTime":"2019-04-04T18:09:40","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:04:28 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wifee'sWifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280500000100000000000000000000000000","00000000"]}}
11:04:28 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":3556,"Heap":14,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":2,"Features":["00000809","0FAAA19C","00078FA0","23B617C2","000028C0"]}}
11:04:28 RSL: STATUS5 = {"StatusNET":{"Hostname":"esp12bb","IPAddress":"192.168.1.96","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"18:FE:34:FE:31:47","Webserver":2,"WifiConfig":4}}
11:04:28 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 05 10:04:28 2019","Local":"Fri Apr 05 11:04:28 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"19:49","Sunset":"07:12"}}
11:04:28 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-04-05T11:04:28"}}
11:04:28 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-04-05T11:04:28","Uptime":"0T00:00:28","Vcc":3.464,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Wifee'sWifi","BSSId":"C8:0E:14:09:A8:8B","Channel":9,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [✔] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
See above
### TO REPRODUCE
_Steps to reproduce the behavior:
enter Rule1 as:
Rule1 on var4#state do event display=4 endon on var5#state do event display=5 endon on event#display do var1 %value% endon
Change the value of var5 and note that no trigger of event#display occurs
Change the value of var4 and note that the trigger event does occur
### EXPECTED BEHAVIOUR
Expect var5#state to behave as a trigger the same as vars1-4.
### SCREENSHOTS
None
### ADDITIONAL CONTEXT
Feel free to contact me if additional details or testing would be helpful.
Thanks in advance for looking into this issue.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-04-05:
Hi,

Duplicated issue from #5525

ascillato2, 2019-04-05:
Closing this issue as it is **duplicated** from #5525. Sorry. Anyway, thanks for reporting. Let's continue this in the original issue. Thanks. 

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------is 

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-05 by ascillato2