arendst/Tasmota #5586: Black web server UI instead of white

Markkyboy, 2019-04-05:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-04-06:
Implemented as options in my_user_config.h

Give it a try!

Markkyboy, 2019-04-06:
Hi Theo,

That's wonderful news, thank you!, when I get 5 minutes, I will most certainly check it out.

Regards,

Mark

  ----- Original Message ----- 
  From: Theo Arends 
  To: arendst/Sonoff-Tasmota 
  Cc: Markkyboy ; Author 
  Sent: Saturday, April 06, 2019 6:26 PM
  Subject: Re: [arendst/Sonoff-Tasmota] Black web server UI instead of white (#5586)

  Implemented as options in my_user_config.h

  Give it a try!

  —

---
This email has been checked for viruses by AVG.
https://www.avg.com

Markkyboy, 2019-04-06:
Hi, not sure if this is the right place to post this, but while compiling this version in Atom, I noticed red text fly by, scrolling back, I took a screenshot as I could not successfully swipe the right bit of text.

[image]

Markkyboy, 2019-04-07:
Request fulfilled.

closed 2019-04-07 by Markkyboy