arendst/Tasmota #5587: Zemismart boiler switch configuration

rhayun, 2019-04-05:
Hi all

i purchase the Zemismart Boiler switch and flash it with tuya-convert and tasmota
now i am trying to set the GPIO to make it work and its bit hard

did someone have template for that or can help?

[image]
[image]
[image]
[image]
[image]
[image]

ascillato, 2019-04-05:
From your photos, seems that your device has another mcu. It can't be seen clearly, but if it has another processors you will need the original firmware to see the communication between the esp8266 and that other microcontroller and make also the driver or rules for that.

If you have backed up the original firmware, you can have some support on that reverse engineering in the Tasmota support chat

rhayun, 2019-04-05:
i did it myself and its seems to be working like this:

[image]

but i am not sure if the tempeture sensor is working

blakadder, 2019-04-08:
Can you submit the device to the [templates database ](https://blakadder.github.io/templates/)

As far as temperature sensor, you can find it with the continuity setting on your multimeter since all the chip gpio's are nicely labelled and easily reached although I didn't spot it in the board images. Maybe its a simple thermistor?

ascillato2, 2019-04-10:
@rhayun

Hi,

Have you managed to find the right temperature sensor and to submit the complete template?

rhayun, 2019-04-12:
{"NAME":"Zemismart WH","GPIO":[53,255,255,17,21,255,255,255,255,255,52,255,255],"FLAG":1,"BASE":18}

blakadder, 2019-04-12:
That is correct. Issue closed.

closed 2019-04-12 by blakadder