arendst/Tasmota #5589: Use different html input field ids/names so password managers work well

some-rando, 2019-04-05:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
When I tried to use LastPass to begin setting unique wifi and web admin and mqtt usernames and passwords per device I learned that the field names aren't unique. `Web Admin Password` field is `p1`, and WiFi `AP1 Password` field is also named `p1`.

**Describe the solution you'd like**
I haven't checked for all name collisions yet, but if each html input field can have a unique name across all web pages it should eliminate the problem. I don't think it would require more than one letter addition or change to each field name. Example: `wp1` for web admin password.

**Describe alternatives you've considered**
As a bonus, it would be awesome if we could have all SSId/Username/passwords (really, ALL fields) accessible on a single page. Maybe using  `<div class="tab">`, so it wouldn't be so visually busy. It would load slower the first load, but switching between tabs would be faster than switching between pages.

**Additional context**
Example for tabs: [w3schools.com](https://www.w3schools.com/howto/howto_js_tabs.asp)

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-05:
Keeping the code as small as possible is the goal...
If this can be done not increasing code size a PR is welcome!

arendst, 2019-04-06:
As @Jason2866 already mentioned code space is a major issue. Processing larger webpages also takes code space not available for real functionality. Switching between pages is nice as long as they do not depend on each other. Current changes to the module page need a restart in order to configure the device for other parameters. There is a lot going on behind the scenes.

Even fieldnames is an issue if you know that they are stored in four bytes as long as you keep them as small as three characters. Making them four characters then they will occupy eight bytes!

I'll consider changing the naming to make it unique. Hell of a job but can be done and if it solves your password issue then it sounds like a valid request.

Does it only need unique password/username fieldnames or are there more dependencies?

some-rando, 2019-04-06:
Thanks for the quick replies and consideration. My bonus idea sounds like a no-go until code space isn't so super tight. No problem, it was "pie in the sky" anyway. Functionality is obviously more valuable than a prettier interface.

I looked at the code and I think I found what would need to be changed for minimal LastPass support. And yes, only unique password/username fieldnames seem to be required for LastPass to recognize things correctly. I can only assume it would be the same for other password managers.

In the `/sonoff/xdrv_01_webserver.ino` file it looks like the only name collision is actually the **p1**  in `D_WEB_ADMIN_PASSWORD "</b><br/><input id='p1' name='p1'`. I'm obviously not familiar enough with the whole code base to know if this would affect other things, but my limited understanding makes me think that `p1` could be changed to something like `wa` (**w**eb **a**dmin password) or `wap` if three characters doesn't steal an additional four bytes. Apologies if I'm underestimating the difficulty of the task!

It's truly impressive what you and everyone have made with Tasmota. Thanks for all the hard work.

some-rando, 2019-04-08:
Speedy fix! Thanks so much. I hope this is useful for people besides myself too.

closed 2019-04-08 by some-rando