arendst/Tasmota #5591: Module Type 52 for Teckin SP20 doesn't work

ptmoy2, 2019-04-06:
I'm running Tasmota v6.5.0 on a Teckin SP20 smart plug with energy monitoring.  When I set module type to "Teckin 52" as recommended in the release notes,  the toggle doesn't work, and no monitored values are displayed on the Tasmota device web page.  However, everything seems to work as expected when I set module type to "Techkin US - 59."

Is this a mistake in the Release Notes, or a problem with the "Teckin -52" settings?

blakadder, 2019-04-07:
Do you have this plug https://www.amazon.de/gp/product/B07D5V139R

ptmoy2, 2019-04-08:
No, I have this one:

https://www.amazon.com/gp/product/B079Q5W22B/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 

Bottom of the plug says Model: SP20

Jason2866, 2019-04-08:
Manufactures do change there products and sell it with same or different names...
If you found a working setting everything is fine. 
That means template isnt false, device "just" changed.
You can add a note to the SP20 template that maybe template Teckin-52 is the right choice for a variant of the SP 20

blakadder, 2019-04-08:
Ah ok. I understand the problem now. 

@ascillato @arendst RELEASE.md incorrectly states that module `Teckin (52)` applies to [Teckin SP20 ](https://www.teckinhome.com/copy-of-sp-22-2 )which is in fact module `Teckin US (59)`. 
Judging by the link in sonoff_template.h module `52` should be [Teckin SP22](https://www.teckinhome.com/copy-of-sp-23).
Furthermore module 45 also names Teckin SP22 
[image]
but the pinouts for `45` and `52` are different.
Reading #3950 does not clarify much.
Is module 52 meant for older board revisions of Teckin SP22? 
If it is obsoleted by a new version compatible with module 45 it might be prudent to remove the module or change the name to avoid user confusion.

Jason2866, 2019-04-08:
@blakadder  Yes there are different Teckin SP22 versions (different GPIO layout)

blakadder, 2019-04-08:
Ok, but still RELEASENOTES.md state its SP20 when it is in fact not, Teckin SP20 is Teckin US module

ascillato2, 2019-04-09:
Closing this issue as the PR #5606 has been made to fix this. Thanks for your contributions.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

PR #5606 (2019-04-09): Update RELEASENOTES.md
Corrected module 52 and 59 Teckin model numbers mixup

## Description:

**Related issue (if applicable):** fixes #5591
## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to update _changelog.ino_ file)
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works.
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#contribut [... cut]
closed 2019-04-09 by ascillato2