arendst/Tasmota #5592: Support for Shelly 2.5

JoergTiedemann, 2019-04-06:
Hi
Is the new Shelly 2.5 supported by Tasmota?
Which firmware version is required in minimum and are there any known open issues with Shelly 2.5 ?
Regards joerg

arendst, 2019-04-06:
Not yet. Mine is on the way.

peterforeman, 2019-04-06:
Do you have the GPIO pin mapping yet? Then I can fix it temporarily using USER pins.

arendst, 2019-04-06:
No, that's the tricky part ;-). If someone knows let him/her make the template as documented here https://github.com/arendst/Sonoff-Tasmota/wiki/Templates

peterforeman, 2019-04-06:
Currently I only have GPIO4 = Relay1. Rest I'm trying to figure out...

curzon01, 2019-04-10:
Correction:
D2/GPIO4 = Relay1
D8/GPIO15 = Relay2

**Be warned:**
After I'd tried Switch input for all of the rest of the GPIOs my Shelly2.5 SW1/SW2 pins has steady 230V~ signal also without input. I couldn't revert this status

arendst, 2019-04-10:
Warning: https://forum.creationx.de/forum/index.php?thread/1539-shelly-2-5-ist-angek%C3%BCndigt/&postID=22683#post22683

Seems to happen on a small amount of production items but it might be usefull to check before power on.

meingraham, 2019-04-10:
Was just coming here to add this information. You beat me too it. I added it to the Shelly wiki page.

arendst, 2019-04-11:
I just received info from the Shelly support group (Many thanks). The Shelly 2.5 GPIO's should be:

GPIO0 - Led1
GPIO2 - Button1
GPIO4 - Relay1
GPIO5 - Switch1
GPIO13 - Switch2
GPIO15 - Relay2

For correct functionality you may need Switchxn and/or Relayxi. I cannot test this yet as my device is still in the mail.

Energy metering is done by a new chip (ADE7953) connected to I2C on GPIO12/14 and IRQ on GPIO16 not supported by Tasmota yet. I will have to implement this.

Temperature is measured using the Analog input. Again, I have to implement this.

So for now the power control should work. The rest will need my attention...

arendst, 2019-04-11:
Pls report back if the above template functions as expected.

Tommi-nrw, 2019-04-11:
Please be careful handling Shelly 2.5!
Better first open every single Device, cut all pins and check for shortcuts **before** connecting to power!

See details here:
https://forum.creationx.de/forum/index.php?thread/1753-wichtige-information-zum-shelly-2-5/&postID=22725#post22725

Regards
Tom

curzon01, 2019-04-11:
This template works for my 2.5:
`{"NAME":"Shelly 2.5","GPIO":[52,255,17,255,21,83,0,0,6,82,5,22,0],"FLAG":1,"BASE":18}`

The difference to yours Theo:
- use SwitchXn instead of SwitchX (SwitchX using pull-up does not work here using 230V~ input)
- Switch1 and 2 are exchanged (it confirms the observe from @Tommi-nrw  link that Shelly group may have exchanged the SW1/SW2 internal documentation)

arendst, 2019-04-11:
Great! I'll post the updated version in the wiki then.

arendst, 2019-04-11:
Updated template with analog temperature readings (if compiled with ``//#define USE_ADC_VCC`` and using latest development release (2c6df2b) or up).

``{"NAME":"Shelly 2.5","GPIO":[52,255,17,255,21,83,0,0,6,82,5,22,0],"FLAG":2,"BASE":18}``

arendst, 2019-04-13:
First readings are in:

[image]

Still a lot to do like separating the outputs and calibration.

joba-1, 2019-04-13:
nice progress
-- 
iphone so i typo

ascillato2, 2019-04-15:
Closing this issue as the support for shelly 2.5 has been added.

Thanks :+1:

closed 2019-04-15 by ascillato2

curzon01, 2019-04-15:
It might be I make a principle failure but can't get power monitor to work on my Shellys 2.5.
I did also flash erase and re-flash to start from scratch but always got the VEML6070 UV sensor instead of power monitor:

[image]

Status 0:
```
stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:37","StartupUTC":"2019-04-15T14:40:47","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":13,"SaveAddress":"FA000"}}
stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.8(2163df5-sensors)","BuildDateTime":"2019-04-15T11:06:27","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dev.prsolution.eu","<*_*>"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":16,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDEE394","0007A3A4","B7FFBFCC","00DABBC0"]}}
stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6422","IPAddress":"172.18.10.151","Gateway":"172.18.10.4","Subnetmask":"255.255.255.0","DNSServer":"172.18.10.4","Mac":"DC:4F:22:5D:B9:16","Webserver":2,"WifiConfig":4}}
stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.local.prsolution.eu","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5DB916","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 15 14:41:25 2019","Local":"Mon Apr 15 15:41:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:59","Sunset":"19:41"}}
stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-04-15T15:41:25","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":51.0},"VEML6070":{"UvLevel":65535,"UvIndex":99.00,"UvIndexText":"OoR","UvPower":2.475},"TempUnit":"C"}}
stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-04-15T15:41:25","Uptime":"0T00:00:38","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"dev.prsolution.eu","BSSId":"5C:49:79:A2:34:FB","Channel":13,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
Template
```
stat/sonoff/RESULT = {"NAME":"Shelly 2.5","GPIO":[52,255,17,255,21,83,0,0,6,82,5,22,0],"FLAG":2,"BASE":18}
```

curzon01, 2019-04-15:
I tried to remove remove the none existing device, I couldn't. :grinning: 

Theo, it's an original Shelly 2.5 without any modification...

arendst, 2019-04-15:
Ah. I see. You did not connect a VEML but the code detects the VEML too even if it is not connected.

Short term solution, remove the VEML from the compile options.
Long term solution, either the VEML code owner finds a way to actively search for the device or I have to come up with some kind of address reservation system; once detected do not allow other to take over (this should work for the ADE7953 as it is a driver which is searched first before the sensor).

To be continued....

arendst, 2019-04-15:
Should be fixed in latest update just released.

curzon01, 2019-04-16:
I still didn't got it work also not with last commit and also not VEML disabled so I dig a little into the code.

The reason is simply the previous communicated template might be wrong. GPIO16 must be defined as `ADE7953 IRQ` and setting this energy monitor works.

But the comment in code confused me (ade7953 driver)
```
void Ade7953DrvInit(void)
...
if (i2c_flg && (pin[GPIO_ADE7953_IRQ] < 99)) {  // Irq on GPIO16 is not supported...
```

My template now is:

`{"NAME":"Shelly 2.5","GPIO":[52,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}`

(new is `...,156],"FLA...` instead of `...,0],"FLA...`) 

@arendst - can you pls confirm?

arendst, 2019-04-16:
@curzon I've updated th wiki some days ago with the correct template. Pls chk

curzon01, 2019-04-16:
Unfortunatley the given [template](https://github.com/arendst/Sonoff-Tasmota/wiki/Shelly-1-and-2#template) does not work - reason is the GIPO16 setting (=0):

`GPIO16` must be setup as `ADE7953 IRQ (156)` to get it started

arendst, 2019-04-16:
So sorry. You're right. Forgot to update (again).

Yes gpio16 should be 156.

Hope that works for you too.

BafHS, 2019-04-30:
Good morning,
thx a lot, its working fine, i use 6.5.0.9. 
i do every step from [https://github.com/arendst/Sonoff-Tasmota/wiki/Shelly-1-and-2#template](url) now one Calibration ... you write 

>execute the commands
but where?  Can you give me a tipp

Yarikx, 2019-04-30:
Hi. 
Does anyone experience random device reboots after switching some of the relays?
I can easily reproduce by quickly switching one relay from web.
I don't have non-tasmotized shelly 2.5 to check if it happens on original firmware =(

pboschi, 2019-04-30:
> Hi.
> Does anyone experience random device reboots after switching some of the relays?
> I can easily reproduce by quickly switching one relay from web.
> I don't have non-tasmotized shelly 2.5 to check if it happens on original firmware =(

Same here... after some minutes i get:
21:25:03 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}

arzaman, 2019-05-04:
hi
Does last binary official release 6.5.0 works with shelly 2.5 or need to compile a more recent development version? 
thks
Davide

curzon01, 2019-05-05:
6.5.0 might be usable for IO-GPIOS (relay, switches) but without energy monitor.

For fully Shelly 2.5 support use the latest developer version (minimum v6.5.0.8 branch 5e810f1f)

You don't need to compile by yourself, download the [latest compiled developer binary](http://thehackbox.org/tasmota/) as described [here](https://github.com/arendst/Sonoff-Tasmota#development)

arzaman, 2019-05-05:
cristal clear thanks, I will try to flash the developer binary on my shelly 2.5  and report any issue in case !

lobocobra, 2019-06-23:
When will the shelly be added to master? 
6.5.0.16 wifi connection is unstable and I want to use my 20+ shelly 2.5 in a productive environnment.

ArturGabrys, 2019-06-23:
> When will the shelly be added to master?
> 6.5.0.16 wifi connection is unstable and I want to use my 20+ shelly 2.5 in a productive environnment.
> Same issues with 6.5.0.16. Reset 5 brought no improvement.

I would really appreciate this, one of main reasons for buying Shelly devices for me was ability to run Tasmota on it. Any approximate time? Thank you.

meingraham, 2019-06-24:
@lobocobra Wi-Fi instability is usually associated with the core you are using. Even when the latest stable development branch becomes the next master, this will not address your Wi-Fi connectivity issues.

If you haven't tried [2.5.2](http://thehackbox.org/tasmota/020502), give that a try. Otherwise, [2.3.0](http://thehackbox.org/tasmota/020300) tends to be foolproof.

MaximilanPower, 2019-07-12:
Hello everyone, 
is there a reason, why there it is not selectable under Module Type or did I not found the commit? 
Best regard Max

meingraham, 2019-07-12:
@MaximilanPower,

Going forward, Tasmota modules will be solely for unique capability requiring special coding. The 2.5 makes use of existing components configurations to specify its [template](https://blakadder.github.io/templates/shelly_25.html). The [Template](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates) feature was added in order to support the explosion in the number of ESP devices on the market.

zkolb, 2019-07-17:
Hello everyone,
I've just installed two shelly 2.5 devices with release 6.6.0. After 10-15 minutes they reach 73C and turn off both relays (the threshold set by SetOption42 is ignored). I've replaced the lightbulbs with LED ones, lowering the power consumption to 14W, that extended the ON time to15-20 minutes. At this point I've turned off temperature sensing (flag 2 -> 0), but I'm not happy with this solution. Can we calibrate the temperature reading? Is there a way to change the 73C threshold? Does anyone else experience this issue?
Best regards, Zoltan

arendst, 2019-07-18:
See #6036 and use the precompiled dev version.

zkolb, 2019-07-22:
Thanks!
The dev version works fine. The temperature does not even go above 72C. I was runnng it for several hours and it never turned off the relays due to overtemp.

stefanbosch, 2019-08-25:
Great fix, dev version did the trick for the temp threshold! 👍

djack2017, 2020-02-11:
Can anyone reveal how to initialize the GPIO13 port to support an external switch in shelly 2.5?
With GPIO5 there is no problem, and GPIO13 always returns "1" regardless of how I initialized it.

JeremiahGillis, 2020-02-23:
Can anyone confirm Shelly 2.5 still fully works with Tasmota 8.1.0? My energy measurements are not working or showing up.

I just flashed it and this is what my interface looks like:
[image]

I followed this guide https://tasmota.github.io/docs/#/devices/Shelly-2.5 and used template:

```
{"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
```

My serial logging is off
[image]

```
01:03:43 CMD: Group 0, Index 1, Command "STATUS", Data "0"
01:03:43 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:03:43 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:17:28","StartupUTC":"2020-02-22T23:46:15","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":44,"SaveAddress":"F8000"}}
01:03:43 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"329/699"}}
01:03:43 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Butchery IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0020880A","0105C8000100060000005A00000000000000","00000000","00000000"]}}
01:03:43 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
01:03:43 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6366","IPAddress":"192.168.3.112","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"80:7D:3A:74:58:DE","Webserver":2,"WifiConfig":4}}
01:03:43 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7458DE","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
01:03:43 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 23 00:03:43 2020","Local":"Sun Feb 23 01:03:43 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:44","Sunset":"18:22"}}
01:03:43 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-02-23T01:03:43","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":43.1},"TempUnit":"C"}}
01:03:43 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-02-23T01:03:43","Uptime":"0T00:17:28","UptimeSec":1048,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Butchery IoT","BSSId":"7A:8A:20:0F:93:DB","Channel":5,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```

```
01:05:33 CMD: Group 0, Index 1, Command "STATUS", Data "8"
01:05:33 RSL: stat/tasmota/STATUS8 = {"StatusSNS":{"Time":"2020-02-23T01:05:33","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.5},"TempUnit":"C"}}
01:05:35 CMD: status 9
01:05:35 SRC: WebConsole from 192.168.1.105
01:05:35 CMD: Group 0, Index 1, Command "STATUS", Data "9"
01:05:35 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
```

What could be wrong?

I have more Shelly 2.5's that I could try. I just did a Shelly 1PM which showed energy measurements correctly. I just don't know if this is a hardware or software issue. This specific Shelly 2.5 was never used with the stock firmware.

ascillato, 2020-02-23:
My Shelly 2.5 is fully working with latest Tasmota. 

If you are self compiling remember to define I2C. Please use the precompiled bins. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Antexa, 2020-10-29:
> Can anyone confirm Shelly 2.5 still fully works with Tasmota 8.1.0? My energy measurements are not working or showing up.
> 
> I just flashed it and this is what my interface looks like:
> [image]
> 
> I followed this guide https://tasmota.github.io/docs/#/devices/Shelly-2.5 and used template:
> 
> ```
> {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
> ```
> 
> My serial logging is off
> [image]
> 
> ```
> 01:03:43 CMD: Group 0, Index 1, Command "STATUS", Data "0"
> 01:03:43 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 01:03:43 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:17:28","StartupUTC":"2020-02-22T23:46:15","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":44,"SaveAddress":"F8000"}}
> 01:03:43 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"329/699"}}
> 01:03:43 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Butchery IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0020880A","0105C8000100060000005A00000000000000","00000000","00000000"]}}
> 01:03:43 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
> 01:03:43 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6366","IPAddress":"192.168.3.112","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"80:7D:3A:74:58:DE","Webserver":2,"WifiConfig":4}}
> 01:03:43 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7458DE","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
> 01:03:43 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 23 00:03:43 2020","Local":"Sun Feb 23 01:03:43 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:44","Sunset":"18:22"}}
> 01:03:43 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-02-23T01:03:43","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":43.1},"TempUnit":"C"}}
> 01:03:43 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-02-23T01:03:43","Uptime":"0T00:17:28","UptimeSec":1048,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Butchery IoT","BSSId":"7A:8A:20:0F:93:DB","Channel":5,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:07"}}}
> ```
> 
> ```
> 01:05:33 CMD: Group 0, Index 1, Command "STATUS", Data "8"
> 01:05:33 RSL: stat/tasmota/STATUS8 = {"StatusSNS":{"Time":"2020-02-23T01:05:33","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.5},"TempUnit":"C"}}
> 01:05:35 CMD: status 9
> 01:05:35 SRC: WebConsole from 192.168.1.105
> 01:05:35 CMD: Group 0, Index 1, Command "STATUS", Data "9"
> 01:05:35 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> ```
> 
> What could be wrong?
> 
> I would expect the energy parameters to show up even if the hardware isn't working correctly.
> 
> I have more Shelly 2.5's that I could try. I just did a Shelly 1PM which showed energy measurements correctly. I just don't know if this is a hardware or software issue. This specific Shelly 2.5 was never used with the stock firmware.

@JeremiahGillis Did you manage to get it works ?

JeremiahGillis, 2020-11-09:
> Can anyone confirm Shelly 2.5 still fully works with Tasmota 8.1.0? My energy measurements are not working or showing up.

Sorry for the delay. Yes, I was able to get mine work on a different Shelly 2.5. I flashed Tasmota 8.1.0 on another Shelly 2.5 and it functioned correctly.

My problem was a hardware issue. I flashed the device in question back to stock firmware. The energy measurements did not work. So, the energy measurements only seem to show up if the energy IC is detected over I2C. Not intuitive when you have faulty hardware.

Antexa, 2020-11-09:
Hey @JeremiahGillis, I also revert back to stock firmware 👎 
I think this issue happen some time, I saw it on Discord channel, this might be helpfull for another people.

kugelkopf123, 2020-11-09:
@Antexa I can't see any problems when using a Shelly2.5 and Tasmota. I have installed 8 of them in my house and they all run wonderfully. For months.
Since version 9 the bug with the high temperature caused by the ADE IC has also been solved. just install the latest version. Take the template from the official site: https://templates.blakadder.com/shelly_25.html
If you like, you can also find links to further information on calibration. 
If they are used as shutter control, just search for "Blinds and Shutters" in the Tasmota Docs.

michalskrivanek, 2020-11-19:
FWIW I'm seeing similar issues. I flashed 3 Shellies via OTA with https://github.com/yaourdt/mgos-to-tasmota, all of them to Tasmota 8.5 , first worked like a charm, the other two do not have power monitoring.  They all did work with stock firmware.
Tried downgrading to Tasmota 8.2 and re-set template (as a desperate try per #9692, I do have both loads connected) - didn't help. Reset power couple times, sometimes i2cscan returned errors (error 4 on 0x39), sometimes no device detected. It did come up eventually on the third shelly (while I was at Tasmota 8.2, with 60s+ power off)...but I still don't see any meaningful pattern - the first shelly always works, the second one doesn't, the third one is currently working. Soft restarts with restart 1 or reset 5 doesn't seem to make any difference.
I have all of them in wall and difficult to power cycle so I didn't do that many tests, but it's definitely fishy...

StarvingDeveloper, 2020-11-21:
I have similar issue - flashed over air with mg2x and upgraded to v9. So had hard time convert to new template format. Then spent some time till made monitoring showup, but for me only 1 line and I cant make both line show power monitoring whatever I do.
So we all should unite and find out it's faulty shelly devices or tasmota firmware got many undocumented settings which make our devices act weird.

kugelkopf123, 2020-11-21:
Dont get it... 
why not simply flash via cable. No problem. Works like a charm.

StarvingDeveloper, 2020-11-22:
> Dont get it...
> why not simply flash via cable. No problem. Works like a charm.

Not everyone want spend life diging deep in this shit, buying programers, dig trought wiki for days and such. Some bought shelly to use it and forget it got instaled. Then see such thing as tasmota exist and posible flash over air, so why dont try? Especialy this first time when I see diference flashing over air and via cable. Problem somewhere else, and even if I would buy programer nd flash firmware via cable problems would stay since people have 3 shelly devices and all 3 act diferent.

Omega236, 2021-02-03:
@michalskrivanek  

i have the same problem that i have no Powermeter with a shelly2.5 i flashed over OTA.

is there any solution?

michalskrivanek, 2021-02-03:
I converted two more and met same issue, but both magically started to work after template changes and 60s+ power off eventually (not on a first try). I would guess it may have something to do with residual charge somewhere or temperature...but I have no idea really and can't reliably reproduce anything.
For reference on 9.2.0 template I use {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}

Omega236, 2021-02-03:
@michalskrivanek 
You are right, after a while and without electricity, the energy display worked.

But unfortunately the ENERGY_Total is not split.

arendst, 2021-02-04:
If you see the same error again where the readings are gone pls try to push the reset button on the device and let me know if this solves the issue.

If so I can make a more neat reset solution.

foobar26, 2021-07-17:
Had the same issue and reverting back to version 7.2.0 and resetting the template fixed the issue for me.
I then went back to 8.5.2 (where I started with the OTA) and energy monitoring was still there. Maybe this helps others to fix their issues as well.

gitmaster2013, 2022-01-02:
Had the same problem ShellyFW(everythingWorking)-OTA->tasmota(10.1.0)-tasmota-Upgrade->tasmota-DE(10.1.0)

The first OTA was:
http://dl.dasker.eu/firmware/mg2tasmota-Shelly25
Working config:
{"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
https://templates.blakadder.com/shelly_25.html

**I just had to turn of the CircuitBreaker (for about  30min). After that the page refreshed and the values appeared.**

AleXSR700, 2022-09-09:
Same here. I had to remove all 2.5 from the grid and then the power monitoring worked.

AleXSR700, 2023-01-19:
> I can confirm this.
> After flashing with the OTA tool, the Shelly must be without power for >30 minutes.
> Only then are the energy values displayed.

For me it was just a few seconds.