arendst/Tasmota #5593: HC-SR04

alvsky, 2019-04-06:
Hi all
I have a situation where I connected HC-SR04 sensor to wemos D1 mini. 
I connected it to pins D1 (Trig) and D2 (Echo), and VCC to 5V (all directly).
Result is that I don't have any output. 
I expected at least wrong readings, but I get literally nothing as you can see on the screenshot.

I tried 2 different sensors and behavior is the same.

Is it possible that both of sensors are faulty or I am doing something wrong?

[image]

[image]

[image]

joba-1, 2019-04-06:
Ideally Wemos operates on 3.3V and HC-SR04 on 5V.
I had bad results with this (on test firmware, not tasmota) - even if I got readings from the HC-SR04, they were very inaccurate.

Your Tasmota log even says VCC is only at 2.784V.
My guess is that the HC-SR04 does not interpret this low voltage as HIGH level, so it does not see a trigger.

Try working with level shifters or voltage dividers. It is also safer for the Wemos: The 5V echos could destroy its inputs.

effelle, 2019-04-06:
If you're using a SR04 you need a voltage divider in your echo (D2). 
Is not necessary only if you use an SR04P (3.3V variant). For more information have a look to the Wiki.
[https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-HC-SR04-Ultrasonic-Sensor](url)

ascillato2, 2019-04-09:
Template Missing.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md)).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-09 by ascillato2

alvsky, 2019-04-09:
Hi all
thank you for you answers.
I solved the problem using voltage divider.

As for the Support chat, since I am not very active here I just did not know that so opening the issue was the logical solution to me.

Thank you all for the answers