arendst/Tasmota #5594: Energy monitoring template & PowerOnState

BostonMikeGetHub, 2019-04-07:
I installed tasmota 6.5.0 on an AWP04L energy monitoring plug. I then installed the template for this plug from https://blakadder.github.io/templates/awp04l.html

All worked including the energy monitoring.  However when power is removed then restored, the relay remains on but the website and the power light indicate off.   Pressing the button corrects the state indicated by the light and website. 

PowerOnState = 3

Can this be fixed by tasmota or the template?

blakadder, 2019-04-07:
That could be a template issue which you will have to debug and correct yourself

BostonMikeGetHub, 2019-04-07:
The template works perfectly and does not have any issues as long as power is not lost to the device  

The problem is when PowerOnState is set to the defaul of 3, the device is on and it lost power then power returns.  It has no problem when PowerOnState is 0.  Apparently tasmota is not setting the state of the light to the same as the relay during the power on. 

I will investigate the template but I hope someone with knowage of tasmota will look into what I said above.   

@blakadder you appear to be the owner of the templates on GitHub.  I tried to put the below note on the template description but I don't have permission.  Please put something like the following note on the template page.  Thanks. 

"Their is an issue when using this template with tasmota 6.5.0. See issue 5594 for additional information."

ascillato2, 2019-04-12:
Hi, 

I could not reproduce your issue. 

Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you? Also, please, add the console output at Tasmota starts. Remember to put the weblog to 4 before that. 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

BostonMikeGetHub, 2019-04-13:
### BUG DESCRIPTION
I installed tasmota 6.5.0 on an AWP04L energy monitoring plug. I then installed the template for this plug from https://blakadder.github.io/templates/awp04l.html

{"NAME":"AWP04L","GPIO":[57,255,255,131,255,134,0,0,21,17,132,56,255],"FLAG":0,"BASE":18}

All worked including the energy monitoring. However when power is removed then restored from/to the device, the relay remains on (supplying power to the load) but the website and the power light indicate off. Pressing the button corrects the state indicated by the light and website.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)

- [x] Device used (i.e. Sonoff Basic) : An AWP04L power monitoring plug from the Amazon website given on the https://www.amazon.com/Smart-Outlet-Compatible-Alexa-Google/dp/B07LGSBFNJ

- [x] Tasmota binary firmware version number used : 6.5.0 pre-complided "6.5.0(release-basic)"
- [x ] Development IDE - Compiler / Upload tools used : 
OTA update from tasmota, sonoff-minimal.bin

- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
Status 0
23:12:01 SRC: WebConsole from 192.168.1.203
23:12:01 RSL: Received Topic /Status, Data Size 1, Data 0
23:12:01 RSL: Group 0, Index 1, Command STATUS, Data 0
23:12:01 MQT: stat/PMPlug-6/STATUS = {"Status":{"Module":0,"FriendlyName":["PMPlug-6"],"Topic":"PMPlug-6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:12:01 MQT: stat/PMPlug-6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:01:55","StartupUTC":"2019-04-13T03:10:06","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":84,"SaveAddress":"F8000"}}
23:12:01 MQT: stat/PMPlug-6/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-basic)","BuildDateTime":"2019-03-19T12:25:14","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:12:01 MQT: stat/PMPlug-6/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UAPAC01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","280500000100000000000000000000000000","00000008"]}}
23:12:01 MQT: stat/PMPlug-6/STATUS4 = {"StatusMEM":{"ProgramSize":432,"Free":568,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","0F082380","000483A0","21021706","000001C0"]}}
23:12:01 MQT: stat/PMPlug-6/STATUS5 = {"StatusNET":{"Hostname":"PMPlug-6-8146","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:6C:1F:D2","Webserver":2,"WifiConfig":4}}
23:12:01 MQT: stat/PMPlug-6/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.232","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6C1FD2","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:12:01 MQT: stat/PMPlug-6/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 13 03:12:01 2019","Local":"Fri Apr 12 23:12:01 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"06:08","Sunset":"19:21"}}
23:12:01 MQT: stat/PMPlug-6/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:12:01 MQT: stat/PMPlug-6/STATUS10 = {"StatusSNS":{"Time":"2019-04-12T23:12:01","ENERGY":{"TotalStartTime":"2019-04-03T20:07:17","Total":1.284,"Yesterday":1.187,"Today":0.016,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
23:12:01 MQT: stat/PMPlug-6/STATUS11 = {"StatusSTS":{"Time":"2019-04-12T23:12:01","Uptime":"0T00:01:55","Vcc":3.138,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"UAPAC01","BSSId":"80:2A:A8:51:E3:F4","Channel":11,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:03"}}}
23:12:10 WIF: Checking connection...
23:12:10 WIF: Connected

```
- [x] Provide the output of console when you experience your issue if apply :

_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

00:00:00 CFG: Loaded from flash at FA, Count 82
00:00:00 SRC: Restart
00:00:00 Project sonoff PMPlug-6 Version 6.5.0(release-basic)-2_3_0
00:00:00 CFG: Saved to flash at F9, Count 83, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 UAPAC01 in mode 11N as PMPlug-6-8146...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on PMPlug-6-8146 with IP address 192.168.1.66
[... cut ...]
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Screen shot with OFF is with the issue (device power removed then restored, the load had power but the device does not indicate this. 
Screen shot with ON is normal condition.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
the output of command status 0  is after I had the issue.  I removed then restored power to the device. 
[image]
[image]

ascillato2, 2019-04-13:
Hi,

Thanks for completing the template and sharing the requested information. Very appreciated. 

After the device boots, if you refresh the web page, it still shows OFF while the relay is ON?

BostonMikeGetHub, 2019-04-13:
Thank you for looking into this. 

Yes, it still shows OFF while the relay is ON after I refresh the web page.  

Pressing the device button or TOGGLE on the web page puts the device's light and web page indicating ON state while the relay remain that. 

Basically the device always indicate OFF when it powers up regardless of its state when the device lost power.

ascillato2, 2019-04-14:
Let's try a test.

Please, could you go to extra configuration menu and disable mqtt?

Then please try again to toggle to ON, remove power and power it again.

BostonMikeGetHub, 2019-04-14:
Samething. The relay is on but the device light and website indicates off. 

Below is console after I reapply power.
I did a status 0 at the end. 

> 
00:00:00 CFG: Loaded from flash at FB, Count 145
00:00:00 SRC: Restart
00:00:00 Project sonoff PMPlug-6 Version 6.5.0(release-basic)-2_3_0
00:00:00 CFG: Saved to flash at FA, Count 146, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 UAPAC01 in mode 11N as PMPlug-6-8146...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on PMPlug-6-8146 with IP address 192.168.1.66
00:00:03 RSL: INFO1 = {"Module":"AWP04L","Version":"6.5.0(release-basic)","FallbackTopic":"cmnd/DVES_6C1FD2_fb/","GroupTopic":"sonoffs"}
00:00:03 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"PMPlug-6-8146","IPAddress":"192.168.1.66"}
00:00:03 RSL: INFO3 = {"RestartReason":"Power on"}
00:00:03 RSL: RESULT = {"POWER":"OFF"}
00:00:03 RSL: POWER = OFF
00:00:03 APP: (UTC) Sun Apr 14 18:43:21 2019, (DST) Sun Mar 10 02:00:00 2019, (STD) Sun Nov 03 02:00:00 2019
14:43:22 HTP: Main Menu
14:43:28 APP: Boot Count 34
14:43:28 CFG: Saved to flash at F9, Count 147, Bytes 3584
14:43:30 RSL: STATE = {"Time":"2019-04-14T14:43:30","Uptime":"0T00:00:12","Vcc":3.137,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"UAPAC01","BSSId":"80:2A:A8:51:E3:F4","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:43:30 RSL: SENSOR = {"Time":"2019-04-14T14:43:30","ENERGY":{"TotalStartTime":"2019-04-03T20:07:17","Total":1.361,"Yesterday":0.000,"Today":0.010,"Period":2,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
14:43:41 WIF: Checking connection...
14:43:41 WIF: Connected
14:43:43 HTP: Console
14:43:53 CMD: Status 0
14:43:53 SRC: WebConsole from 192.168.1.203
14:43:53 RSL: Received Topic /Status, Data Size 1, Data 0
14:43:53 RSL: Group 0, Index 1, Command STATUS, Data 0
14:43:53 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["PMPlug-6"],"Topic":"PMPlug-6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
14:43:53 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:00:35","StartupUTC":"2019-04-14T18:43:18","Sleep":50,"CfgHolder":4617,"BootCount":34,"SaveCount":147,"SaveAddress":"F9000"}}
14:43:53 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-basic)","BuildDateTime":"2019-03-19T12:25:14","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:43:53 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UAPAC01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008021","280500000100000000000000000000000000","00000008"]}}
14:43:53 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":432,"Free":568,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","0F082380","000483A0","21021706","000001C0"]}}
14:43:53 RSL: STATUS5 = {"StatusNET":{"Hostname":"PMPlug-6-8146","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:6C:1F:D2","Webserver":2,"WifiConfig":4}}
14:43:53 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 14 18:43:53 2019","Local":"Sun Apr 14 14:43:53 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"06:05","Sunset":"19:23"}}
14:43:53 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:43:53 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-04-14T14:43:53","ENERGY":{"TotalStartTime":"2019-04-03T20:07:17","Total":1.362,"Yesterday":0.000,"Today":0.011,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
14:43:53 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-04-14T14:43:53","Uptime":"0T00:00:35","Vcc":3.136,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"UAPAC01","BSSId":"80:2A:A8:51:E3:F4","Channel":11,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:03"}}}
14:44:01 WIF: Checking connection...
14:44:01 WIF: Connected

BostonMikeGetHub, 2019-04-17:
I am able to reproduce bug #5663

1. PowerOnState = 0. 
2. Power on the device by pressing the button.
3. Restart the device by pressing Restart on its web page. 

After restart, the relay, and light are on (should be off).  Power monitoring is also active. 

The above is regardless of mqtt being enabled or not. 

However, with PowerOnState = 0, the device now operates correctly when I remove power then restore power to the device.

BostonMikeGetHub, 2019-04-17:
I believe issues #5594 and #5663 are related.  Issue #5594 (this issue) occurs when PowerOnState = 3.  Issue #5663 occurs when PowerOnState = 0.

BostonMikeGetHub, 2019-04-18:
I will be happy to test both error conditions when a possible solution is found.

arendst, 2019-04-18:
Could you try changing your template from
```
{"NAME":"AWP04L","GPIO":[57,255,255,131,255,134,0,0,21,17,132,56,255],"FLAG":0,"BASE":18}
```
to 
```
{"NAME":"AWP04L","GPIO":[57,255,255,131,255,134,0,0,29,17,132,56,255],"FLAG":0,"BASE":18}
```
This will change GPIO12 from "Relay1" to "Relay1i" inverting the relay. You might also have to play with GPIO0 (Led2i) and/or GPIO15 (Led1i).

BostonMikeGetHub, 2019-04-18:
Thank you for your help. 

Unfortunately the relay is now opposite to the stated indicate by the device web page.

arendst, 2019-04-18:
OK. In that case set it back to Relay1 (21).

arendst, 2019-04-18:
Could you pls verify what the statetext are:
```
15:43:16 CMD: statetext1
15:43:16 SRC: Serial
15:43:16 RSL: Group 0, Index 1, Command STATETEXT, Data 
15:43:16 MQT: stat/wemos2/RESULT = {"StateText1":"OFF"}
15:43:24 CMD: statetext2
15:43:24 SRC: Serial
15:43:24 RSL: Group 0, Index 2, Command STATETEXT, Data 
15:43:24 MQT: stat/wemos2/RESULT = {"StateText2":"ON"}
```

arendst, 2019-04-18:
If equal pls change code in sonoff.ino line 2694 from:
```
  // Issue #526 and #909
  for (uint8_t i = 0; i < devices_present; i++) {
    if ((i < MAX_RELAYS) && (pin[GPIO_REL1 +i] < 99)) {
      bitWrite(power, i, digitalRead(pin[GPIO_REL1 +i]) ^ bitRead(rel_inverted, i));
    }
    if ((i < MAX_PULSETIMERS) && (bitRead(power, i) || (POWER_ALL_OFF_PULSETIME_ON == Settings.poweronstate))) {
      SetPulseTimer(i, Settings.pulse_timer[i]);
    }
  }
```
in
```
  // Issue #526 and #909
  for (uint8_t i = 0; i < devices_present; i++) {
/*
    if ((i < MAX_RELAYS) && (pin[GPIO_REL1 +i] < 99)) {
      bitWrite(power, i, digitalRead(pin[GPIO_REL1 +i]) ^ bitRead(rel_inverted, i));
    }
*/
    if ((i < MAX_PULSETIMERS) && (bitRead(power, i) || (POWER_ALL_OFF_PULSETIME_ON == Settings.poweronstate))) {
      SetPulseTimer(i, Settings.pulse_timer[i]);
    }
  }
```

arendst, 2019-04-18:
Using command ``SetOption63 1`` should solve your issue.

BostonMikeGetHub, 2019-04-18:
```
10:28:38 CMD: statetext1
10:28:38 MQT: stat/PMPlug-6/RESULT = {"StateText1":"OFF"}
10:28:44 CMD: statetext2
10:28:44 MQT: stat/PMPlug-6/RESULT = {"StateText2":"ON"}
```

I am only able to download .bin files.  I am unable to change the code.  Sorry

meingraham, 2019-04-18:
I also no longer compile my own code.

Is SetOption63 (6.5.0.9) in the most recent thehackbox binaries?

BostonMikeGetHub, 2019-04-18:
```
00:00:00 Project sonoff PMPlug-6 Version 6.5.0(release-basic)-2_3_0
00:00:00 WIF: Connecting to AP1 UAPAC01 in mode 11N as PMPlug-6-8146...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on PMPlug-6-8146 with IP address 192.168.1.66
00:00:03 RSL: INFO1 = {"Module":"AWP04L","Version":"6.5.0(release-basic)","FallbackTopic":"cmnd/DVES_6C1FD2_fb/","GroupTopic":"sonoffs"}
00:00:03 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"PMPlug-6-8146","IPAddress":"192.168.1.66"}
00:00:03 RSL: INFO3 = {"RestartReason":"Power on"}
00:00:03 RSL: RESULT = {"POWER":"OFF"}
00:00:03 RSL: POWER = OFF
10:39:14 RSL: STATE = {"Time":"2019-04-18T10:39:14","Uptime":"0T00:00:12","Vcc":3.133,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"UAPAC01","BSSId":"80:2A:A8:51:E3:F4","Channel":11,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:39:14 RSL: SENSOR = {"Time":"2019-04-18T10:39:14","ENERGY":{"TotalStartTime":"2019-04-03T20:07:17","Total":4.172,"Yesterday":0.380,"Today":1.763,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
10:39:24 CMD: poweronstate
10:39:24 RSL: RESULT = {"PowerOnState":3}
10:39:45 CMD: SetOption63
10:39:45 RSL: RESULT = {"SetOption63":"ON"}
```
SetOption63 1 did not fix the problem.
However I was unable to add the code.

arendst, 2019-04-18:
@BostonMikeGetHub your version does not contain the fix.

It should be in Hackbox soon as commit e8e7b25

meingraham, 2019-04-18:
Mike, you need to OTA from http://thehackbox.org/tasmota/020300/ when the new binary is ready.

meingraham, 2019-04-18:
`SetOption63 1` resolved my issue.

So I can document it properly in the wiki, `On` enables or disables scanning at restart?

arendst, 2019-04-18:
ON (1) disables scanning at restart (default is OFF (0))

BostonMikeGetHub, 2019-04-18:
I won't be available to verify the issue is resolved on my device until the weekend.

BostonMikeGetHub, 2019-04-18:
Thank you Theo and everyone for your help. 

I got some time for testing. I found one case where their is still a problem.

```
Tasmota Info:
Program Version6.5.0.9(b800db4-sonoff)
Build Date & Time2019-04-18T18:01:37
Core/SDK Version2_3_0/1.5.3(aec24ac9)
```

My issue is resolved when PoweOnState is 3 for both lost then return of power and when restarting the device. 

When PoweOnState is 0 and the relay is on:

The issue is resolved with lost then return of power.  (Relay is off after power up)

However, the relay remains on when restarting the device using its web site. 

To me, it is important the power on state is correct after lost then return of power.   A power lost could be beyond the user knowage.  But the user will be the one rebooting so they can easily correct the relay state. 

So I am satisfied and will close this bug upond Theo's or @ascillato2 request. 

Thank you again.

BostonMikeGetHub, 2019-04-18:
Console of error case: poweronstate 0, Relay on, restart use web site. 

```
00:00:00 CFG: Loaded from flash at F5, Count 442
00:00:00 Project sonoff PMPlug-6 Version 6.5.0.9(b800db4-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 UAPAC01 in mode 11N as PMPlug-6-8146...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on PMPlug-6-8146 with IP address 192.168.1.66
00:00:03 RSL: INFO1 = {"Module":"AWP04L","Version":"6.5.0.9(b800db4-sonoff)","FallbackTopic":"cmnd/DVES_6C1FD2_fb/","GroupTopic":"sonoffs"}
00:00:03 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"PMPlug-6-8146","IPAddress":"192.168.1.66"}
00:00:03 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:03 RSL: RESULT = {"POWER":"ON"}
00:00:03 RSL: POWER = ON
19:53:34 RSL: STATE = {"Time":"2019-04-18T19:53:34","Uptime":"0T00:00:12","Vcc":3.137,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"UAPAC01","BSSId":"80:2A:A8:51:E3:F4","Channel":11,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:53:34 RSL: SENSOR = {"Time":"2019-04-18T19:53:34","ENERGY":{"TotalStartTime":"2019-04-03T20:07:17","Total":4.172,"Yesterday":0.380,"Today":1.763,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":124,"Current":0.000}}
19:53:49 CMD: Status 0
19:53:49 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["PMPlug-6"],"Topic":"PMPlug-6","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
19:53:49 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:27","StartupUTC":"2019-04-18T23:53:22","Sleep":50,"CfgHolder":4617,"BootCount":97,"SaveCount":443,"SaveAddress":"F4000"}}
19:53:49 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0.9(b800db4-sonoff)","BuildDateTime":"2019-04-18T18:01:37","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:53:49 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UAPAC01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008021","280500000100000000000000000000000000","00002008"]}}
19:53:49 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":508,"Free":492,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","01003BC0"]}}
19:53:49 RSL: STATUS5 = {"StatusNET":{"Hostname":"PMPlug-6-8146","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:6C:1F:D2","Webserver":2,"WifiConfig":4}}
19:53:49 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 18 23:53:49 2019","Local":"Thu Apr 18 19:53:49 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"05:58","Sunset":"19:28"}}
19:53:49 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:53:49 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-04-18T19:53:49","ENERGY":{"TotalStartTime":"2019-04-03T20:07:17","Total":4.172,"Yesterday":0.380,"Today":1.763,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":124,"Current":0.000}}}
19:53:49 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-04-18T19:53:49","Uptime":"0T00:00:27","Vcc":3.137,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"UAPAC01","BSSId":"80:2A:A8:51:E3:F4","Channel":11,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

arendst, 2019-04-19:
This is a very old issue and currently as designed.

Bottomline is that a restart is different from a power cycle. During a restart PowerOnState is NOT checked for the simple reason the relays are being set to the state before the restart. A restart is supposed to happen unexpectedly and should be virtually unnoticed for the relay state.

In this case it's no different; PowerOnState is not being used when a restart is executed for the simple reason it's not a power On situation.

This functionality was discussed at that time (see old issues).

meingraham, 2019-04-19:
OK, now the obvious strikes! PowerOnState means exactly what it says - when power is turned on.

Is not scanning the relay state on restart (i.e., `SetOption63 1`) equivalent to `keep relay(s) OFF after restart`? By that then, is `SetOption63 0` equivalent to `turn relay(s) on as last saved` (A.K.A. leave them as they were when the device was restarted)? Or is the relay state on restart set from other Tasmota settings in effect?

arendst, 2019-04-19:
Scanning the relay state has, in my eyes, always been a very strange piece of code... It was added to satisfy ElectroDragon users but what it does is reading from GPIO's configured as OUTPUT! The result, as you noticed will not always be what you expect as it all depends on how the relays are hardwired to the GPIO. To make this scan optional command SetOption63 was introduced.

Tasmota keeps score of the relays state in a masked variable called "power". A 1 in a bit location always means a relay turned ON (depending on configured relay as relay1 or relay1i the GPIO is turned low or high). Every change in relay state is only executed after a change in "power". When "power" changes a copy is saved in flash.

So after a restart "power" is reinitialised with the saved state from flash and the relay is set to the state that "power" is in. PowerOnState is not used in this case.

meingraham, 2019-04-19:
If `SetOption63 = 0`; yes?
> So after a restart "power" is reinitialised with the saved state from flash and the relay is set to the state that "power" is in.

If `SetOption63 = 1`, on restart (i.e., not power on), Tasmota does nothing to set the state of relay(s). They come back after reboot in whatever state the hardware puts them in. Is this accurate?

arendst, 2019-04-19:
Nope.

In both cases the relays are set as stored in "power".

SetOption63 is executed AFTER PowerOnState or Restart init.

The difference with SetOption63 is that when disabled (0) it will scan the relay GPIO's (outputs) and read it's state using "digitalRead". It will then update the "power" variable with it's detected values. No relay states will be changed. These values may be different from what there was before the relays were set. This solved the ElectroDragon issue but made your issue. Sonoff devices do not need this but were not affected by it's functionality.

meingraham, 2019-04-19:
This may have done it... put it beyond my ability to comprehend :wink:

So, with `SetOption63 1`, on a restart, it may leave the internal power bits in a state that differs from the actual output GPIOs since the Tasmota scan of the output GPIOs is not performed?

arendst, 2019-04-19:
It should never do a scan. There is no valid way of getting input status from an output configured GPIO.

What I tried to say is that for the ElectroDragon it worked. For many others it also worked and didn't change anything. It just doesn't work for your and Mike's device. There should be no need to run this piece of code.

Weekend!

meingraham, 2019-04-19:
TTFN. Thanks.

ascillato2, 2019-04-19:
Closing this issue as it is solved. Thanks everyone :+1:

closed 2019-04-19 by ascillato2

BostonMikeGetHub, 2019-04-20:
@arendst

Thank you for fixing the bug, the explanation and for creating & maintaining tasmota.