arendst/Tasmota #5595: Many devices crash after update to 6.5.0.5

tobox, 2019-04-07:
### BUG DESCRIPTION

I flashed 3 Devices with my own compiled 6.5.0.5, without any code changes (just some settings, i.e. Wifi), using core 2.3.0.They all worked properly. Then I updated around 20 other devices, and 5 of them seem to be stuck in a boot/crash loop:

```

00:00:00 CFG: Loaded from flash at FA, Count 696
00:00:00 Project sonoff Sonoff-Dose-5 Version 6.5.0.5(sonoff)-2_3_0
00:00:04 WIF: Connecting to AP1 ToBoX in mode 11N as Sonoff-Dose-5...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on Sonoff-Dose-5 with IP address 192.168.207.110
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff-dose-5/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff-dose-5/POWER = 
00:00:07 MQT: tele/sonoff-dose-5/INFO1 = {"Module":"Sonoff Basic","Version":"6.5.0.5(sonoff)","FallbackTopic":"cmnd/DVES_1528
56_fb/","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff-dose-5/INFO2 = {"WebServerMode":"Admin","Hostname":"Sonoff-Dose-5","IPAddress":"192.168.207.110"}
00:00:07 MQT: tele/sonoff-dose-5/INFO3 = {"RestartReason":"Power on"}
00:00:07 MQT: stat/sonoff-dose-5/RESULT = {"POWER":"OFF"}
00:00:07 MQT: stat/sonoff-dose-5/POWER = OFF

Exception (9):
epc1=0x401042b9 epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys 
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  000000c5 00000001 40103a66 3ffee7d8  
3fffff60:  4023891c ffffffff 00000000 3ffe8004  
3fffff70:  402391d3 ffffffff 00000000 40239144  
3fffff80:  3ffedda8 40238abe 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffea6f0 00000001 4020e35f  
3fffffa0:  3ffea6f0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

```

I cannot provide STATUS0 output, because the devices crash immediately. Only reflashing the devices seemed to help only on some devices, I erased the whole flash and and reconfigured them again.

tobox, 2019-04-07:
I initiated the update using MQTT, by setting Upgrade to 1. I did not change the OtaUrl, and all devices flashed the sonoff.bin directly (without flashing MINIMAL first). I remember that during other updates, the devices automatically installed minimal first and then the main firmware?

ascillato, 2019-04-07:
Sorry, I can't reproduce your problem. Something has gone wrong in your own compilation. 

Please, try to do a reset 5

If that don't solve your issue, try the precompiled bins.

ascillato2, 2019-04-09:
Please, when you have time, tell us if reseting or using the precompiled bins, solve your issue. Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-09 by ascillato2

tobox, 2019-04-09:
I was unable to track down the error any further. My own compiled binaries are small enough to update without minimal firmware, so that was probably not the reason for the crashes. Crashing devices were Sonoff-POW, POW R2, Basic, Touch and Generic (Wemos D1 Mini), so it was probably not related to the specific hardware. For the devices that were built into walls, I was able to recover them by running the following shell code and flipping the fuse to power-cycle them:

```
while `sleep 0.2` 
do 
     mosquitto_pub -h localhost -p 1883 -u USER -P PASSWORD -t cmnd/sonoff-rf-wohnzimmer/Reset -m "5"
done
```
After that, I reflashed my own compiled firmware again, without any issues. I checked the loggfiles of my HTTP server and verified that the binaries were downloaded properly during the first update. 

Lessons learned: always keep backups of the configuration files (I am running a download script weekly now), and don't update all devices at once.

Thanks for the help.