arendst/Tasmota #5596: Yunshan reset to factory

rebelian, 2019-04-07:
Yunshan TASMOTA 6.5 relay module (sonoff.bin)
set Module type: Yunshan relay
set Configure Template: Yunshan relay

this is my configuration:
https://ucexperiment.files.wordpress.com/2016/12/wifi_relay2.jpg?w=600&h=350
A: 7-30V + DC power supply
B: Power supply ground
C: Normally closed (NC) relay contact
D: Common (COM) relay contact
E: Normally open (NO) relay contact
F: 5V + out
G: ESP8266 GPIO5 Optocoupler Input
H: Ground (isolated optocoupler input)

ground (A) I have a connect  with (H)

if I short for a short time (max 2s) (F G) everything is ok:
console out:
15:00:10 APP: Button1 multi-press 1
15:00:10 SRC: Button
15:00:10 MQT: domoticz/in = {"idx":2,"nvalue":1,"svalue":"","Battery":83,"RSSI":3}
15:00:10 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
15:00:10 MQT: stat/sonoff/POWER = ON

 - but if I make a connect  (F G) and leave it after a few seconds (10-15s), the system resets and returns to the factory settings, why ?

console:
15:08:40 APP: Button1 multi-press 1
15:08:40 APP: Button1 multi-press 2
15:08:41 SRC: Button

Jason2866, 2019-04-07:
Cant reproduce. My Yunshan device works without any issue.

ascillato2, 2019-04-09:
Template missing.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md)).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-09 by ascillato2