arendst/Tasmota #5597: Sonoff Rf Bridge V2 not reading Sonoff PIR2 output

sndipz, 2019-04-07:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I have Flashed Sonoff RF V2 Bridge with Tasmota 6.5.0. I have also successfully configured MQTT as well.
I am trying to read Sonoff PIR2 using this RF Bridge, however, on the Console, I could see information only related to Uptime and State. It is not capturing Data. 

Also, when ever a motion is detected on Sonoff PIR2, I could see the RED Led blinking (assuming it as Hardware acknowledging the signal sent from PIR2), but console does not show any information about Data captured.

00:00:00 APP: Set Baudrate to 19200
00:00:00 Project sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 WIFISSID in mode 11N as sonoff-1858...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-1858 with IP address IP.AD.DR.ES.S
00:00:06 MQT: Attempting connection...
15:13:47 MQT: Connected
15:13:47 MQT: tele/sonoff/LWT = Online (retained)
15:13:47 MQT: cmnd/sonoff/POWER = 
15:13:47 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Bridge","Version":"6.5.0(release-sonoff)","FallbackTopic":"cmnd/DVES_BFC742_fb/","GroupTopic":"sonoffs"}
15:13:47 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1858","IPAddress":"192.168.0.116"}
15:13:47 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
15:13:56 MQT: tele/sonoff/STATE = {"Time":"2019-04-07T15:13:56","Uptime":"0T00:00:16","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"Wifi":{"AP":1,"SSId":"WIFISSID","BSSId":"BSSID","Channel":10,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:18:57 MQT: tele/sonoff/STATE = {"Time":"2019-04-07T15:18:57","Uptime":"0T00:05:17","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"Wifi":{"AP":1,"SSId":"WIFISSID","BSSId":"BSSID","Channel":10,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:23:42 APP: Serial logging disabled
15:23:58 MQT: tele/sonoff/STATE = {"Time":"2019-04-07T15:23:57","Uptime":"0T00:10:17","Vcc":3.159,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"WIFISSID","BSSId":"BSSID","Channel":10,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:05"}}

Jason2866, 2019-04-07:
Please post status 0 output and log file when you fire the PIR

sndipz, 2019-04-07:
Status 0 output.

19:03:58 MQT: stat/sonoff/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:03:59 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:35","StartupUTC":"2019-04-07T18:03:23","Sleep":50,"CfgHolder":4617,"BootCount":12,"SaveCount":20,"SaveAddress":"F8000"}}
19:03:59 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-classic)","BuildDateTime":"2019-03-19T12:25:42","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:03:59 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFISSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
19:03:59 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","6010A390","008403A8","210217C6","000000C0"]}}
19:03:59 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1858","IPAddress":"HOME.ASSIST.MQTTBROKER.IP","Gateway":"GATEWAY.IP","Subnetmask":"255.255.255.0","DNSServer":"GATEWAY.IP","Mac":"MACADD","Webserver":2,"WifiConfig":4}}
19:03:59 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"MQTTHOST","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BFC742","MqttUser":"MQTTUSERNAME","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:03:59 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 07 18:03:59 2019","Local":"Sun Apr 07 19:03:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
19:03:59 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-04-07T19:03:59"}}
19:03:59 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-04-07T19:03:59","Uptime":"0T00:00:36","Vcc":3.156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"WIFISSID","BSSId":"BSSID","Channel":10,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:05"}}}

I would need help in getting log file. All I can see is console Log. WHen Sonoff PIR2 is triggered (by Motion), I could only see Red Led blink on the SOnoff RF Bridge and nothing on console.

Jason2866, 2019-04-08:
Tasmota DOESNT change anything in the RF receiving part. 
Have you closed the little switch on the PCB after successful flash of Tasmota?
If not you will receive nothing ;-)

herik63, 2019-04-09:
I have similar issue, my rfbridge is receiving signal only when I digit rfraw 177 in console, i successfully used it for triggering my backyard door and my led strip, but now I've just received some door sensors and I'm a little bit stuck.

does make sense that at power on the rfbridge is not receiving?

Jason2866, 2019-04-09:
What do you mean command via bridge isnt working? Can you explain?

herik63, 2019-04-10:
hallo, my previous configuration wasn't using sensor, only I had a light switch and a door switch, those used this form:
- platform: mqtt
    name: "LED strip"
    command_topic: "cmnd/sonoffbridge/backlog"
    payload_on: "rfraw AA B0 21 03 08 01C2 0492 2F62 28181818181819081818181908181819081818181818181908 55;rfraw 177"
    payload_off: "rfraw AA B0 21 03 08 01C2 0492 2F62 28181818181819081818181908181819081818181818181908 55;rfraw 177"
    retain: false

now, when the rfraw is triggered the unit sends back from the sensor meaningless data, is possible to anchieve the correct results putting rfraw 0: 

07:16:29 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA436BA01CC0546336D0E55"}}
07:16:31 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA436B001CC0546336D0A55"}}
07:16:41 CMD: rfraw 0
07:16:41 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"OFF"}
07:16:49 MQT: tele/sonoffbridge/RESULT = {"RfReceived":{"Sync":14010,"Low":460,"High":1360,"Data":"336D0A","RfKey":"None"}}
07:16:51 MQT: tele/sonoffbridge/RESULT = {"RfReceived":{"Sync":13930,"Low":470,"High":1350,"Data":"336D0A","RfKey":"None"}}
07:19:04 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
07:19:04 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
07:19:18 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA4367E01CC0546336D0A55"}}
07:19:20 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA436F601D60546336D0E55"}}

Jason2866, 2019-04-10:
It looks like that using RfRaw isnt necessary anymore. Code is recognized in standard modus

sndipz, 2019-04-10:
> Solved! i had a firmware that was old or incorrect, with the one found in
> latest release seems working!
> RF-Bridge-EFM8BB1-20181127.hex seems fine, now the bridge is receiving by
> default and also the message seems fine with my door sensors.
>
> edit: now the command via bridge are not working anymore :(

Hi Herik,

Can you please share how you updated firmware and also option to check
existing firmware.

I am a new to all this. All I am aware of is to Tasmota Sonoff using
Espeasy tool.

Also, I have looked in wiki, I could find the instructions (may be I am
looking at wrong place),

Any help is much appreciated.

herik63, 2019-04-10:
Hi, I uploaded the firmware via the usual FTDI adapter, following the normal path :) I left the 2 wires for upgrading the EFM8BB1, i needed to learn some commands and to send via HA.

now I have uploaded via OTA the latest release of Tasmota: 

Program Version | 6.5.0.6(0dbbf41-sonoff)
-- | --
2019-04-09T19:01:51
2_5_0/3.0.0-dev(c0f7b44)

and also via OTA I uploaded the Keil 8051 v9.53 version of Portisch.

unfortunately when the RfRaw is on the feedback from the sensors is still garbage, thus I'm trying to program the few keys I need directly on the Tasmota interface, and keep RfRaw off.

herik63, 2019-04-10:
no way, also with different version the behaviour is the same: when powered on the read RF data make sense, when a "RFraw" command is issued the next rf reading are garbage, is a problem of mine?

00:00:00 APP: Set Baudrate to 19200
00:00:00 Project sonoff Sonoff (Topic sonoffbridge, Fallback DVES_B2ECA2, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: Connecting to AP1 Home Sweet Home in mode 11N as sonoffbridge-3234...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoffbridge-3234.local with IP address 192.168.2.223
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoffbridge/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoffbridge/POWER = 
00:00:07 MQT: tele/sonoffbridge/INFO1 = {"Module":"Sonoff Bridge","Version":"6.2.1","FallbackTopic":"DVES_B2ECA2","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoffbridge/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffbridge-3234","IPAddress":"192.168.2.223"}
00:00:07 MQT: tele/sonoffbridge/INFO3 = {"RestartReason":"Software/System restart"}
09:26:13 MQT: tele/sonoffbridge/STATE = {"Time":"2019-04-10T09:26:13","Uptime":"0T00:00:16","Vcc":3.182,"Wifi":{"AP":1,"SSId":"Home Sweet Home","RSSI":56,"APMac":"20:C9:D0:1F:46:49"}}
09:27:42 MQT: tele/sonoffbridge/RESULT = {"RfReceived":{"Sync":14050,"Low":470,"High":1350,"Data":"336D0E","RfKey":"None"}}
09:27:45 MQT: tele/sonoffbridge/RESULT = {"RfReceived":{"Sync":13960,"Low":460,"High":1350,"Data":"336D0A","RfKey":"None"}}
09:27:49 MQT: tele/sonoffbridge/RESULT = {"RfReceived":{"Sync":13940,"Low":460,"High":1350,"Data":"336D0A","RfKey":"None"}}
09:28:00 MQT: stat/sonoffbridge/RESULT = {"RfKey1":"Learned sent"}
09:28:01 MQT: stat/sonoffbridge/RESULT = {"RfKey1":"Learned sent"}
09:28:11 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:12 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:13 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:14 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:14 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:15 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:16 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:16 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:17 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:17 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
09:28:27 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA475B201CC0546336D0E55"}}
09:28:30 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA4368801CC0546336D0A55"}}
09:28:32 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA4366A01D60546336D0E55"}}
09:28:33 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA436B001CC0546336D0A55"}}
09:28:35 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA4367E01CC0546336D0E55"}}
09:28:36 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA4366A01CC0546336D0A55"}}
09:28:37 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA4294A01D60546336D0A55"}}

Jason2866, 2019-04-10:
Update your sonoff version to latest. You use a old version...

herik63, 2019-04-11:
Hi, that was another test, now I'm using the latest version but the issue is still present

Jason2866, 2019-04-11:
Please visit us in [Discord](https://discord.gg/Ks2Kzd4). More user experience :-)

herik63, 2019-04-11:
ok :)

ascillato2, 2019-04-12:
Hi, any news on this?

herik63, 2019-04-12:
Hi, 
I tried to reset the module, to erase the entire flash and reflash, but the issue is still there, I managed to workaround it adding a "rfraw 0" at the end of the command, like this:

- platform: mqtt
    name: "LED strip"
    command_topic: "cmnd/sonoffbridge/backlog"
    payload_on: "rfraw AA B0 21 03 08 01C2 0492 2F62 28181818181819081818181908181819081818181818181908 55;rfraw 0"
    payload_off: "rfraw AA B0 21 03 08 01C2 0492 2F62 28181818181819081818181908181819081818181818181908 55;rfraw 0"
    retain: false

it seems working, the string are not garbage, I needed to flash the Portisch firmware, or the command will not work.

ascillato2, 2019-04-13:
Great to know that it is working for you now

ascillato2, 2019-04-13:
@sndipz 

Hi, 

Any news on this?

sndipz, 2019-04-13:
Hi,

I will be sharing my information by End of Sunday IST.

I am new to all of this, I will try respond quick.

Thanks!

ascillato2, 2019-04-14:
When you have time, please, tell us if the solution proposed by @herik63 solved your issue. Thanks 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-14 by ascillato2

sndipz, 2019-04-14:
It turn out to be a faulty Switch issue. I had to hold it for Console to
show the Receive signal.

For me, it was an hardware issue.

Sorry for holding it long.

sagedamscus, 2019-07-08:
> It turn out to be a faulty Switch issue. I had to hold it for Console to show the Receive signal. For me, it was an hardware issue. Sorry for holding it long.
> […](#)

Which switch on your unit was faulty i am having same issue. Was it the on /off switch on the sonoff rf bridge for programming ?