arendst/Tasmota #5598: Template Activation Causes Device Reset

meingraham, 2019-04-07:
### BUG DESCRIPTION
Configuring a device using the Generic Module works as expected. When trying to convert the configuration to a Template, activating the template causes the device to reset back to Sonoff Basic.

The Kesen KS-604S is working with Tasmota using Generic (18). I can live with it as such.

I was thinking that perhaps there is an issue in the Template facility (i.e., not _necessarily_ the device) which could be resolved through this troubleshooting. @digiblur mentioned that he'd seen other similar references regarding Template use (i.e., boot loop resets). Perhaps there is a lurking bug.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Kesen KS-604S
- [x] Tasmota binary firmware version number used : 6.5.0.5 pre-compiled
- [x] Development IDE - Compiler / Upload tools used : N/A / Tuya-convert
- [x] Provide the output of command ``status 0`` :
```
STATUS = {"Status":{"Module":18,"FriendlyName":["Upper","Lower","USB","Upper Indicator"],"Topic":"kesen","ButtonTopic":"","Power":4,"PowerOnState":0,"LedState":0,"SaveData":1,"SaveState":0,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020300/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:31","StartupUTC":"2019-04-07T15:17:47","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":41,"SaveAddress":"FA000"}}
STATUS2 = {"StatusFWR":{"Version":"6.5.0.5(387e3c5-sonoff)","BuildDateTime":"2019-04-06T20:02:23","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008108","280500000100000000000000000000000000","000000C0"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":505,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
STATUS5 = {"StatusNET":{"Hostname":"kesen-<deviceID>","IPAddress":"192.168.1.<nnn>","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"EC:FA:BC:<myMAC>","Webserver":2,"WifiConfig":4}}
STATUS6 = {"StatusMQT":{"MqttHost":"myMQTTHost","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_<myMAC>","MqttUser":"myUserID","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 07 15:18:18 2019","Local":"Sun Apr 07 11:18:18 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"07:05","Sunset":"19:52"}}
STATUS10 = {"StatusSNS":{"Time":"2019-04-07T11:18:18"}}
STATUS11 = {"StatusSTS":{"Time":"2019-04-07T11:18:18","Uptime":"0T00:00:31","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"mySSID","BSSId":"<myBSSID>","Channel":6,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

- Merge the module configuration into a Template
- Change the Template name...
- Turn off A0 for good measure (have tried this with and without A0)

```
CMD: template 255
RESULT = {"NAME":"Merged","GPIO":[56,255,255,17,32,18,255,255,22,21,33,23,255],"FLAG":1,"BASE":18}
CMD: template
RESULT = {"NAME":"KS-604S","GPIO":[56,255,255,17,32,18,255,255,22,21,33,23,255],"FLAG":0,"BASE":18}
CMD: weblog 4
MQT: stat/kesen/RESULT = {"WebLog":4}
```

- [x] Provide the output of console when you experience your issue if apply :

The output resulting from the Activate and Save action:
```
CFG: Saved to flash at F6, Count 45, Bytes 3584
WIF: Checking connection...
WIF: Connected
HTP: Configuration
HTP: Configure Other
HTP: Configure Other
OTH: MQTT enable ON, Emulation 0, FriendlyName Upper, Lower, USB, Upper Indicator
SRC: WebGui from 192.168.1.42
RSL: Received Topic /Template, Data Size 90, Data {"NAME":"KS-604S","GPIO":[56,255,255,17,32,18,255,255,22,21,33,23,255],"FLAG":0,"BASE":18}
RSL: Group 0, Index 1, Command TEMPLATE, Data {"NAME":"KS-604S","GPIO":[56,255,255,17,32,18,255,255,22,21,33,23,255],"FLAG":0,"BASE":18}
MQT: stat/kesen/RESULT = {"NAME":"KS-604S","GPIO":[56,255,255,17,32,18,255,255,22,21,33,23,255],"FLAG":0,"BASE":18}
SRC: WebGui from 192.168.1.<nnn>
RSL: Received Topic /Module, Data Size 1, Data 0
RSL: Group 0, Index 1, Command MODULE, Data 0
MQT: stat/kesen/RESULT = {"Module":"0 (KS-604S)"}
HTP: Restart
SRC: WebGui from 0.0.0.0
CFG: Saved to flash at F5, Count 46, Bytes 3584
MQT: Subscribe to cmnd/sonoffs/#
MQT: Subscribe to cmnd/DVES_<myMAC>_fb/#
MQT: tele/kesen/INFO1 = {"Module":"Sonoff Basic","Version":"6.5.0.5(387e3c5-sonoff)","FallbackTopic":"cmnd/DVES_<myMAC>_fb/","GroupTopic":"sonoffs"}
MQT: tele/kesen/INFO2 = {"WebServerMode":"Admin","Hostname":"kesen-<deviceID>","IPAddress":"192.168.1.<nnn>"}
MQT: tele/kesen/INFO3 = {"RestartReason":"Software/System restart"}
```

After the device finally successfully boots:
```
CMD: template
MQT: stat/kesen/RESULT = {"NAME":"KS-604S","GPIO":[56,255,255,17,32,18,255,255,22,21,33,23,255],"FLAG":0,"BASE":18}
CMD: module
MQT: stat/kesen/RESULT = {"Module":"1 (Sonoff Basic)"}
```

### TO REPRODUCE

- Merge the Template and rename it.
- Separate console window open to capture the log (level 4)
- Activate the Template and click save
- Device restarts and immediately (i.e., no 40 second pause) a relay begins to click (toggle) multiple times. After a few (which I have to assume are boot loops, but I have not counted how many), the device boots and is reset back to Sonoff Basic. Wi-Fi, MQTT, etc. appear to all be as before. The custom template is still there.
- To get back to full operation I restore my configuration backup. This puts the configuration back to a Generic (18) with the configured GPIOs.
- Device functions properly (using Generic rather than the custom Template)

BTW, I set `SetOption36` to 0... but bootloop reset still occurs.

Any additional information I should trap to try to troubleshoot this?

### EXPECTED BEHAVIOUR
When creating a Template from a working device configuration, activating the Template should not result in a configuration reset.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

arendst, 2019-04-08:
Just for my information, what is a KS-604S? When searching the net I find this:

https://hidin.en.alibaba.com/productgrouplist-807823258/KS_604_US_Style_Smart_Wall_Socket.html

A two socket smart wall socket. 

How does that equal a configuration with five (!) relays based on your template:

[image]

blakadder, 2019-04-08:
Its this one (allegedly, I don't own it either) https://blakadder.github.io/templates/kesen_KS-604S.html

Jason2866, 2019-04-08:
It is a mess anyway because settings doesnt match with text!
[image]

But there is a error behaviour! If you set (starting from generic(18)):
GPIO2 = Led1
GPIO5 = Button1
GPIO12 = Relay2
GPIO13 = Relay1
GPIO16 = Button2

It resets to Sonoff Basic. Button2 on GPIO16 is doing this

arendst, 2019-04-08:
Bug is a regression from https://github.com/arendst/Sonoff-Tasmota/commit/5e1cbcf94fc10f338b1d95258a0b7019df8d5e79

Will fix soon.

arendst, 2019-04-08:
The issue here is NOT a bootloop problem due to misconfiguration but the fact that a code regression from 6.5.0.4 did not allow activation or selection of a user-template; it just selected the default one.

Because of this the SetOption36 feature wasn't active either.

Should be ok now.

meingraham, 2019-04-08:
Theo,

Thanks for looking into this and finding the issue... and fixing it so quickly! I'll update the firmware later today and confirm that the fix took care of the behavior on my device.

[This is a better description of the current state of the device configuration](https://github.com/arendst/Sonoff-Tasmota/wiki/Kesen-KS-604S). The template was created by a different user and may have not been completely configured. Also, the device GPIO hardware setup for the devices I just received appears to have changed.

The reason I have defined five relays is that three control the physical devices (two receptacles and the USB port) plus I have two "virtual relays" to control the power indicator LEDs because configuring them as LEDs did not function as it should. @digiblur helped me get that setup right. I'd described this in the Tasmota wiki.

I submitted an edit to the Templates wiki page for the [KS-604S](https://blakadder.github.io/templates/kesen_KS-604S.html). I sent a note on Discord to @blakadder that I was not able to complete the pull request.

Cheers!

Mike

meingraham, 2019-04-08:
Fix confirmed.

closed 2019-04-08 by meingraham

agecrabahaykubo, 2019-07-22:
I have the same issue when using a user created template.  It seems not holding the template configuration and the device will reset back into sonoff-xxxx, needed to access again 192.168.4.1 to enter wifi credentials etc.

{"NAME":"Viflykoo 3xStr","GPIO":[0,0,53,0,0,23,0,0,21,56,17,24,22],"FLAG":1,"BASE":18}

digiblur, 2019-07-22:
Does this one hold on?  

{"NAME":"Viflykoo 3xStr","GPIO":[0,0,53,0,0,23,0,0,21,56,9,24,22],"FLAG":1,"BASE":18}

agecrabahaykubo, 2019-07-23:
Thanks for the response @digiblur. I managed to get the correct configuration of the relays thru trial & error.  For now I will stick to this settings and not use the template.

Here's the basic configuration that I am using to toggle the relays ON/OFF .
D6 - Relay1 (21)
D5 - Relay2 (22)
D8 - Relay3 (23)
D1 - Relay4 (24)

BTW, what is Led2(53),  Led1i(56) & Switch1(9) for?