arendst/Tasmota #5599: Updating channel4 and channel5 not updating CT value on Sonoff B1

benas1989, 2019-04-07:
I try to update individual color channel on sonoff B1. 
This is my output from web console

19:40:30 CMD: channel4 100
19:40:30 MQT: tasmota/stat/Sonoff_B1/CHANNEL = {"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"0,0,0","Channel":[0,0,0,100,0],"CT":153}
19:40:34 CMD: channel5 100
19:40:34 MQT: tasmota/stat/Sonoff_B1/CHANNEL = {"POWER":"ON","Dimmer":100,"Color":"000000FFFF","HSBColor":"0,0,0","Channel":[0,0,0,100,100],"CT":500}

blakadder, 2019-04-07:
I don't understand the issue. Log is showing CT changes after channel<x> commands

benas1989, 2019-04-08:
wrong CT value. full cold and full warm white is 326.5 CT value ant dimmer value 100%

blakadder, 2019-04-08:
Technically you shouldn't be able to go full CW and WW so the console is displaying the highest CT value. When using the CT slider in Tasmota or a smart home hub you cannot put both channels above 50 at the same time.

I'm wondering which formula you used to calculate that.

ascillato2, 2019-04-12:
Hi, any news on this?

ascillato2, 2019-04-14:
Closing this issue as there is no feedback. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-14 by ascillato2