arendst/Tasmota #5600: Unsupported "xy" color encoding from Alexa App in Philips Hue emulation

shadinger, 2019-04-07:
I'm using the Arilux LC01 module with Tasmota 6.5 to control LED, through Philips Hue emulation. When using the Alexa app on iOS: brightness works well but color selection has no effect.

Looking at the logs, I realized that the Alexa app (iOS) sens HSI (xy) color information instead of bri/hue/sat. Looking in the source, it appears that "xy" color space is currently not supported.

Ex (blue):
	`20:05:24 HTP: Hue POST args ({ "on": true, "xy": [0.13550301400290363,0.03987867049354716] })`

I found here how to convert xy to rgb: http://blog.saikoled.com/post/44677718712/how-to-convert-from-hsi-to-rgb-white

It would be great to add "xy" color space to Philips Hue emulation.

Jason2866, 2019-04-08:
Color isnt supported in Tasmota. 
The fact color functions a little is just luck...
If you can bring to work (without big code increase) a PR is highly appreciated

Jason2866, 2019-04-08:
Duplicate see https://github.com/arendst/Sonoff-Tasmota/issues/4809

shadinger, 2019-04-08:
Adding xy color space does not seem like a big change. I'll work later on it and hopefully submit a PR.

Interestingly, when Alexa detects a light supporting White Temperature and Color, it does work.

As far as I can see, Alexa sends an "xy" color and listens the actual color back. Since the "xy" value is currently hardcoded to [0.5, 0.5], it always returns Orange.

ascillato2, 2019-04-09:
>Adding xy color space does not seem like a big change. I'll work later on it and hopefully submit a PR.

Great!!! If you need help with the code, you can find us in the Tasmota Support Chat or you can submit part of your PR and we can discuss it there.

Waiting for your PR :+1:

PR #5667 (2019-04-24): Support for color and colortone in Philips Hue emulation via Alexa
## Description:

This provides full support for colors and white color-tone to the Philips Hue emulation via Alexa devices or Alexa smartphone app.

Specifically the Alexa app is very picky about receiving the exact same value that it sent initially. For ex, if it sends Brightness of '43', sometimes due to rounding errors it would send back '42'. The new version sends exactly the same value to the app.

 - to avoid receiving XY colors, don't send any XY colors in HUE lights status
 - adding rounding when converting from float t [... cut]
closed 2019-04-09 by ascillato2