arendst/Tasmota #5601: How to determine Smart Bulb Module Parameters

garnern2, 2019-04-08:
I have successfully flashed a non-upgraded Teckin Smart Bulb (https://www.amazon.com/Dimmable-Multicolor-Compatible-TECKIN-Equivalent/dp/B07GT2Q5RW). I can access the bulb's settings in the Tasmota web interface, but I cannot for the life of be set the parameters so that the bulb works as expected. I've attempted to use the color commands and can only get green, white, and occasionally red--albeit in the wrong order. I know this is because my parameter settings are incorrect, but is there a way that I can determine what they should be? I'm willing to go the trial and error route, but I've been doing that four 5 hours and have still had no luck.

If this is a dumb question, I sincerely apologize--but I have indeed attempted to find the answer myself and I'm now at an impasse.

Jason2866, 2019-04-08:
Take a look at https://github.com/arendst/Sonoff-Tasmota/issues/4883
You have enter manualy in the web console the command
Color 1 should be light red
Color 2 should be light green
Color 3 should be light blue
Read the Wiki -> Commands
If you read https://github.com/arendst/Sonoff-Tasmota/issues/1867 you get a idea how it works...

Jason2866, 2019-04-08:
You can try ready made templates too. Maybe one fits to your bulb
https://blakadder.github.io/templates/light.html

blakadder, 2019-04-08:
Here you go https://blakadder.github.io/templates/teckin-sb50.html

closed 2019-04-09 by ascillato2