arendst/Tasmota #5602: Led Status toggle 1 and toggle 2 sonoff-display.bin

oigroigm, 2019-04-08:
I installed the display version on a sonoff basic but when the toggle 2 turns on the display (i2c oled 128x64) the LED associated with relay 1 also lights up, it is possible to separate the status of the led

arendst, 2019-04-08:
Search wiki for ``ledstate``

oigroigm, 2019-04-08:
I searched but nothing solves the problem, I can only completely turn off the LED, I would like it to turn on only when toggle 1 is active

Jason2866, 2019-04-08:
Try, not tested, a (virtual) Relais2. This should split Led1 and Led2

arendst, 2019-04-09:
With the default Sonoff Basic template it's not possible to separate the led from any GPIO. 

You could edit the template by changing the GPIO connected to the led (Led1i) with another relay which allows to control the led as you like albeit.

oigroigm, 2019-04-09:
I use the release sonoff-display.bin with generic configuration (18) configured in the following way

D3 GPIO0 Button1        Button1 (17)
TX GPIO1 Serial Out      I2C SCL (5)      display 0led 128x64
D4 GPIO2                      AN2301 (2)      temperature & humidity sensor
RX GPIO3 Serial In        I2C SDA (6)      display oled 128x64
D2 GPIO4
D1 GPIO5
D6 GPIO12 Relay1         Relay1 (12)
D7 GPIO13 Led1i           Led1i (56)
D5 GPIO14 Sensor
D8 GPIO15

the sonoff LED lights up both when I press the sonoff button and when I turn on the display I would like to know if it is possible to separate the switching on of the LED from the display on.

arendst, 2019-04-09:
See my previous comment.

oigroigm, 2019-04-09:
Can you explain how to do it?

arendst, 2019-04-09:
Change GPIO13 to relay2
Make a rule which will toggle relay2 when relay1 is switched.

See wiki how to use rules

oigroigm, 2019-04-09:
thank you I will try

oigroigm, 2019-04-09:
Perfect so it works well, thanks.

closed 2019-04-09 by oigroigm