arendst/Tasmota #5603: 2CH USB DC5V/7-32 V and Sonoff- Tasmota

vwtune, 2019-04-08:
Hi all,

I have a Sonoff board which i can't get it to be flashed. Any help would be nice.
I already flashed lots off Sonoffs such as a S20, basic, Th16, 4 channel Pro etc... But this one kills me...
[IMG]http://i64.tinypic.com/2m2eyxc.jpg[/IMG]
I think i get it in flash modes to push the two buttons nearest the relays at te same time when powerd it up with usb, then release the two buttons, then push the third button, the both leds will stay on, so i think it is in flash modus. But then i tried several places with the tx en rx, but always get the: error: espcomm_upload_mem failed.

Anyone already flashed this board succesfully?

blakadder, 2019-04-08:
Are you sure its an Itead Sonoff board?
You can always short GPIO0 to GND directly from the chip as mentioned in https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation

vwtune, 2019-04-08:
> Are you sure its an Itead Sonoff board?
> You can always short GPIO0 to GND directly from the chip as mentioned in https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation

Honestly i don't know, i just bought it as a Sonoff: https://nl.aliexpress.com/item/Sonoff-Smart-Wifi-Schakelaar-Universele-Module-2CH-USB-DC5V-7-32-v-Draadloze-Schakelaar-Timer-Telefoon/32950334093.html?spm=a2g0z.search0604.3.9.163cb35bzZ3t9e&ws_ab_test=searchweb0_0%2Csearchweb201602_5_10065_10068_10890_319_10546_317_10548_10696_10084_453_454_10083_10618_10304_10307_10820_537_536_10843_10059_10884_10887_321_322_10103%2Csearchweb201603_35%2CppcSwitch_0&algo_pvid=78029046-65ff-4427-97e2-5f79f8a0ef80&algo_expid=78029046-65ff-4427-97e2-5f79f8a0ef80-1

Where an i find GPIO0 and GND on the board?

blakadder, 2019-04-08:
https://www.google.com/search?safe=strict&q=psf-b04+pinout&tbm=isch&source=univ&sa=X&ved=2ahUKEwimnaHQ7sDhAhWEmIsKHYhCDTUQsAR6BAgHEAE&biw=1920&bih=979

vwtune, 2019-04-08:
> https://www.google.com/search?safe=strict&q=psf-b04+pinout&tbm=isch&source=univ&sa=X&ved=2ahUKEwimnaHQ7sDhAhWEmIsKHYhCDTUQsAR6BAgHEAE&biw=1920&bih=979
> 
> edit: even better, we have support for something similar https://github.com/arendst/Sonoff-Tasmota/wiki/Geekcreit-Sonoff-2-Channel-Relay-(AC-85V-250V)

Lovely! I managed to flash it :-)

blakadder, 2019-04-08:
Resolved!

closed 2019-04-08 by blakadder