arendst/Tasmota #5605: Shelly 2 power metering not working

BlackAlpha700, 2019-04-08:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Power metering does not work. All values show "0". Switch and relays are working fine.
I've configured tasmota as device type "Shelly 2 (47)". Is there anything else I have to configure to make it working?

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Device used (i.e. Sonoff Basic) : Shelly 2
- [x] Tasmota binary firmware version number used : 6.5.0 / (pre-compiled)
- [x] Provide the output of command ``status 0`` :
```
20:32:36 MQT: stat/shelly2_86/STATUS = {"Status":{"Module":47,"FriendlyName":["Shelly2_86_1","Shelly2_86_2"],"Topic":"shelly2_86","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:32:36 MQT: stat/shelly2_86/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:08","StartupUTC":"2019-04-08T19:26:28","Sleep":50,"CfgHolder":4617,"BootCount":20,"SaveCount":148,"SaveAddress":"F8000"}}
20:32:36 MQT: stat/shelly2_86/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019.03.19 12:30:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:32:36 MQT: stat/shelly2_86/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"192.168.x.x","LogPort":514,"SSId":["xxxxxx",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["0000C009","280500000100000000000000000000000000","00000000"]}}
20:32:36 MQT: stat/shelly2_86/STATUS4 = {"StatusMEM":{"ProgramSize":508,"Free":492,"Heap":14,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000407","0FDAE394","000783A0","23B617CE","00003BC0"]}}
20:32:36 MQT: stat/shelly2_86/STATUS5 = {"StatusNET":{"Hostname":"shelly2_86-3414","IPAddress":"192.168.x.x","Gateway":"192.168.x.x","Subnetmask":"255.255.255.0","DNSServer":"192.168.x.x","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":4}}
20:32:36 MQT: stat/shelly2_86/STATUS6 = {"StatusMQT":{"MqttHost":"openhab.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A4D56","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:32:36 MQT: stat/shelly2_86/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 08 19:32:36 2019","Local":"Mon Apr 08 20:32:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:13","Sunset":"19:30"}}
20:32:36 MQT: stat/shelly2_86/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:32:36 MQT: stat/shelly2_86/STATUS10 = {"StatusSNS":{"Time":"2019-04-08T20:32:36","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2019-04-05T17:08:26","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
20:32:36 MQT: stat/shelly2_86/STATUS11 = {"StatusSTS":{"Time":"2019-04-08T20:32:36","Uptime":"0T00:06:08","Vcc":3.109,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":3,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```

arendst, 2019-04-08:
Is it a Shelly 2 or a Shelly 2.5?

BlackAlpha700, 2019-04-08:
Shelly 2

arendst, 2019-04-08:
Then it should just work. What is your (shelly power) load?

BlackAlpha700, 2019-04-08:
An old lamp with 60W

arendst, 2019-04-08:
That should work just fine.

To get more information enable more logging with command ``weblog 4``. This should show metering information every second.

BlackAlpha700, 2019-04-08:
```
20:53:07 CMD: weblog 4
20:53:07 MQT: stat/shelly2_86/RESULT = {"WebLog":4}
20:53:07 DMP: FF FF FF
20:53:08 CFG: in Flash gespeichert am F8, zählen 156, Bytes 3584
20:53:08 DMP: FF FF FF FF
20:53:08 DMP: FF
20:53:09 DMP: FF
20:53:10 DMP: FF FF
20:53:11 DMP: FF FF
20:53:12 DMP: FF FF
20:53:13 DMP: FF
20:53:13 DMP: FF
20:53:13 DMP: FF
20:53:14 DMP: FF
20:53:15 DMP: FF FF
20:53:15 DMP: FF
20:53:16 MQT: tele/shelly2_86/STATE = {"Time":"2019-04-08T20:53:16","Uptime":"0T00:26:48","Vcc":3.109,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"xxxxx","Channel":3,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:06"}}
20:53:16 MQT: tele/shelly2_86/SENSOR = {"Time":"2019-04-08T20:53:16","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2019-04-05T17:08:26","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
20:53:16 DMP: FF
20:53:16 DMP: FF
20:53:16 WIF: Prüfe Verbindung...
20:53:16 WIF: verbunden
20:53:16 DMP: FF
20:53:16 DMP: FF
20:53:17 DMP: FF FF FF FF
20:53:18 DMP: FF
20:53:19 DMP: FF
20:53:20 DMP: FF
20:53:21 DMP: FF
20:53:21 DMP: FF
20:53:22 DMP: FF
20:53:23 DMP: FF
20:53:24 DMP: FF
20:53:24 DMP: FF
20:53:25 DMP: FF
20:53:25 DMP: FF
20:53:25 DMP: FF FF FF
20:53:26 DMP: FF
20:53:26 MQT: tele/shelly2_86/STATE = {"Time":"2019-04-08T20:53:26","Uptime":"0T00:26:58","Vcc":3.115,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"xxxxx","Channel":3,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:06"}}
20:53:26 MQT: tele/shelly2_86/SENSOR = {"Time":"2019-04-08T20:53:26","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2019-04-05T17:08:26","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
20:53:26 DMP: FF
20:53:26 DMP: FF
20:53:26 DMP: FF FF
20:53:26 DMP: FF
20:53:26 DMP: FF FF FF FF
20:53:27 DMP: FF FF
20:53:28 DMP: FF
20:53:29 DMP: FF
20:53:29 DMP: FF
20:53:29 DMP: FF FF
20:53:30 DMP: FF
20:53:31 DMP: FF
20:53:31 DMP: FF
20:53:31 DMP: FF
20:53:32 DMP: FF FF FF FF
20:53:32 DMP: FF FF
20:53:32 DMP: FF FF
20:53:33 DMP: FF FF
20:53:33 DMP: FF
```

arendst, 2019-04-08:
That's no good. Should be like this:
```
22:08:38 CMD: weblog 4
22:08:38 SRC: WebConsole from 192.168.2.1
22:08:38 RSL: Group 0, Index 1, Command WEBLOG, Data 4
22:08:38 MQT: stat/shelly2/RESULT = {"WebLog":4}
22:08:39 CFG: Saved to flash at F9, Count 1758, Bytes 3584
22:08:39 DMP: 06 19 E9 00 00 00 FA 08 E2 00 00 00 0E 00 00 00 0C 02 00 00 34 37 06 C3 3C
22:08:40 DMP: 06 19 E9 00 00 00 FC 08 E3 00 00 00 0C 00 00 00 0C 02 00 00 73 37 00 C3 76
22:08:41 DMP: 06 19 E9 00 00 00 FB 08 E0 00 00 00 0C 00 00 00 0C 02 00 00 B7 36 C2 C2 76
```

BlackAlpha700, 2019-04-09:
Does it make sense to try an older Tasmota - 6.41 for example?

arendst, 2019-04-09:
No. There are no changes that could be related to this behaviour.

What you could try is:
- Go to the web configure template and select Shelly 2 in Based on field:
[image]
- Save it.
- Then go to web configure module and select module 0 being the just saved template.
- After the reboot open web configure template again and play with the MCP93F5 GPIO's to find if the Shelly manufacturer changed the hardware regarding the energy monitoring chip. This is a tough task but maybe changing GPIO15 to none fixes things.

BlackAlpha700, 2019-04-09:
I've tried a lot, but nothing works.

This is the back of my Shelly 2 (Rev. E) - maybe the photo can help to find anything different.

[image]

arendst, 2019-04-09:
Rev.E is the same as I have.

You might want to read this https://github.com/arendst/Sonoff-Tasmota/wiki/Shelly-1-and-2 although it doesn't help in solving your metering problem.

Did the device work ok with the original firmware?

blakadder, 2019-04-10:
Have you done the classic erase flash completely and then flash a precompiled binary?

ascillato2, 2019-04-11:
Hi,

Any news on this?

BlackAlpha700, 2019-04-13:
> Did the device work ok with the original firmware?

I'm using some Sonoff devices with tasmota already and everything works fine. So I haven't thougt about testing the original firmware before flashing Tasmota. :-(

> Have you done the classic erase flash completely and then flash a precompiled binary?

No, but I'll try it...

fernandopereiranewby, 2019-04-14:
I have the exact same behavior with my shelly2, after upgrade from 6.4.1 to 6.5.0.

ascillato2, 2019-04-14:
Hi @BlackAlpha700 

Any news on this? Have you tried erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins?

ascillato2, 2019-04-18:
Closing this issue as there isn't feedback. Please, when you have time, tell us if erasing and flashing again solve your issue. Thanks

closed 2019-04-18 by ascillato2

gasparirob, 2019-04-25:
Same issue here, Shelly 2 working from couple of months but never metering power. It doesn't really matter to me but... it's not working.

PS: weblog 4 doesn't show powering information (no DMP)