arendst/Tasmota #5607: MQTT full topic field has parameters reversed

flavin21222, 2019-04-08:
### BUG DESCRIPTION
After OTA update to version 6.5.0 from version 6.4.1, the MQTT configuration full topic fields were reversed.  I.e., "%prefix%/%topic%/" became "%topic%/%prefix/".  For example, a message "stat/EpickaSwitch1/STATUS" would become "EpickaSwitch1/stat/STATUS".

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Happened with multiple devices: Sonoff Basic, Sonoff SV, Epicka smart switch flashed with Tasmota, HoveBeatty smart outlet flashed with Tasmota, Sonoff iFan02.
- [X] Tasmota binary firmware version number used : 6.5.0 / pre-compiled
- [X] Development IDE - Compiler / Upload tools used : NA - this was an OTA update flash
- [X] Provide the output of command ``status 0`` :  NOTE: This is output AFTER the problem was fixed by manually reversing the MQTT Full Topic fields on the Configure MQTT page.
```
22:26:59 MQT: stat/EpickaSwitch1/STATUS = {"Status":{"Module":18,"FriendlyName":["EpickaSwitch1"],"Topic":"EpickaSwitch1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T16:55:21","StartupUTC":"2019-04-08T04:31:38","Sleep":50,"CfgHolder":4617,"BootCount":19,"SaveCount":202,"SaveAddress":"F7000"}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ENIAC",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","280500000100000000000000000000000000","00000220"]}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS5 = {"StatusNET":{"Hostname":"EpickaSwitch1-2682","IPAddress":"REDACTED","Gateway":"REDACTED","Subnetmask":"255.255.255.0","DNSServer":"REDACTED","Mac":"DC:4F:22:84:CA:7A","Webserver":2,"WifiConfig":4}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS6 = {"StatusMQT":{"MqttHost":"REDACTED","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_84CA7A","MqttUser":"myMQTTuser","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 08 21:26:59 2019","Local":"Mon Apr 08 22:26:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:13","Sunset":"19:30"}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS10 = {"StatusSNS":{"Time":"2019-04-08T22:26:59"}}
22:26:59 MQT: stat/EpickaSwitch1/STATUS11 = {"StatusSTS":{"Time":"2019-04-08T22:26:59","Uptime":"0T16:55:21","Vcc":3.192,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"CC:2D:21:0D:59:59","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Provide the output of console when you experience your issue if apply : NA
```
### TO REPRODUCE
_Steps to reproduce the behavior:_Six (6) devices were OTA flashed with the update.  The %prefix% and %topic% fields in the MQTT Full Topic were reversed in each case.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
See attached view of Configure MQTT page with the MQTT Full Topic fields reversed.

### ADDITIONAL CONTEXT
Problem can be fixed by manually reversing the %prefix% and %topic% fields on the Configure MQTT page, MQTT Full Topic text box.

**(Please, remember to close the issue when the problem has been addressed)**
![MQTT_configuration_page](https://user-images.githubusercontent.com/37983195/55758796-cceec280-5a25-11e9-93d3-37b4c0399a78.png)

ascillato2, 2019-04-09:
Hi,

If you use the Home Assistant Auto discovery, it will change your fulltopic to that. If you don't want to use it, go to the console and type:

SetOption19 0
SetOption59 0
and correct your fulltopic to the one you need.

More information for this can be found in the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#setoption-overview and https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-09 by ascillato2

Fenix999, 2019-04-09:
Hi,

On this subject, this also cougt me by surprise but reading the documentation about home assistant I should be able to have a custom fulltopic of tasmota/%topic%/%prefix%/ but every time I try to have it it goes back to %topic%/%prefix%/.
I did actually tried with the tasmota name but like home/basement/machine_room/hot_water_control. 

Is this behavior normal?

ascillato, 2019-04-09:
Yes, it is as designed. 

If you use the Home Assistant Auto discovery, it will change automatically your fulltopic to that. If you don't want to use it, go to the console and type:

SetOption19 0
SetOption59 0
and correct your fulltopic to the one you need.