arendst/Tasmota #5608: Adding a new Driver

nosaj66au, 2019-04-09:
Hi All,
   Looking for some guidance around adding a new driver (version 6.5.0.2) . I have worked through the current driver list and think I have a reasonable handle on the way it works. However I stumbled on what I assumed would be the easy bit and its left me scratching my head.
    In the sonoff_template.h I have added  a new entry to UserSelectablePins "GPIO_RF433TX" just before the GPIO_SENSOR_END. I also added it to the bottom of the nice list kGpioNiceList as stated. This indeed creates a new list in the Web drop down at the end of the original list, however the name is not allocated, and in my case is displayed as "(154)". Assuming I need to add an entry in kSensorNames. I added my D_SENSOR_RF433_TX  "|" entry follow the pattern of the other entries. This throws a compile error "sonoff_template.h:245:3: error: expected ',' or ';' before 'D_SENSOR_RF433_TX'".  Not sure what I'm doing wrong or even if Im making the correct edits.
One thing that does confuses me is the names in the web interface drop down which do not  match the kGpioNiceList,  As an example the "ALux IrRvc" (147) list entry. "ALux_IrRvc" or "Alux IrRvc" or even "Alux" doesn't arrpear in the code anywhere???? 

Please help Jason

arendst, 2019-04-09:
kSensorNames are references to text in the selected language file. If you search for D_SENSOR_ARIRFRVC you'll find an entry in every language file.

To solve your issue you will have to add a text for D_SENSOR_RF433_TX in each language file.

nosaj66au, 2019-04-09:
Thanks @arendst endst as soon as I sore your comment I understood my mistake.The languages files are in a separate folder, so not in my search path.

nosaj66au, 2019-04-09:
Hi Again, 
   Can you also help with this error
 I added my D_SENSOR_RF433_TX "|" entry following the pattern of the other entries to the kSensorNames which is referenced in the language file. However it is still throws a compile error "sonoff_template.h:245:3: error: expected ',' or ';' before 'D_SENSOR_RF433_TX'". Keep in mind I have only edited the en-GB.h.

nosaj66au, 2019-04-09:
Sorry Looks like I had a typo.

closed 2019-04-09 by nosaj66au