arendst/Tasmota #5609: Simple EVSE-WB / EVSE-DIN with Modbus RS485

roesl, 2019-04-09:
Hi,
I want to control my EVSE-DIN which has Modbus protocol with Sonoff Tasmota ( MQTT) to set charging current or read the status of the device, so I could also use several Inputs or outputs of the tasmota device too.
Because there are also modules for IP-Symcon and other Smarthome applications available it would be easy to charge vehicles with solar power.

Jason2866, 2019-04-09:
What is your question?

roesl, 2019-04-09:
I want to control the evse-din with modbus rs485 so I need a device for tasmota to do this

Jason2866, 2019-04-09:
Modbus Protocol isnt supported from Tasmota. 
You can connect Modbus hardware as described here
https://github.com/arendst/Sonoff-Tasmota/issues/3694#issuecomment-424743666

roesl, 2019-04-09:
there is the SDM630 Energy meter ( this has Modbus protocol)  as device in tasmota so I think it would be also possible to add the device EVSE-DIN

Jason2866, 2019-04-09:
There is a specific driver made for this device. This would be needed for the device you want too.
There is no general solution...

roesl, 2019-04-09:
is anybody able to make a driver for the EVSE - I think there are many people interested because they  will drive Electric vehicles in the future

ascillato2, 2019-04-10:
@roesl 

Thanks for sharing your ideas. It is indeed very interesting. I personally like it a lot, but would be very difficult to develop a driver for it without having the actual hardware and obviously without the car for testing.

I'm adding the label feature request (hold over) for future reference and work.

Also, I'm posting here some links with extra information about this.

EVSE-WB =  Electric Vehicle Supply Equipment for Wallbox
EVSE-DIN = Electric Vehicle Supply Equipment for DIN

EVSE is for European Standard

Compatibility:
• Tesla Model S
• Nissan Leaf
• Mitsubishi iMiev (Peugeon iOn, Citroen cZero)
• Opel Ampera
• eGolf
• Mitsubishi Outlander PHEV
• Citroen Berlingo Electrique
• and others

Board explanation: http://openwb.de/manuals/evse-wb-din_20180611.pdf
Protocol: https://www.openevse.com/
To Buy: https://www.off-grid-systems.de/Simple-EVSE-WB-Wallbox-Bausatz

ascillato2, 2019-04-10:
For now you can use as a workaround the serial-mqtt bridge feature of Tasmota as @Jason2866 has explained.

Thanks

closed 2019-04-10 by ascillato2