arendst/Tasmota #5613: Sonoff Device (RF Bridge in my case) as permament Wifi Access Point?

haegar33, 2019-04-10:
I want to use an RF Bridge in my vacation home **without any Wifi Access Point**. Can I use my tasmota-flashed Bridge (V6.5) as a permanent Wifi Access Point to embed and control other Wifi Devices (i..e. EPS8266 based sensors etc..).
Of course the main function would be to operate as RF controller but including additional Wifi sensors would be interessting..?

Thanks, Haegar

blakadder, 2019-04-10:
You're expecting way too much from an ESP8266 chip. Do yourself a favor and buy a cheap wireless router instead.

closed 2019-04-10 by blakadder

haegar33, 2019-04-10:
So the answer is "no"? I understood that the AP feature is only for setup purposes but as it is present can it not act permanently as AP?

meingraham, 2019-04-10:
The answer is probably yes but not worth the effort... and then the long term effort of maintaining your customized code.

Take the advice from @blakadder, for example: https://www.amazon.com/TP-Link-Wireless-Portable-Travel-Router/dp/B00TQEX8BO