arendst/Tasmota #5615: Module parameters don't show gpios options

pablozg, 2019-04-10:
<GUIDE>

### BUG DESCRIPTION

In Configure Module window, the gpios parameters aren't showed on a sonoff pow r2

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : sonoff pow r2
- [X] Tasmota binary firmware version number used : 6.5.0.4 and 6.5.0.6 / self-compiled
- [X] Development IDE - Compiler / Upload tools used : Visualcode / platformio
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

21:10:48 CMD: status 0
21:10:48 MQT: monitorCasa/stat/STATUS = {"Status":{"Module":43,"FriendlyName":["monitorCasa"],"Topic":"monitorCasa","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:10:48 MQT: monitorCasa/stat/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T08:29:09","StartupUTC":"2019-04-10T10:41:39","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":65,"SaveAddress":"3FA000"}}
21:10:48 MQT: monitorCasa/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0.4(sonoff)","BuildDateTime":"2019-04-04T10:47:43","Boot":7,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
21:10:48 MQT: monitorCasa/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Isabella_T","Isabella_H"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["002A8009","28B400000100000000000000000000000000","00000200"]}}
21:10:48 MQT: monitorCasa/stat/STATUS4 = {"StatusMEM":{"ProgramSize":547,"Free":3528,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE394","040783A1","23B617CE","00003BC0"]}}
21:10:48 MQT: monitorCasa/stat/STATUS5 = {"StatusNET":{"Hostname":"monitorCasa-6274","IPAddress":"10.10.0.115","Gateway":"10.10.0.254","Subnetmask":"255.255.255.0","DNSServer":"10.10.0.254","Mac":"80:7D:3A:32:B8:82","Webserver":2,"WifiConfig":2}}
21:10:48 MQT: monitorCasa/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.0.110","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32B882","MqttUser":"htpc","MqttCount":12,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:10:48 MQT: monitorCasa/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 10 19:10:48 2019","Local":"Wed Apr 10 21:10:48 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:09","Sunset":"20:33"}}
21:10:48 MQT: monitorCasa/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:10:48 MQT: monitorCasa/stat/STATUS10 = {"StatusSNS":{"Time":"2019-04-10T21:10:48","ENERGY":{"TotalStartTime":"2019-02-23T11:35:38","Total":93.689,"Yesterday":12.864,"Today":11.811,"Power":472,"ApparentPower":484,"ReactivePower":109,"Factor":0.97,"Voltage":223,"Current":2.174}}}
21:10:48 MQT: monitorCasa/stat/STATUS11 = {"StatusSTS":{"Time":"2019-04-10T21:10:48","Uptime":"0T08:29:09","Vcc":3.462,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"POWER":"ON","Wifi":{"AP":1,"SSId":"Isabella_T","BSSId":"D4:6E:0E:D0:43:EC","Channel":6,"RSSI":36,"LinkCount":8,"Downtime":"0T00:00:24"}}}

```

### SCREENSHOTS
[image]

```

curzon01, 2019-04-10:
Duplicate #4732 and similary.

It's dangerous to connect something to a pow so there is no gpio definition enabled.

pablozg, 2019-04-10:
Ok, I'm sorry, I don't find it, thanks,

closed 2019-04-10 by pablozg