arendst/Tasmota #5616: NEO Coolcam NAS-WR02 template

cypeks, 2019-04-10:
**Have you look for this feature in other issues and in the wiki?**
yes
**Is your feature request related to a problem? Please describe.**
yes

**Describe the solution you'd like**
I give a ready template for the device NEO Coolcam NAS-WR02 (WiFi Socket 3680W 16A Power Energy Monitoring) :
{"NAME":"NAS-WR02","GPIO":[17,0,0,0,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":45}

**Describe alternatives you've considered**
{"NAME":"NAS-WR02","GPIO":[17,0,0,0,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":45}

**Additional context**
[image]

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-11:
Please provide link where to buy and link to picture(s)
Thx

blakadder, 2019-04-11:
Is it this one https://blakadder.github.io/templates/neocoolcam_NAS-WR02.html? If yes, close the issue please

ascillato2, 2019-04-11:
Closing this issue as the device is already in the template database. Anyway, thanks for sharing it. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-11 by ascillato2

LouiS22, 2019-04-14:
> Closing this issue as the device is already in the template database. Anyway, thanks for sharing it.
>
> ### Support Information [...]

The device in the database isn't measuring power consumption or at least it's not present.

blakadder, 2019-04-14:
Are you using sonoff.bin?

cypeks, 2019-04-18:
I'm using sonoff-PL.bin ver. 6.5.0