arendst/Tasmota #5617: How to ignore double press for X time

MikeVRT, 2019-04-10:
Hello,
I have Sonoff RF with Tasmota FW.
Sonoff RF controlled by push button with 433MHz transmitter.
SetOption13 set to 1 for immediate relay reaction.
Sometimes, users can push button with to long hold, and relay makes double switch like on->off->on. or off->on->off.
Is in possible configure something for immediate relay reaction and than ignoring for 500-1000ms any presses (in my case - that push button still hold and Sonoff RF's receiver interpretating it as next button press).
p.s. if we hold push button, transmitter sends press to receiver and relay makes not stop "...on-off-on-off-on-off..".

Jason2866, 2019-04-11:
`SetOption1 1`  and `SetOption11 1`
See https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices

ascillato2, 2019-04-11:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-11 by ascillato2

MikeVRT, 2019-04-11:
> `SetOption1 1` and `SetOption11 1`
> See https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices

Sorry, and how long will be "ignoring period" after accepted push?

MikeVRT, 2019-04-11:
> `SetOption1 1` and `SetOption11 1`
> See https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices
I tried this settings. They don't help unfortunately, long press still equal 2 presses (on-off-on for example)

ascillato, 2019-04-11:
Please, check the example of the wiki and adapt it to the behaviour you desire https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

MikeVRT, 2019-04-11:
> Please, check the example of the wiki and adapt it to the behaviour you desire https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

In link "only if defining your GPIO as button, ... assuming GPIO0 is configured as Button1", but Sonoff RF have no GPIO0 at all. What GPIO to change instead?: [Screenshot](url=https://ibb.co/09SvPLx)

MikeVRT, 2019-04-12:
Any ideas?