arendst/Tasmota #5620: Dimmer always turn on to high intensity

SylvainGa, 2019-04-11:
On my KS-7011 dimmer, whenever it is toggled on, although the LED will display its last known intensity setting and the Console output will also display the last know intensity setting, the intensity is always set to 50%.

Example of the dimmer can be seen here: [https://www.amazon.ca/dp/B07JNRFBG3/ref=pd_luc_rh_crh_rh_sbs_02_01_t_img_lh?_encoding=UTF8&psc=1](url)

Jason2866, 2019-04-11:
Issue Template is missing. Without no help can be provided

ascillato2, 2019-04-11:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

SylvainGa, 2019-04-12:
Sorry, didn't know (or read) that. Actually, my first assumption was wrong. It's not 50% but changes as the device reboots. Also, every power on / power off combo drops the dimmer value. Debugging as I type this (code is compiling). Might just want to close this bug request.

closed 2019-04-12 by ascillato2