arendst/Tasmota #5623: Reboots in DIGOO SP202 after choosing module type.

machadof1, 2019-04-11:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
In 3 of my 5 DIGO SP202, after choosing the DIGOO SP202 module type, when I click save, the SP202 does a series of reboots and then looses all configuration, I have to select the sonoff wifi network again and put in my wifi information. If I try to flash the firmware again with the sonoff. bin I get the error "
Upload buffer miscompare“,  I can however flash it again if I first use the minimal firmware. 
Once I choose the model type DIGOO SP202 again  the cycle repeats again. 
The pcb of these 3 DIGOO SP202 are a little different from the other 2.
[image]
[image]

## REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

22:11:59 CMD: STATUS 0
22:11:59 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:11:59 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T06:05:21","StartupUTC":"2019-04-11T15:06:38","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":24,"SaveAddress":"F9000"}}
22:11:59 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:11:59 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxx","xxxxxxxx"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
22:11:59 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
22:11:59 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"DSwitch 2","IPAddress":"xxx. xxx. xxx. xxx","Gateway":"xxx. xxx. xxx. xxx","Subnetmask":"255.255.255.0","DNSServer":"xxx. xxx. xxx. xxx","Mac":"80:7D:3A:7C:99:A4","Webserver":2,"WifiConfig":4}}
22:11:59 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7C99A4","xxxxxxxxx":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:11:59 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 11 21:11:59 2019","Local":"Thu Apr 11 22:11:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:07","Sunset":"19:35"}}
22:11:59 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-04-11T22:11:59"}}
22:11:59 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-04-
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Choose DIGOO SP202 in module type and click save

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Save the configuration and reboot once and keep all configuration. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

machadof1, 2019-04-11:
I used tuya-convert to upload the precompiled 6.5.0 firmware version

ascillato, 2019-04-11:
Hi,

Please, could you try to do update to latest Tasmota of dev branch http://thehackbox.org/tasmota/sonoff.bin

Also, to clean any incorrect data left on the flash, you can try to do a `reset 5` in the Tasmota console.

machadof1, 2019-04-12:
No luck, even with the new firmware, as soon as I choose the DIGOO SP202 as the model type and click save, I start to hear the relay clicking, more than 10 times, when they stop clicking all configuration is gone.

Jason2866, 2019-04-12:
Please do this configuration manually in module generic:

```
     GPIO_KEY1,                 // GPIO00 Button1
     GPIO_HJL_CF,             // GPIO04 BL0937 or HJL-01 CF power
     GPIO_NRG_CF1,         // GPIO05 BL0937 or HJL-01 CF1 current / voltage
     GPIO_NRG_SEL_INV,  // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
     GPIO_LED1,               // GPIO13 Blue Leds - Link Status
     GPIO_REL2,               // GPIO14 Relay2 (0 = Off, 1 = On) and Red Led
     GPIO_REL1,               // GPIO15 Relay1 (0 = Off, 1 = On) and Red Led
     GPIO_KEY2_NP,         // GPIO16 Button2, externally pulled up
```
**Try it first with NOT configuring GPIO16.** Does it work without reset now?
If yes configure GPIO16 as suggested. Is reseting back again?

Jason2866, 2019-04-12:
@arendst  I tested the Module 63 (Digoo-SP202) config on a wemos mini and had this behaviour too.
It is caused from Setting GPIO 16 to Button2. Deleting this assignment to GPIO 16 and no more resetting.
It is a bug.

machadof1, 2019-04-12:
Just to clarify this problem, it does not happen to all DIGOO SP202, I have 2 with a different pcb that are ok, the other 3 have this problem, the ones that do have this problem have V1.3 written on the pcb, the others have V2.0. 
I will try to discover the pin assignments of the v1.3 pcb.

Jason2866, 2019-04-12:
Please try setting GPIO 16 to none (0) for the failing devices

machadof1, 2019-04-12:
I will try that later today.

Jason2866, 2019-04-12:
GPIO 16 isnt a fully useable GPIO has restrictions
https://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations

machadof1, 2019-04-12:
I managed to get it to work correctly, the pin assignment on this pcb is different, the default assignment of this module is:
GPIO0	Button1(17)
GPIO1	None
GPIO2	None
GPIO3	None
GPIO4	BL0937 CF(134)
GPIO5	HLWBL CF1(132)
GPIO9	None
GPIO10	None
GPIO12	HLWBL SEL1(131)
GPIO13	Led1(52)
GPIO14	Relay2(22)
GPIO15	Relay1(21)
GPIO16	Button2n(91)

For this pcb I got it to work using:
GPIO0	Led1(52)
GPIO1	None
GPIO2	None
GPIO3	HLWBL SELi(131)
GPIO4	Button2n(91)
GPIO5	BL0937 CF(134)
GPIO9	None
GPIO10	None
GPIO12	Relay2(22)
GPIO13	Button1(17)
GPIO14	HLWBL CF1(132)
GPIO15	Relay1(21)
GPIO16	None
[image]

Jason2866, 2019-04-12:
Could you make a template for this? 
And provide the info here https://blakadder.github.io/templates/

machadof1, 2019-04-12:
Done, the template is for model with no SN below the "MADE IN CHINA"
[image]

machadof1, 2019-04-12:
not this one
[image]

Jason2866, 2019-04-13:
Fixed https://blakadder.github.io/templates/digoo_NX-SP202.html
Thx for your finding and providing.
Please close issue

closed 2019-04-13 by machadof1

haselchen, 2019-04-16:
I got the same error. Flashing with Tuya Convert. Choose Model Digoo SP202. After one oder two reboots the device lost connection to my WIFI. No chance to connect. So i open the device. I remove the screws , but the board is fixed so hard. How will i get the board out of the device for flashing?

machadof1, 2019-04-16:
No need to flash again, after the module loses the configuration, the sonoff ssid is up again. Just connect to it again and set your WiFi  information

haselchen, 2019-04-16:
There is no AP , nothing. The Device is dead. Not even electricity is working. Inside , I can see the blue Leds if I push the buttons left or right.  No more no less.

machadof1, 2019-04-16:
That is not the behavior I got, in my case every time I lost the configuration I always got the sonoff ssid back. Other than trying to flash directly with the tx/rx pins I don't know what you can do.

netpok, 2019-04-17:
@haselchen It's not easy to disassemble the device but definetely not impossible, the method I used when I made the original pinout:

1. Pull the ground bar on the plug by the hole so you can insert a spudger under it
2. While prying it with the spudger you need to desolder it between the two relays the joint is very close to the red-black wire
3. Unscrew the pcb
4. Take a solid core wire (I used 1.5mm² - 16AWG) and hook the two ends under the two socket assemblies
5. While pulling the pcb with the wire desolder the two main pins, apply modest force and switch sides often to avoid step 6
6. If you applied to much force in the previous step and removed the via, make sure to create a replacement with multi-core wire or it can be a fire hazard and/or destroy the device (I did not in the first try and voltage regulator did not get connected to ground because its connected to the underside of the pcb and the esp and the bl0937 blow-up)

step 1 and 2:
[image]

step 2, 3, 4 and 5:
[image]

haselchen, 2019-04-17:
@ netpok
Thank you for the detailed instructions. It will certainly help others. Unfortunately, technically I am not talented at all. Unscrew screws, no problem. I left the easiest way and ordered 2 more. 
You are doing a fantastic job. Thanks again.

nikfo953, 2020-11-20:
Hi! Just found this as i was trying to Convert my Device. Tuya-convert doesnt work for me, so left with serial option. Having trouble opening the Device ”lid” to get to the PCBs. Any hints there? Saw the pictures above but it didnt get me inside.
Thnx!