arendst/Tasmota #5626: My Tuya plug restarts randomly

vaidyasr, 2019-04-12:
I have successfully flashed the Tasmota on my Tuya smart plug. But the plug is switched off randomly. I have checked the settings and PowerOnState is 3 and ButtonRetain is off and SwitchRetain is off only. I can't find any useful info in console. How do I fix it?. 
The plug is [here](https://www.aliexpress.com/item/US-Standard-Mini-Smart-Wifi-Socket-Remote-Control-Power-Strip-Timing-Switch-for-Smart-Home-Automation/32969667372.html?spm=a2g0s.9042311.0.0.1b134c4dmKc1iK).

Jason2866, 2019-04-12:
Issue template is missing. Status 0....

ascillato2, 2019-04-12:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-12 by ascillato2