arendst/Tasmota #5630: My Tuya plug restarts randomly

vaidyasr, 2019-04-12:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Tasmota flashed Tuya plug restarts randomly.  I have checked the settings and PowerOnState is 3 and ButtonRetain is off and SwitchRetain is off only.
  

### REQUESTED INFORMATION

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : Tuya SP
- [ ] Tasmota binary firmware version number used : 6.5.0
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
```
20:30:56 CMD: status 0
20:30:56 MQT: stat/tuya1/STATUS = {"Status":{"Module":0,"FriendlyName":["Tuya"],"Topic":"tuya1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:30:56 MQT: stat/tuya1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:05:21","StartupUTC":"2019-04-12T14:55:35","Sleep":50,"CfgHolder":4617,"BootCount":60,"SaveCount":176,"SaveAddress":"FB000"}}
20:30:56 MQT: stat/tuya1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-basic)","BuildDateTime":"2019-03-19T12:25:14","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:30:56 MQT: stat/tuya1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Anuradha Sriram",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
20:30:56 MQT: stat/tuya1/STATUS4 = {"StatusMEM":{"ProgramSize":432,"Free":568,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F082380","000483A0","21021706","000001C0"]}}
20:30:56 MQT: stat/tuya1/STATUS5 = {"StatusNET":{"Hostname":"tuya1-7260","IPAddress":"192.168.1.137","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.108","Mac":"C4:4F:33:8E:BC:5C","Webserver":2,"WifiConfig":4}}
20:30:56 MQT: stat/tuya1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.108","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8EBC5C","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:30:57 MQT: stat/tuya1/STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 12 15:00:56 2019","Local":"Fri Apr 12 20:30:56 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+05:30","Sunrise":"10:35","Sunset":"00:06"}}
20:30:57 MQT: stat/tuya1/STATUS10 = {"StatusSNS":{"Time":"2019-04-12T20:30:57"}}
20:30:57 MQT: stat/tuya1/STATUS11 = {"StatusSTS":{"Time":"2019-04-12T20:30:57","Uptime":"0T00:05:22","Vcc":3.186,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Anuradha Sriram","BSSId":"E8:DE:27:FB:F0:5A","Channel":11,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:08"}}}
20:31:01 MQT: tele/tuya1/STATE = {"Time":"2019-04-12T20:31:01","Uptime":"0T00:05:26","Vcc":3.186,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Anuradha Sriram","BSSId":"E8:DE:27:FB:F0:5A","Channel":11,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:08"}}
```

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-12:
DONT use sonoff-basic version. Use sonoff.bin.
Read wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisites#needed-software

vaidyasr, 2019-04-13:
My log output after flashing sonoff.bin
07:48:48 MQT: stat/tuya1/STATUS = {"Status":{"Module":0,"FriendlyName":["Tuya"],"Topic":"tuya1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:48:48 MQT: stat/tuya1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:36","StartupUTC":"2019-04-13T02:18:12","Sleep":50,"CfgHolder":4617,"BootCount":61,"SaveCount":180,"SaveAddress":"F9000"}}
07:48:48 MQT: stat/tuya1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:48:48 MQT: stat/tuya1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Anuradha Sriram",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
07:48:48 MQT: stat/tuya1/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
07:48:48 MQT: stat/tuya1/STATUS5 = {"StatusNET":{"Hostname":"tuya1-7260","IPAddress":"192.168.1.137","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.108","Mac":"C4:4F:33:8E:BC:5C","Webserver":2,"WifiConfig":4}}
07:48:48 MQT: stat/tuya1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.108","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8EBC5C","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:48:48 MQT: stat/tuya1/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 13 02:18:48 2019","Local":"Sat Apr 13 07:48:48 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+05:30","Sunrise":"10:33","Sunset":"00:08"}}
07:48:48 MQT: stat/tuya1/STATUS10 = {"StatusSNS":{"Time":"2019-04-13T07:48:48"}}
07:48:48 MQT: stat/tuya1/STATUS11 = {"StatusSTS":{"Time":"2019-04-13T07:48:48","Uptime":"0T00:00:36","Vcc":3.187,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Anuradha Sriram","BSSId":"E8:DE:27:FB:F0:5A","Channel":11,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:04"}}}
07:48:59 WIF: Checking connection...
07:48:59 WIF: Connected
07:49:19 WIF: Checking connection...
07:49:19 WIF: Connected
07:49:39 WIF: Checking connection...
07:49:39 WIF: Connected
07:49:59 WIF: Checking connection...
07:49:59 WIF: Connected
```

vaidyasr, 2019-04-13:
Below log in console too when it reboots
~~~
00:00:00 CFG: Loaded from flash at F5, Count 184
00:00:00 SRC: Restart
00:00:00 Project sonoff Tuya Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Anuradha Sriram in mode 11N as tuya1-7260...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
~~~

ascillato2, 2019-04-13:
@vaidyasr 

Hi, 

Thank you very much for completing the template and providing all the information requested. Very appreciated. 

In your last log, it says `SRC: Restart` that means that Tasmota has restarted due to it received the command to restart. After your device randomly restarts, can you post the `status 0` in order to see the restart reason?

Can you also check if you have retained messages? You are using home assistant right? Well, there is an addon from the home assistant community to monitor your mqtt messages called MQTT SERVER AND WEBCLIENT. When you starts the webclient all the retained messages will pop up and you can see if there is something weird like sending the restart command to Tasmota.

ascillato2, 2019-04-13:
Another question, how have you flashed Tasmota? Using esptool.py? Have you erased all flash before flashing Tasmota? Or have you done a reset 5 to clean the flash at first usage?

vaidyasr, 2019-04-13:
> Another question, how have you flashed Tasmota? Using esptool.py? Have you erased all flash before flashing Tasmota? Or have you done a reset 5 to clean the flash at first usage?

I have used  tuya-convert to flash the Tasmota. I have followed their instructions to flash from original firmware to Tasmota.

vaidyasr, 2019-04-13:
Console log right after the restart
~~~
16:54:30 MQT: stat/tuya1/STATUS = {"Status":{"Module":0,"FriendlyName":["Tuya"],"Topic":"tuya1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:54:30 MQT: stat/tuya1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:00:40","StartupUTC":"2019-04-13T11:23:50","Sleep":50,"CfgHolder":4617,"BootCount":66,"SaveCount":192,"SaveAddress":"F5000"}}
16:54:30 MQT: stat/tuya1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:54:30 MQT: stat/tuya1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Anuradha Sriram",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
16:54:30 MQT: stat/tuya1/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
16:54:30 MQT: stat/tuya1/STATUS5 = {"StatusNET":{"Hostname":"tuya1-7260","IPAddress":"192.168.1.137","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.108","Mac":"C4:4F:33:8E:BC:5C","Webserver":2,"WifiConfig":4}}
16:54:30 MQT: stat/tuya1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.108","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8EBC5C","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:54:30 MQT: stat/tuya1/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 13 11:24:30 2019","Local":"Sat Apr 13 16:54:30 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+05:30","Sunrise":"10:33","Sunset":"00:08"}}
16:54:30 MQT: stat/tuya1/STATUS10 = {"StatusSNS":{"Time":"2019-04-13T16:54:30"}}
16:54:30 MQT: stat/tuya1/STATUS11 = {"StatusSTS":{"Time":"2019-04-13T16:54:30","Uptime":"0T00:00:40","Vcc":3.186,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Anuradha Sriram","BSSId":"E8:DE:27:FB:F0:5A","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}
~~~

vaidyasr, 2019-04-13:
Also, the plug restarts only if it is switched on. If it is off, I don't see any activity apart from regular messages.

ascillato, 2019-04-13:
Hi,

The restart reason is power on. It means that the esp8266 chip was left without power and was powered again. 

So, seems that the problem is just a faulty power regulator. Besides, the fact that this problem only occurs when the relay is on. So, sorry but seems to be a hardware defect.

vaidyasr, 2019-04-14:
Thank you. But if I make the config as Sonoff basic, the switch stays on without any issue. The only problem is the switch can't be switched-on via the physical switch.
On the console log, it says '00:00:00 WIF: Patch issue 2186' is that normal?

Jason2866, 2019-04-14:
> On the console log, it says '00:00:00 WIF: Patch issue 2186' is that normal

Yes, this is normal.

ascillato2, 2019-04-14:
So, if you choose sonoff basic, your button doesn't work for obvious reasons (different gpio mapping) but your device stays on without resetting right?

But if you choose your template, your button works but if the device is ON it will randomly resets?

ascillato2, 2019-04-15:
@vaidyasr 

Hi, 

Any news on this?

vaidyasr, 2019-04-16:
Strangely the plug stays on when I press the physical button on. But when switching on from web, the plug restarted and switched off.

ascillato, 2019-04-16:
Ok. And after the restart, what is the restart reason?

vaidyasr, 2019-04-16:
Why my Tuya plug is showing 'Checking connection' every 20 seconds, whereas my other sonoff's doesn't
~~~
10:40:48 WIF: Checking connection...
10:40:48 WIF: Connected
10:41:08 WIF: Checking connection...
10:41:08 WIF: Connected
10:41:28 WIF: Checking connection...
10:41:28 WIF: Connected
10:41:48 WIF: Checking connection...
10:41:48 WIF: Connected
10:42:08 WIF: Checking connection...
10:42:08 WIF: Connected
10:42:28 WIF: Checking connection...
10:42:28 WIF: Connected
10:42:48 WIF: Checking connection...
10:42:48 WIF: Connected
10:43:08 WIF: Checking connection...
10:43:08 WIF: Connected
10:43:28 WIF: Checking connection...
10:43:28 WIF: Connected
10:43:48 WIF: Checking connection...
10:43:48 WIF: Connected
10:44:10 WIF: Checking connection...
10:44:10 WIF: Connected
10:44:26 HTP: Main Menu
10:44:30 WIF: Checking connection...
10:44:30 WIF: Connected
10:44:47 HTP: Console
10:44:52 WIF: Checking connection...
~~~

stepic, 2019-04-17:
I have the same issue described above. My plug is an AWP07L and I used tuya-convert too. I tried reflashing via OTA and configuration reset without no luck.
My template is https://blakadder.github.io/templates/houzetek_AWP07L.html

ascillato, 2019-04-17:
@vaidyasr 

Ok. And after the restart, what is the restart reason?

vaidyasr, 2019-04-18:
@ascillato 
the reason is same in the above log. but it take a while to restart comparing to the switch-on via web.

ascillato2, 2019-04-18:
If the restart reason is power on after you click, I'm sorry but that is a power regulator failure. Your hardware has a defect.

ascillato2, 2019-04-18:
Being this issue a hardware problem, there isn't anything thay can be done from the software side.

closed 2019-04-19 by ascillato2

vaidyasr, 2019-05-20:
@ascillato2 I just unchecked 'ADC0 input' and the plug is stable without any restart for the past 4 days.

masseyy, 2019-07-30:
@vaidyasr I have been experiencing similar issues - to clarify my issues is only when i manually press the button the relay comes on then after 5..10..15..20 mins the relay goes. Controlling via home assistant or via tasmota using the IP address in a web browser, there is no issue works 100%.
I have taken your advise and turned off 'ADC0 input', hopefully this does the job!