arendst/Tasmota #5631: Publishing Counter1 value over a specific MQTT topic

HobbytronicsPK, 2019-04-12:
Hello,

I have set up a counter with button using rules with Nodemcu. Counter value increments when button is pressed (I have made changes in certain settings using commands). I have implemented it successfully. Here is my rule.

`Rule1 on Switch4#state=0 do Counter1 +1 endon`

Now I want to publish Counter1 value to a specific MQTT topic such as watchdog. Here is an example.

`Rule1 on Switch4#state=0 do Backlog Counter1 +1; Publish watchdog {"idx":1,"count":%Counter1%} endon`

It is just publishing `Counter1` on MQTT topic rather it's value. May someone please help?

ascillato, 2019-04-13:
Hi,

There is not any variable in Tasmota called %counterX%. For all available variables, please check the wiki at rules. 

Counter is meant for setting a GPIO as a pulse counter. Then its value is being sent every teleperiod automatically.

If you want just to count the times you press a switch, it should be better to use a VAR instead. You can use the command ADD to add 1 to a var, and you can publish a var using %var1%.

HobbytronicsPK, 2019-04-13:
Got it. It was the second solution I thought. But Counter value is being saved in Flash. Can we retrieve this value?

Jason2866, 2019-04-13:
No
Yes

ascillato2, 2019-04-14:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-14 by ascillato2

gbansen, 2019-09-01:
Hello,

When I try to publish a %var1% I don't get the value. See below.
For example: "publish garage_grant/cmnd/POWER8 %var1%"
I get: "MQT: garage_grant/cmnd/POWER8 = %var1%"

Below is a clip from my MQTT Explorer

[image]

I've tried different bracket around the variable but nothing seems to work.
I'm I doing something wrong. I really wished this worked.

ascillato, 2019-09-01:
%Var1% is only supported inside a rule. Is not valid to be used as a direct command or request

gbansen, 2019-09-02:
Thanks for the response. This is good to know as I was actual trying to test this for a rule that had failed. Your comment made me look closer at the rule I had made and I found an extra space in the line that I think wasn't allowing it to run properly. I still have a problem though. Below are the rules.

rule1 on switch8#state=1 do publish garage_grant/cmnd/POWER8 ON; mem1 ON endon on switch8#state=0 do publish garage_grant/cmnd/POWER8 OFF; mem1 OFF endon

rule2 on mqtt#connected do publish garage_grant/cmnd/POWER8 %mem1% endon 
This works good if the state changes from OFF to ON while disconnected from MQTT, but from ON to OFF for some reason It publishes an ON see below.

[image]

I'm sure I'm doing something wrong. Thanks again for your help.
Grant...

Jason2866, 2019-09-02:
Your rule syntax is wrong. No `;` allowed here. See https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook for examples.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-09-02:
Hi,

your rule:

```
rule1 on switch8#state=1 do publish garage_grant/cmnd/POWER8 ON; mem1 ON endon on switch8#state=0 do publish garage_grant/cmnd/POWER8 OFF; mem1 OFF endon
```

is wrong and should be:

```
rule1 on switch8#state=1 do backlog publish garage_grant/cmnd/POWER8 ON; mem1 ON endon on switch8#state=0 do backlog publish garage_grant/cmnd/POWER8 OFF; mem1 OFF endon
```

note the backlog command.

gbansen, 2019-09-02:
Thank you ascillato that fix the problem.
Sorry Jason2866 I have spent the last few days trying to figure this out by searching and I did look at the cookbook examples but nothing seem to fit my issue until I came across this post. Hopefully in time I will better understand the syntax. I defiantly will not post here again until less it is to report a bug. Thank you both for your patience.