arendst/Tasmota #5635: Teckin SP21 - No power measurements

SlayDictator, 2019-04-13:
Hi there,

I just got those:

https://www.amazon.de/dp/B07HHFKWJJ

They are labelled with Teckin SP21.

I flashed them with Tasmota 6.5.0 OTA without opening them using tuya-convert.
They are working fine with profile 45, Blitzwolf SHP.

However, I wasn't able to get the power measurements running, all values show 0.

I tried all the Teckin profiles from the database and profile 45 and 52 from the stock firmware. 
Is there any possibility to find out how to configure them, without cracking them open? They are glued together, so I would like to avoid opening them...

I hope there is anyone out there who knows how to configure this.

bleisoldat, 2019-04-13:
same Problem here

bleisoldat, 2019-04-13:
[image]
[image]

Jason2866, 2019-04-13:
In the link you provided is nowhere mentioned that devices supports power monitoring

ascillato2, 2019-04-14:
@SlayDictator 

Hi, 

Did you test the power monitoring before flashing Tasmota? 

There are several versions of this device and only very few of them have power monitoring. From the photos seems that it doesn't have that feature.

SlayDictator, 2019-04-15:
Hi,

I was sure that I saw the power monitoring feature in the product description. But @Jason2866 is right:  It isn't mentioned there.
I did not test them without Tasmota, I didn't want to create an account and without Tasmota they are useless for me, so I did not waste time before flashing them.

So I guess this issue can be solved - I need some new plugs ;p

closed 2019-04-15 by ascillato2

ei-ke, 2019-04-18:
@SlayDictator @Bohrfutter I also first got the SP21 and after cracking them open (beating on the sides with a rubber mallet worked well) I noticed that there is no power monitoring feature and additionally got the [SP22](https://www.amazon.de//dp/B07D5V139R/) which work.
Further have a look at [the "soft" way](https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html) to flash the firmware... No need to open them...

Grüße
Eike

seritos, 2020-02-10:
Hi, 

Is it possible to disassembly this device without broke it? 

Thanx

ei-ke, 2020-02-10:
@seritos nope, you have to break the bonding between the socket and the enclosure as I've described.

seritos, 2020-02-10:
> @seritos nope, you have to break the bonding between the socket and the enclosure as I've described.

Ok, Thnx!

ortegafernando, 2021-09-22:
> you have to break the bonding between the socket and the enclosure as I've described

HI @ei-ke , I have "opened" it, but how to you get to extract the pcb board? is it glued to bottom? hot gun? do I must to broke also the left side of the enclosure as one photo of this post? Thanks.

ortegafernando, 2021-09-22:
> same Problem here

Hi @Bohrfutter I have opened the enclosure, I have seen that you have "broken" the left/right side to access ESP8266/TUYA IC.
But, did you find glue? how is the pcb board "joined" to bottom part? Any help ? I want to extract pcb board without damaging enclosure.
Thanks.