arendst/Tasmota #5636: Support of more than one MCP23017 at one sonoff device

ladislav37, 2019-04-13:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would like to ask to add support for more than one MCP23017 (MCP230XX) I/O expander per sonoff device. In general you can have 8 MPC devices per one I2C line.. (because of address selection).

**Thank You**

**Describe alternatives you've considered**
No alternatives...

**Additional context**
No additional context...

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-13:
No supported

ladislav37, 2019-04-13:
Ok than, as soon as i will have hardware to test solution, i will do update and PR.. Closing the issue..

closed 2019-04-13 by ladislav37

ridikuel, 2019-08-27:
@ladislav37 have you started coding yet?

I have started implementing multiplicity of MCP230xx in my [fork](https://github.com/ridikuel/Sonoff-Tasmota-stepper).

I am preparing a testbed in the next days. Using Node MCU and at first 2 MCP230xx DIP Modules.

djpankdo, 2020-06-22:
Hi @ridikuel have you made this feature already available in the main rep?

SilentBob83, 2021-10-03:
Hello is the function of multiple MCPs now available ?

doublenick, 2023-02-08:
Hello!
Is this not implemented in Tasmota? I see it still supports only one MCP230xx device.

What is the way to not just specify one device address? 
#define USE_MCP230xx_ADDR 0x20

Jason2866, 2023-02-08:
@doublenick if you want it you have to do.

nccchirag, 2023-03-29:
@ladislav37 @doublenick 

Did you read the new mode2 posibility/functionality? It supports several MCP23xxx chips so no need to change any code. Just define your template data as shown in the examples.

_Originally posted by @arendst in https://github.com/arendst/Tasmota/issues/6316#issuecomment-1488346569_