arendst/Tasmota #5637: Factory Reset after a few minutes

42Rhino, 2019-04-13:
I have some Problems with my Shelly1. I have just flash it with Tasmota and selected the Shelly1 profile. Because i need Button functionality i changed the switch to a button. A few minutes everything works, but then Tosmota got a factory reset. 
I have tried it with DE and Basic version. All the same.

`{"NAME":"Shelly1 - Butt","GPIO":[0,0,0,0,21,90,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}`

Jason2866, 2019-04-13:
Issue template is missing. Since we are no clairvoyants no help can be provided.

blakadder, 2019-04-13:
have you properly configured the button
you might be activating factory reset after 40 seconds since buttons have extra functionality https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

Jason2866, 2019-04-13:
@blakadder  Good finding! This is the issue. You cant change the switch to a button on a Shelly1.

42Rhino, 2019-04-14:
Thanks for the help. Now i changed the SwitchMode to 3 and it works. Sorry, but i don't know anything about the commands bevore.

closed 2019-04-14 by 42Rhino