arendst/Tasmota #5638: SSerialReceived#Data=<value> doesnt work

Jason2866, 2019-04-13:
### BUG DESCRIPTION
Sending from wemosmini1 via Sserial5 6f6e to connected wemosmini2 didnt fire rule
as expected.

`17:11:39 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":421,"Rules":"on SSerialReceived#Data=on do power1 1 endon on SSerialReceived#Data=off do power1 0 endon"}`

```
17:07:34 CMD: rule on SSerialReceived#Data=on do power1 1 endon on SSerialReceived#Data=off do power1 0 endon
17:07:34 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":421,"Rules":"on SSerialReceived#Data=on do power1 1 endon on SSerialReceived#Data=off do power1 0 endon"}
```
wemosmini1:
```
17:10:27 CMD: SSerialSend5 6f6e
17:10:27 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
```
wemosmini2:
`17:10:28 MQT: tele/sonoff/RESULT = {"SSerialReceived":"on"}`

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : _____
- [ x] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ x] Development IDE - Compiler / Upload tools used : ____ / ____
- [ x] Provide the output of command ``status 0`` :
```
17:12:57 CMD: status 0
17:12:57 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["149"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:12:57 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:37:54","StartupUTC":"2019-04-13T14:35:03","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":125,"SaveAddress":"F4000"}}
17:12:57 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.8(sonoff)","BuildDateTime":"2019-04-13T16:06:07","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(c0eb301)"}}
17:12:57 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
17:12:57 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0F8AA384","040403A1","80041496","001028C2"]}}
17:12:57 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1459","IPAddress":"192.168.2.149","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:15:85:B3","Webserver":2,"WifiConfig":5}}
17:12:57 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1585B3","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:12:57 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 13 15:12:57 2019","Local":"Sat Apr 13 17:12:57 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:26","Sunset":"20:00"}}
17:12:57 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-04-13T17:12:57"}}
17:12:57 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-04-13T17:12:57","Uptime":"0T00:37:54","Vcc":3.041,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":9,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:05"}}}

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect to Wemos mini with configured serialbridges and connect RX-TX and TX-RX
Use command `SSerialSend5 6f6e` on Transmitter Wemosmini
rule on `SSerialReceived#Data=on do power1 1 endon on SSerialReceived#Data=off do power1 0 endon` on receiver wemosmini
GPIO2 is defined as Relay1i on receiver wemosmini

### EXPECTED BEHAVIOUR
Light up Led on GPIO2 when sending `on` and switching out when sending `off`

arendst, 2019-04-13:
I guess it want to receive JSON data. 

How about sending this over serial: ``{"data":"on"}``

Jason2866, 2019-04-13:
Tried:
```
18:32:23 CMD: sserialsend2 {"data":"on"}
18:32:23 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
18:32:55 CMD: sserialsend2 "data":"on"
18:32:55 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
18:33:07 CMD: sserialsend2 "data":"on"
18:33:07 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
18:33:42 CMD: sserialsend2 data:on
18:33:42 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
18:34:01 CMD: sserialsend2 {"on"}
18:34:01 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
18:34:16 CMD: sserialsend2 {"on"}
18:34:16 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
18:35:21 CMD: sserialsend2 {"data":"on"}
18:35:21 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
```
None did it

Received:
```
18:26:35 MQT: tele/sonoff/RESULT = {"SSerialReceived":""data":"on"
"}
18:27:32 MQT: tele/sonoff/RESULT = {"SSerialReceived":"{"data":on"}
"}
18:27:58 MQT: tele/sonoff/RESULT = {"SSerialReceived":"{"dapa":"on"}
"}
18:28:09 MQT: tele/sonoff/RESULT = {"SSerialReceived":"{"data":"on"}
"}
18:32:22 MQT: tele/sonoff/RESULT = {"SSerialReceived":"{"data":"on"}"}
18:32:55 MQT: tele/sonoff/RESULT = {"SSerialReceived":""data"8"on""}
18:33:07 MQT: tele/sonoff/RESULT = {"SSerialReceived":""data":"on""}
18:33:41 MQT: tele/sonoff/RESULT = {"SSerialReceived":"data:on"}
18:34:01 MQT: tele/sonoff/RESULT = {"SSerialReceived":"x"on"}"}
18:34:16 MQT: tele/sonoff/RESULT = {"SSerialReceived":"{"on"}"}
18:35:20 MQT: tele/sonoff/RESULT = {"SSerialReceived":"{"data":"on"}"}
```

Frogmore42, 2019-04-18:
Be aware that disabling interrupts for than tens of microseconds will eventually lead to failure https://github.com/espressif/ESP8266_NONOS_SDK/issues/90

The occurance rate is not high, but it isn't zero either.

Jason2866, 2019-05-06:
I update to latest nonos SDK 2.2.x from espressif and used latest Arduino ESP8266 feature stage.

Sended:
```
17:11:01 CMD: sserialsend2 {"data":"on"}
17:11:01 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:07 CMD: sserialsend2 {"data":"on"}
17:11:07 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:10 CMD: sserialsend2 {"data":"on"}
17:11:10 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:14 CMD: sserialsend2 {"data":"on"}
17:11:14 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:16 CMD: sserialsend2 {"data":"on"}
17:11:16 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:19 CMD: sserialsend2 {"data":"on"}
17:11:19 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:25 CMD: sserialsend2 {"data":"on"}
17:11:25 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:27 CMD: sserialsend2 {"data":"on"}
17:11:27 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:30 CMD: sserialsend2 {"data":"on"}
17:11:30 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:32 CMD: sserialsend2 {"data":"on"}
17:11:32 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:33 CMD: sserialsend2 {"data":"on"}
17:11:33 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:35 CMD: sserialsend2 {"data":"on"}
17:11:35 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:37 CMD: sserialsend2 {"data":"on"}
17:11:37 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:38 CMD: sserialsend2 {"data":"on"}
17:11:38 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:40 CMD: sserialsend2 {"data":"on"}
17:11:40 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
17:11:42 CMD: sserialsend2 {"data":"on"}
17:11:42 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
```
received:

```
17:11:00 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:06 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:10 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:13 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:15 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:16 MQT: tele/sonoff-53CF5F/STATE = {"Time":"2019-05-06T17:11:16","Uptime":"0T00:10:15","Vcc":2.945,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":9,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:05"}}
17:11:19 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:24 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:26 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:30 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:32 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:33 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:35 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:36 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:38 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:40 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
17:11:41 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"{"data":"on"}"}
```
No serial receive error. What i am doing wrong?

Jason2866, 2019-05-06:
Status 0
```

17:17:54 CMD: status 0
17:17:54 MQT: stat/sonoff-53CF5F/STATUS = {"Status":{"Module":18,"FriendlyName":["Light"],"Topic":"sonoff-53CF5F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:53","StartupUTC":"2019-05-06T15:01:01","Sleep":0,"CfgHolder":4617,"BootCount":74,"SaveCount":480,"SaveAddress":"F9000"}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS2 = {"StatusFWR":{"Version":"6.5.0.9(sonoff)","BuildDateTime":"2019-05-06T16:24:02","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(2a5f9d5)"}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F8AA384","040003A1","00041496","000000C0"]}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-53CF5F-3935","IPAddress":"192.168.2.154","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:53:CF:5F","Webserver":2,"WifiConfig":5}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-%06X","MqttClient":"sonoff-53CF5F","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS7 = {"StatusTIM":{"UTC":"Mon May 06 15:17:54 2019","Local":"Mon May 06 17:17:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:44","Sunset":"20:34"}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS10 = {"StatusSNS":{"Time":"2019-05-06T17:17:54"}}
17:17:54 MQT: stat/sonoff-53CF5F/STATUS11 = {"StatusSTS":{"Time":"2019-05-06T17:17:54","Uptime":"0T00:16:53","Vcc":2.946,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":9,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

Jason2866, 2019-05-15:
@arendst  could you take a look? Thx

arendst, 2019-05-15:
Some background:

1) Rules only trigger on JSON messages like ``{"TriggerName":{"ValueName":Value}}``
2) SSerialReceived data like ``{"SSerialReceived":"on"}`` won't work as there is no ValueName
3) SSerialReceived data like ``{"SSerialReceived":"{"data":"on"}"}`` won't work either as there are additional quotes around the TriggerValue

Bottomline, I have to redesign the (S)SerialReceived rule trigger along the line of providing something like ``{"Trigger":{"SSerialReceived":"on"}}`` where the rule would be like ``on Trigger#SSerialReceived=on do power1 1 endon``

Stay tuned...

arendst, 2019-05-15:
Redesigned to allowing all single JSON value pairs by inserting the Data ValueName.

So to test on a SSerialReceived value you would use a trigger like ``SSerialReceived#Data=``

Jason2866, 2019-05-15:
Thx! Does work now as needed :-)
Sended on transmitter wemos: 
```
17:32:36 CMD: sserialsend2 on
17:32:36 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
17:32:45 CMD: sserialsend2 off
17:32:45 MQT: stat/sonoff-53CF5F/RESULT = {"SSerialSend":"Done"}
```
results on receiver wemos with rule
`on SSerialReceived#Data=on do power1 1 endon on SSerialReceived#Data=off do power1 0 endon`: in
```
17:32:37 MQT: tele/sonoff-1585B3/RESULT = {"SSerialReceived":"on"}
17:32:37 RUL: SSERIALRECEIVED#DATA=ON performs "power1 1"
17:32:37 MQT: stat/sonoff-1585B3/RESULT = {"POWER":"ON"}
17:32:37 MQT: stat/sonoff-1585B3/POWER = ON
17:32:45 MQT: tele/sonoff-1585B3/RESULT = {"SSerialReceived":"off"}
17:32:45 RUL: SSERIALRECEIVED#DATA=OFF performs "power1 0"
17:32:45 MQT: stat/sonoff-1585B3/RESULT = {"POWER":"OFF"}
17:32:45 MQT: stat/sonoff-1585B3/POWER = OFF
```

closed 2019-05-15 by Jason2866

blakadder, 2019-05-16:
is SerialReceived possible as well? I'm asking for a friend!

arendst, 2019-05-16:
@blakadder Yes, since yesterday.

fluppie, 2019-05-17:
So this means I can also use this for ACK? See https://github.com/arendst/Sonoff-Tasmota/issues/4965

Jason2866, 2019-05-17:
@fluppie  yes