arendst/Tasmota #5639: RGB state always saved to flash eeprom

h3ndrik, 2019-04-13:
Changing the color or dimmer value of my rgb plug always causes a write to flash memory.

I tried fiddling with SetOption0, PowerOnState and PowerRetain but this only changes the situation for turning the relais and leds on and off. I can still see the "Flash write Count" increase if i change the color or brightness.

The device is an Tuya RGB smart plug, flashed with Sonoff-Tasmota 6.5.0(release-basic) (pre-compiled)
Configured with the template from the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/RGB-Smart-Plug-16A
Controlled via MQTT.

Can i somehow make tasmota save the state of the relais (power1) but prevent saving the led color values? Or just discard everything? I'd like to set the color from node-red every few minutes.

STATUS 0 OUTPUT:
```
16:22:19 MQT: tasmota/sonoff/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:22:19 MQT: tasmota/sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T01:32:30","StartupUTC":"2019-04-13T13:49:49","Sleep":50,"CfgHolder":4617,"BootCount":23,"SaveCount":1185,"SaveAddress":"FB000"}}
16:22:19 MQT: tasmota/sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-basic)","BuildDateTime":"2019-03-19T12:25:14","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:22:19 MQT: tasmota/sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ssid","test"],"TelePeriod":300,"Resolution":"558380C0","SetOption":["00028009","280500000100000000000000000000000000","00000000"]}}
16:22:19 MQT: tasmota/sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":432,"Free":568,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0F082380","000483A0","21021706","000001C0"]}}
16:22:19 MQT: tasmota/sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3288","IPAddress":"172.16.85.246","Gateway":"172.16.85.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.85.1","Mac":"CC:50:RE:DA:CT:ED","Webserver":2,"WifiConfig":4}}
16:22:20 MQT: tasmota/sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.mysecretserver.org","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_XXXXXX","MqttUser":"tasmota","MqttCount":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:22:20 MQT: tasmota/sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 13 15:22:20 2019","Local":"Sat Apr 13 16:22:20 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:03","Sunset":"19:38"}}
16:22:20 MQT: tasmota/sonoff/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:22:20 MQT: tasmota/sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2019-04-13T16:22:20","ENERGY":{"TotalStartTime":"2019-04-13T00:00:00","Total":0.095,"Yesterday":0.071,"Today":0.095,"Power":13,"ApparentPower":39,"ReactivePower":37,"Factor":0.33,"Voltage":230.9,"Current":0.169}}}
16:22:20 MQT: tasmota/sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2019-04-13T16:22:20","Uptime":"0T01:32:31","Vcc":3.256,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER1":"ON","POWER2":"ON","Dimmer":63,"Color":"93,160,17","HSBColor":"88,89,63","Channel":[36,62,6],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ssid","BSSId":"EE:08:RE:DA:CT:ED","Channel":11,"RSSI":36,"LinkCount":2,"Downtime":"0T00:00:09"}}}
```

CONSOLE OUTPUT:
```
16:29:46 WIF: Connected
16:29:51 SRC: MQTT
16:29:51 RSL: Received Topic tasmota/sonoff/cmnd/Dimmer, Data Size 2, Data 21
16:29:51 RSL: Group 0, Index 1, Command DIMMER, Data 21
16:29:52 MQT: tasmota/sonoff/stat/RESULT = {"POWER2":"ON","Dimmer":21,"Color":"31,53,5","HSBColor":"87,91,21","Channel":[12,20,1]}
16:29:52 CFG: Saved to flash at F9, Count 1187, Bytes 3584
```

curzon01, 2019-04-15:
`SaveData 0` ?

ascillato2, 2019-04-15:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-15 by ascillato2

h3ndrik, 2019-04-16:
thank you very much. must have read past this when scrolling through the wiki.

i suppose there isn't an option to handle LED and Relais differently? Somehow i can't find the code which prevents the write for normal power operation to dig into the code by myself:-/