arendst/Tasmota #5642: BlitzWolf SHP6 EU v1.4 LEDs not working as expected

svh1985, 2019-04-14:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
The blue and red LED do not work as expected. When I turn it on the blue LED starts flashing and does not stop. The red LED does not work, not even when the relay is powered.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : BlitzWolf SHP6 EU 1.4
- [x] Tasmota binary firmware version number used : 6.5.0.8 / (pre-compiled)
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
16:06:58 RSL: stat/sonoff/STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:06:58 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:05:42","StartupUTC":"2019-04-14T15:01:16","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":12,"SaveAddress":"F8000"}}
16:06:58 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.8(4fa4397-sonoff)","BuildDateTime":"2019-04-13T16:01:47","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:06:58 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
16:06:58 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":480,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
16:06:58 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7702","IPAddress":"172.16.0.24","Gateway":"172.16.0.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.0.2","Mac":"CC:50:E3:2A:3E:16","Webserver":2,"WifiConfig":4}}
16:06:58 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2A3E16","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:06:58 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 14 15:06:58 2019","Local":"Sun Apr 14 16:06:58 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:01","Sunset":"19:39"}}
16:06:58 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:06:58 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-04-14T16:06:58","ENERGY":{"TotalStartTime":"2019-04-14T15:52:25","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
16:06:58 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-04-14T16:06:58","Uptime":"0T00:05:42","Vcc":3.504,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WiFi","BSSId":"00:1D:AA:69:D5:88","Channel":9,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:34"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
16:08:01 CMD: weblog 4
16:08:01 RSL: stat/sonoff/RESULT = {"WebLog":4}
16:08:02 CFG: Saved to flash at F7, Count 13, Bytes 3584

```

### TO REPRODUCE
Flash, set Module type to BlitzWolf SHP 

### EXPECTED BEHAVIOUR
The expected behavior for me is:
-Blue blinking LED while connecting to WiFi connection 
-Blue LED off when WiFi is connected
-Red LED on when relay is powered
-Red LED off when relay is powered off

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-14:
Please use for Power Monitoring devices NOT version sonoff-basic.
The needed drivers are missing.

svh1985, 2019-04-14:
Not sure I understand, only the LED are an issue all other functionality works as expected:
[image]

ascillato, 2019-04-14:
Hi,

The blue led blinking means that your mqtt is not configured/connected. If you are not going to use mqtt, just disable it in the configuration menu. 

If you want the red led to behave like that, you will need to use the command ledmask. Please check the wiki for more information on this new command.

If you need further assistant on this configuration, please find us in the Tasmota support chat.

Thanks

closed 2019-04-14 by ascillato2