arendst/Tasmota #5649: problem GPIO sonoff Basic R2 cutted track

pilot1981, 2019-04-15:
Hi,
I have a SONOFF BASIC R2 (new layout): using 220v, I notice I don't receive signal of dry contact connected to GND and RX PIN...I try to open and close dry contact but SONOFF don't log any changes....

Using 3,3v, it work fine! Logging output say all works fine, GPIO contact communicate open/close signal...

I already tried to erase and flash again, I tried to do again soldering points...but problem remain!!!

blakadder, 2019-04-15:
You have physically modified the device beyond the factory specifications so we cannot offer support

pilot1981, 2019-04-15:
you give support only original HW? I can find any support on chat?

blakadder, 2019-04-15:
You can always ask in chat, issues on github is for code issues first of all.

closed 2019-04-15 by ascillato2