arendst/Tasmota #5650: Tuya Dimmer - ability to add a second button/switch

burnshomeautomation, 2019-04-15:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

burnshomeautomation, 2019-04-15:
I have a zemismart fan/light switch combo switch....
http://www.zemismart.com/smart-wifi-switch-for-fan-light-compatible-with-alexa-google-home-smart-life-app-control_p0153.html

I have flashed it with Tasmota and it partially works with the tuya-dimmer template.  All of the control is handled via tuya tx/rx.   The issue is, the tuya dimmer, doesnt allow me to add a second button/switch.  So matter which button i push locally on the switch, it sends the same messages.  If I leave the light switch turned off, the fan dimmer and switch work properly from tasmota or mqtt.

Please add ability to add additional switches to this template.

burnshomeautomation, 2019-04-16:
Is this going to be possible, what additional information do you need from me?

ascillato, 2019-04-16:
Hi, 

To add support to zemismart fan/light switch combo switch, it is needed to modify the Tasmota's tuya driver: https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_16_tuyadimmer.ino

@digiblur
@chaosmaster

Any idea?

chaosmaster, 2019-04-16:
@burnshomeautomation 
Please provide a Weblog4 when you use the different functions (ideally with commentary what actions you did)
So we can see what messages are exchanged.
Do the +/- buttons control only the light or only the fan? Do they control both?

burnshomeautomation, 2019-04-16:
I will get this uploaded as soon as I get back to the house.  The +/- buttons control the fan only, not the light.

digiblur, 2019-04-16:
@burnshomeautomation

Trying to remember when I spoke with you on discord but weren't you able to get the light relay to work when you didn't have the Tuya stuff picked? Or am I thinking wrong here.

burnshomeautomation, 2019-04-17:
@digiblur

No, everything is handled with the tuya rx/tx.  Everytime a button is pressed locally on the switch, you see the tx/rx pop up on the console.

@chaosmaster - Here are the logs from the various button presses.....

02:43:07 TYA: Set WiFi LED to state 3 (0)
02:43:07 TYA: TX Packet: "55aa000300010306"
02:43:08 TYA: Set WiFi LED to state 3 (0)
02:43:08 TYA: TX Packet: "55aa000300010306"
02:43:09 TYA: Set WiFi LED to state 3 (0)
02:43:09 TYA: TX Packet: "55aa000300010306"
02:43:10 TYA: Set WiFi LED to state 3 (0)
02:43:10 TYA: TX Packet: "55aa000300010306"
##### PRESSED FAN SPEED DOWN BUTTON ######################################
02:43:10 TYA: RX Packet: "55aa01070005010100010110"
02:43:10 TYA: RX - On State
02:43:10 SRC: Switch
02:43:10 TYA: Send Serial Packet Dim Value=255 (id=2)
02:43:10 TYA: TX Packet: "55aa0006000802020004000000ff14"
02:43:10 MQT: zemi/stat/office_switch/RESULT = {"POWER":"ON","Dimmer":100}
02:43:10 MQT: zemi/stat/office_switch/POWER = ON
02:43:10 TYA: RX Packet: "55aa0107000802020004000000f80f"
02:43:10 TYA: RX Dim State=248
02:43:10 TYA: Send CMND_DIMMER_STR=Dimmer 97
02:43:10 SRC: Switch
02:43:10 RSL: Received Topic /Dimmer, Data Size 2, Data 97
02:43:10 RSL: Group 0, Index 1, Command DIMMER, Data 97
02:43:10 MQT: zemi/stat/office_switch/RESULT = {"POWER":"ON","Dimmer":97}
02:43:10 TYA: RX Packet: "55aa01070005030100010011"
02:43:10 TYA: RX - Off State
02:43:10 SRC: Switch
02:43:10 TYA: Send Dim Level skipped due to 0 or already set. Value=0
02:43:10 MQT: zemi/stat/office_switch/RESULT = {"POWER":"OFF","Dimmer":97}
02:43:10 MQT: zemi/stat/office_switch/POWER = OFF
02:43:10 TYA: RX Packet: "55aa0107000802020004000000ff16"
02:43:10 TYA: RX Dim State=255
02:43:10 TYA: Send CMND_DIMMER_STR=Dimmer 100
02:43:10 SRC: Switch
02:43:10 RSL: Received Topic /Dimmer, Data Size 3, Data 100
02:43:10 RSL: Group 0, Index 1, Command DIMMER, Data 100
02:43:10 SRC: Light
02:43:10 TYA: Send Dim Level skipped due to 0 or already set. Value=255
02:43:10 TYA: SetDevicePower.rpower=1
02:43:10 TYA: TX Packet: "55aa0006000501010001010e"
02:43:10 MQT: zemi/stat/office_switch/RESULT = {"POWER":"ON","Dimmer":100}
02:43:10 MQT: zemi/stat/office_switch/POWER = ON
02:43:10 MQT: zemi/stat/office_switch/RESULT = {"POWER":"ON","Dimmer":100}
02:43:10 TYA: RX Packet: "55aa01070005010100010110"
02:43:10 TYA: RX - On State
02:43:10 TYA: RX Packet: "55aa0107000802020004000000ff16"
02:43:10 TYA: RX Dim State=255
02:43:10 TYA: RX Packet: "55aa01070005030100010011"
02:43:10 TYA: RX - Off State
02:43:10 SRC: Switch
02:43:10 TYA: Send Dim Level skipped due to 0 or already set. Value=0
02:43:10 MQT: zemi/stat/office_switch/RESULT = {"POWER":"OFF","Dimmer":100}
02:43:10 MQT: zemi/stat/office_switch/POWER = OFF
02:43:10 TYA: RX Packet: "55aa01070005010100010110"
02:43:10 TYA: RX - On State
02:43:10 SRC: Switch
02:43:10 TYA: Send Serial Packet Dim Value=255 (id=2)
02:43:10 TYA: TX Packet: "55aa0006000802020004000000ff14"
02:43:10 MQT: zemi/stat/office_switch/RESULT = {"POWER":"ON","Dimmer":100}
02:43:10 MQT: zemi/stat/office_switch/POWER = ON
02:43:10 TYA: RX Packet: "55aa0107000802020004000000ff16"
02:43:10 TYA: RX Dim State=255
02:43:11 TYA: Set WiFi LED to state 3 (0)
02:43:11 TYA: TX Packet: "55aa000300010306"
02:43:11 CFG: Saved to flash at F9, Count 402, Bytes 3584
[... 324 lines cut ...]
02:44:46 TYA: TX Packet: "55aa000300010306"
##### PRESSED LIGHT POWER BUTTON (OFF) ####################################
02:44:47 TYA: RX Packet: "55aa01070005010100010110"
02:44:47 TYA: RX - On State
02:44:47 TYA: RX Packet: "55aa0107000802020004000000f50c"
02:44:47 TYA: RX Dim State=245
02:44:47 TYA: RX Packet: "55aa01070005030100010011"
02:44:47 TYA: RX - Off State
02:44:47 SRC: Switch
02:44:47 TYA: Send Dim Level skipped due to 0 or already set. Value=0
02:44:47 MQT: zemi/stat/office_switch/RESULT = {"POWER":"OFF","Dimmer":97}
02:44:47 MQT: zemi/stat/office_switch/POWER = OFF
02:44:47 TYA: Set WiFi LED to state 3 (0)
02:44:47 TYA: TX Packet: "55aa000300010306"
02:44:47 CFG: Saved to flash at F6, Count 405, Bytes 3584
02:44:48 TYA: Set WiFi LED to state 3 (0)
02:44:48 TYA: TX Packet: "55aa000300010306"
02:44:49 TYA: Set WiFi LED to state 3 (0)
02:44:49 TYA: TX Packet: "55aa000300010306"
02:44:50 TYA: Set WiFi LED to state 3 (0)
02:44:50 TYA: TX Packet: "55aa000300010306"
02:44:51 WIF: Checking connection...
02:44:51 WIF: Connected

Hopefully this will give you some kind of an idea of what its doing

burnshomeautomation, 2019-04-17:
Is this the kind of info you were needing from the logs?

chaosmaster, 2019-04-17:
Yes, that looks good, I haven't had time to look into this yet.

burnshomeautomation, 2019-04-24:
Anything I can do to help on this one?

chaosmaster, 2019-04-24:
So the deivice has two power IDs (1 and 3):
55aa01070005**03**0100010011
55aa01070005**01**010001000f

Not sure which one is used for fan and which for light.

The code currently doesn't check the ID when it receives a power message.
That would have to be changed.
I currently don't have much time, @digiblur maybe you could look into that?

blakadder, 2019-05-06:
I'll just tack on to this discussion. I have an Oil Diffuser that has a TYWE3S chip but everything is controlled through the MCU. I've [documented most of the codes](https://blakadder.github.io/templates/oil_diffuser_550ml.html)

As you can see it uses different id for commands and introduces a few new things (RGB control, different power modes). I'm well aware all this support would require a major rewrite.

How difficult would it be to define command id's in a way like the SetOption34 defines tuyaid?

This device also has multiple power IDs:
- diffuser on/off/mode select - `55AA00060005**02**0400010011`
- fan speed (high/low) - `55AA00060005**65**0400010175`
- light on/off/modes - `55AA00060005**06**0400010318`

Currently the device is manageable using rules and my smart home hub. 
The only problem is there is no return state if the device buttons are used for control

shanekuz, 2019-05-10:
I have a very similar problem with some new three and four relay switches i have purchased, these units flash over the air the same way the Tuya Dimmer does they take the Tasmota firmware fine this way but do not seem to be GPIO controlled.  If i select Tuya Dimmer (54) then switching the switch on and off controls switch one on them and when i press button 2, 3, & 4 i get serial codes coming through the console so its serial controlled in the same way.     Can i control these with rules or do i need code changes to get these working?  It would be really good to get these going as these are certified for use in NZ and AU and look the same as Sonoff switches.
[image]
[image]
[image]
[image]
[image]

shanekuz, 2019-05-11:
The console under debug send status updates every few seconds and seems to show in the switched are on or off ok.

01:09:22 TYA: RX   Packet: "55aa0107000501010001000f"
--
01:09:22 TYA: RX - Off State
01:09:22 TYA: RX Packet:   "55aa01070005020100010010"
01:09:22 TYA: RX - Off State
01:09:22 TYA: RX Packet:   "55aa01070005030100010011"
01:09:22 TYA: RX - Off State
01:09:22 TYA: RX Packet:   "55aa01070005040100010012"
01:09:22 TYA: RX - Off State
 
 
01:09:51 TYA: RX Packet:   "55aa01070005010100010110"
01:09:51 TYA: RX - On State
01:09:51 TYA: RX Packet:   "55aa01070005020100010111"
01:09:51 TYA: RX - On State
01:09:51 TYA: RX Packet:   "55aa01070005030100010112"
01:09:51 TYA: RX - On State
01:09:51 TYA: RX Packet:   "55aa01070005040100010113"
01:09:51 TYA: RX - On State

blakadder, 2019-05-11:
You can use rules to control them like I did above, its not a perfect solution but it'll make the switch remotely operable atleast

shanekuz, 2019-05-12:
This will leave it so that if someone manually operates the switch it will not send MQTT updates though right?

blakadder, 2019-05-13:
Yes, until we can have a rule trigger on serial received packets

blakadder, 2019-05-16:
We got a rule trigger on serial received packets (SerialRecevied#data=value), so it will now be possible to trigger relay states on serial messages using rules.
Gonna do some testing this evening on my tuya device with MCP controller and see how it works out

burnshomeautomation, 2019-05-17:
how do i use the rules to send the command?  I can't seem to get it to work.

blakadder, 2019-05-17:
for example `serialsend5 55AA00060005020400010213` but you need to be using Generic module and `backlog baudrate 9600; seriallog 2`. See [my device](https://blakadder.github.io/templates/oil_diffuser_550ml.html) and how I handled it.

For a complete setup you'd need to capture codes by using the app to issue commands and monitor serial traffic with a serial terminal capable of reading hex codes.

thirug010, 2019-05-17:
@burnshomeautomation I have the solution for the issue , I twas tested with the same switch refer This Post https://github.com/arendst/Sonoff-Tasmota/issues/5545#issuecomment-493599301

@shanekuz , I believe Tuya Protocol is standard it worked for 4 Gang should work for 3 Gang with setoption38 =3 

Thanks 
Thiru

burnshomeautomation, 2019-05-19:
can you do an OTA firmware update for this?

burnshomeautomation, 2019-05-20:
nevermind, i got it on there, ill test it tonight and let you know how it performs.  Thank you!

thirug010, 2019-05-20:
@burnshomeautomation , Please use this Branch this latest and fully Tested Dev Branch : https://github.com/thirug010/Sonoff-Tasmota

burnshomeautomation, 2019-05-21:
@thirug010, That worked like a champ.  I was able to set it all up in home assistant with mqtt, and it keeps the status correctly either pushing the buttons locally or through mqtt.  You are awesome!

thirug010, 2019-05-21:
@burnshomeautomation Thank you for testing it.

closed 2019-07-02 by ascillato2

meingraham, 2019-07-02:
@thirug010 Is your code something that should/could be incorporated into the main Tasmota codebase? If so, could you create a pull request to add your solution?

Mike

thirug010, 2019-07-06:
@meingraham, Code was merged to dev branch. 

Thanks 
Thiru

tyjtyj, 2019-07-15:
Can anyone suggest how to flash this to tasmota ? does Tuya convert works ? I have the Fan light switch UK model.

ascillato, 2019-07-15:
@tyjtyj

Please, address your issue to the Tasmota support chat. Thanks