arendst/Tasmota #5652: Question about the Shelly 2.5 GPIO connector

greensouth, 2019-04-15:
Hi!!

Does anybody know what is the name of the wires, or the connector type used for Shelly 2.5 programming? They are smaller than the ones used by -e.g.- the Raspberry GPIO port. They just don't fit and I don't know the name so I can order one.

Thanks in advance!
;)

ascillato2, 2019-04-15:
https://github.com/arendst/Sonoff-Tasmota/wiki/Shelly-1-and-2#serial-connection-1

closed 2019-04-15 by ascillato2

kroon040, 2019-04-23:
Hi,

Here is not the answer, the shelly 1 and shelly 2 uses the normal dupont cables, but what cable is used for flashing the shelly 2.5? It is a debug header cable, but what size of pins on de shelly2.5 side?

regards Richard

greensouth, 2019-04-29:
> Hi,
> 
> Here is not the answer, the shelly 1 and shelly 2 uses the normal dupont cables, but what cable is used for flashing the shelly 2.5? It is a debug header cable, but what size of pins on de shelly2.5 side?
> 
> regards Richard

it's like the connectors used when linking two LED stripes. I soldered that tiny wire which is used for closing bread bags, to dupont connectors :) and worked like a charm

jasase, 2019-04-29:
I have the same problem. @greensouth do you have some pictures or a little bit more details?

greensouth, 2019-04-30:
Take some soldering iron, a thin wire and solder it to the metallic tip of the Dupont connector. Then you can insert easily in the Shelly.

thxthx0, 2019-05-02:
or buy some **1.27 mm Pin Headers** :)

e.g. https://www.aliexpress.com/item/1-27mm-1-27-Pin-Header-Single-Row-Male-Breakaway-PCB-Board-Connector-Pinheader-1-2/32914658645.html?spm=a2g0s.9042311.0.0.5c1f4c4dvITIsn

Frigorio, 2019-05-19:
I did my own using a 5 rigid wire intercom cable, some hot glue and a momentary switch... works like a charm 😉

cinemarene, 2019-06-07:
For me LED pin connectors are a real help.

[image]
[image]
[image]
[image]

joba-1, 2019-06-08:
Clever to ask.
Whatever you use, insert it very gently. 
I broke off the sockets including pcb routes trying to insert small dupont pins :( 
At least I broke no connections. The device still works fine with the original fw.

supermaninja, 2019-09-21:
I broked the connector on the shelly 2.5, are on the shelly other point where I can find rx tx.....ecc?

Frigorio, 2019-09-23:
> Whatever you use, insert it very gently.

[image]
[image]
[image]
[image]

hallard, 2021-11-16:
Something like that also
[image]
https://www.digikey.com/en/products/detail/chip-quik-inc/F127T254P06/5978243

Need to build my own with push button :-)