arendst/Tasmota #5655: T1 US 3 gangs random switching @ Tasmota 6.5.0 bin not @ 6.4.1 basic

ornolio, 2019-04-15:
I would like to report a problem on T1 US 3 gangs Tasmota @ 6.5.0 bin randomly switch on (3 to 5 times a day (never switch off) on any channels, one channel at time.
I have been using 21 x Sonoff Touch US 1 gang since 2 years, some flashed with Tasmota: 6.4.1 sensor some other run Tasmota 6.5.0 basic version.
I also run Tasmota 6.5.0 basic ["Version":"6.5.0(release-sonoff)")] on a Sonoff T1 US 2 gangs. 
The home automation is performed on a Raspberry host running Hassio v. 0.91.3. Everything works like a charm.

Recently I installed 1 x Sonoff T1 US 3 gangs with Tasmota 6.5.0 bin, recommended by the Tasmota wiki. 

Unfortunately the T1 us 3 gangs  randomly switch on, particularly at night, more frequently on gang 2, but also on gang 3 and sometimes on gang 1. 

Even if the Hassio logs DO NOT show any loss of WIFI connection, but simply random power on.
I setup the Hook UP's configuration suggested here: https://www.youtube.com/watch?v=31IyfM1gygo by the Hook UP: "Fix Random Switching in Tasmota by Adjusting Retain Settings" , but problem was not solved.

As loads on T1 are 50W COB LEDs on channel 1 and 30W on channels 2 and 3 (see https://www.banggood.com/High-Power-30W-50W-80W-COB-LED-Flood-Light-Waterproof-Iodine-tungsten-Lamp-for-Outdoor-AC220V-p-1326561.html) and the power lines leaving the 3 relays of T1 run for about 6-8 meters and  Sonoff T1 US also has an RF command at 315 Mhz which I do not use, I arbitrary assumed that the unwanted switch-on could be due to RF noise detected by the radio frequency decoder, su I interrupted the output of the chip SYN470R dedicated to RF decoding. Unfortunately that didn't solve the problem too.

After several attempts to change Tasmota versions  (sensor, classic...), I was  finally successful by downgrading Tasmota to version 6.4.1 (basic).

Any idea why new version seems to have this impact on random switch T1 ?
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

STATUS 0 OUTPUT HERE:
08:06:10 MQT: stat/sala-123/STATUS = {"Status":{"Module":30,"FriendlyName":["sala-123","Sonoff2","Sonoff3"],"Topic":"sala-123","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":60,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:06:10 MQT: stat/sala-123/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff123.ino.bin","RestartReason":"Software/System restart","Uptime":"1T07:36:39","StartupUTC":"","Sleep":50,"BootCount":18,"SaveCount":188,"SaveAddress":"FA000"}}
08:06:10 MQT: stat/sala-123/STATUS2 = {"StatusFWR":{"Version":"6.4.1(basic)","BuildDateTime":"2018-12-24T14:43:30","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
08:06:10 MQT: stat/sala-123/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["TIM-24084959","TIMB-24084959"],"TelePeriod":300,"SetOption":["00002009","558180C0","00000000"]}}
08:06:10 MQT: stat/sala-123/STATUS4 = {"StatusMEM":{"ProgramSize":445,"Free":556,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F082780","000083A0","21021706","000001C0"]}}
08:06:10 MQT: stat/sala-123/STATUS5 = {"StatusNET":{"Hostname":"sala-123-0007","IPAddress":"192.168.1.126","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:92:00:07","Webserver":2,"WifiConfig":5}}
08:06:10 MQT: stat/sala-123/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.56","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_920007","MqttUser":"ornolio","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:06:10 MQT: stat/sala-123/STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 02 08:06:10 1970","Local":"Fri Jan 02 08:06:10 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99,"Sunrise":"07:43","Sunset":"16:04"}}
08:06:10 MQT: stat/sala-123/STATUS10 = {"StatusSNS":{"Time":"1970-01-02T08:06:10"}}
08:06:10 MQT: stat/sala-123/STATUS11 = {"StatusSTS":{"Time":"1970-01-02T08:06:10","Uptime":"1T07:36:39","Vcc":3.543,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"TIM-24084959","BSSId":"10:13:31:70:95:F3","Channel":11,"RSSI":58}}}

ascillato, 2019-04-15:
Hi, 

Sorry, this issue is duplicated. Exactly this had been addressed before. One case was a hardware defect, other was retained messages in the mqtt broker. As the troubleshooting template is incompleted, there is some information missing to properly help you. 

Let's try the following:

Upload the same firmware 6.5.0 you used before (provide the status 0 of that) and go to the console and type `weblog 4`. This will increase the logging level in order to see where this switching is coming from. Please, copy and paste here the output of the console when this switching happens.
You can have SRC: BUTTON if the issue is hardware related or SRC: MQTT if it is retained messages.

ascillato, 2019-04-17:
@ornolio

Hi, any news on this?

ornolio, 2019-04-17:
> A: arendst/Sonoff-Tasmota
> Tasmota 6.5.0 bin not @ 6.4.1 basic (#5655)

Hello Adrian,

 

Thank You for Your replay.

 

Yes, there is a great news, it seems it is my fault because I upgraded using
the development version of 6.5.0 which caused the described problem,
solved both with regular 6.4.1 bin and 6.5.0 bin. Im still performing some
more test before closing the issue, because the switch on event are rare (5
times a day or so) and cannot be reproduced.

 

Concerning your reporting of duplicated issue, I reported this problem for
Sonoff Touch (never observed on my T1 2&3 gangs, as I fixed it applying a
better configuration following the Hook Up suggestion here:
<https://www.youtube.com/watch?v=31IyfM1gygo>
https://www.youtube.com/watch?v=31IyfM1gygo that I mentioned.

 

Ill close to finish and close the issue today or tomorrow morning.

 

Best Regards.

P

 

Pier Giacomo Ornolio
mobile: +39.335.7831.433
ornolio@gmail.com or <mailto:ornolio@gmail.com%20or>  ornolio@libero.it
posta certificata: ing.piergiacomo.ornolio@pec.it
www.skype.com 'pgornolio'

  _____  

<https://github.com/arendst/Sonoff-Tasmota/issues/5655#issuecomment-48391415
7>  it on GitHub, or mute
8UOo-lLks5vhohCgaJpZM4cwgHX>  the thread.
<https://github.com/notifications/beacon/ADR9vzvN1vDSB_vid821zpjzkkCwYdA9ks5
vhohCgaJpZM4cwgHX.gif>

closed 2019-04-17 by ornolio