arendst/Tasmota #5659: Browser suggestions often change Tasmota settings unwittingly

melikyuksel, 2019-04-16:
Interesting "bug". For example, Safari suggests previously visited websites when you type random letters they start with. And then it actually goes and pre-visits these websites for you sometimes, even if you're not going to them. This causes it to often apply Tasmota settings that are changed with simply visiting the URL. Is there a way to turn off these explicit URL settings so that doesn't happen?

blakadder, 2019-04-16:
Are you asking whether Tasmota can change your browser's behavior? If so the answer is: No.

ascillato2, 2019-04-16:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-16 by ascillato2

joba-1, 2019-04-16:
Hm, somehow I doubt, Safari (or any other half decent browser) reopens POST requests by itself, and if GET requests are used to change settings somewhere within tasmota, that would be a valid issue. Anyways, @melikyuksel currently not enough info. Please follow the guidelines. I use Safari and have never seen this symptom.

melikyuksel, 2019-04-27:
I don't know why everyone got so offended over this issue. When you change MQTT settings in 6.5.0, the results are sent to the device through GET requests (so it is a valid issue @joba-1). For example, when I set MQTT settings, my URL bar is populated as such:

`http://lobby-plug.local/mq?mh=server&ml=1883&mc=lobby-plug_%2506X&mu=username&mp=password&mt=lobby-plug&mf=%25prefix%25%2F%25topic%25%2F&save=`

Even the *password* is sent in plaintext, and thus stored in my browser history. If I start to type in "lobby-plug" in the URL bar later, my browser may suggest this previous configuration and visit it (even if I changed MQTT settings later), wreaking havoc on my settings without me finding out until too late.

meingraham, 2019-04-27:
Honest question... how is Tasmota to avoid having the browser record this URL, entered in the browser, from being recorded in the browser's history? Once it's in the browser's history, if the user types and then hits enter on this command, how is Tasmota to know to ignore this particular command... even if it's "old"?

melikyuksel, 2019-04-27:
Tasmota should be sending the data in the form as POST, not as GET, so that the settings don't get recorded in browser histories (and the passwords not sent in plaintext).

meingraham, 2019-04-27:
Understood. A browser will not put into its history the POST message body; yes?

What would the URL of the same command example you used above be as formatted as a POST request?

Thanks.

melikyuksel, 2019-04-29:
Can we re-open this issue?

blakadder, 2019-05-01:
Reason being?

meingraham, 2019-05-01:
@melikyuksel 

I think your request is best addressed as a new feature request issue. When you open the new issue, please complete the template fully providing the examples contrasting the GET URL vs. the POST URL and describing the challenges posed by GET (sensitive information held in browser history, inadvertent resetting to older configuration etc.) vs. the benefits of using POST.

Even better would be if you are able to code the change yourself and provide it as a pull request. Understand that there may need to be a mechanism to keep backward compatibility unless the new POST method allows transparent or easy transition for any older deployments that use the current HTTP mechanism.

Regards.

Mike

melikyuksel, 2019-05-01:
@blakadder Because the code as it stands causes problems?