arendst/Tasmota #5660: SHELLY RGBW2 Tasmota

GPL71, 2019-04-16:
Is SHELLY RGBW2 supported by tasmota?
I've seen someone who sell SHELLY RGBW2 flashed TASMOTA

blakadder, 2019-04-16:
It is not officially supported but it might work with templates.

closed 2019-04-16 by blakadder

GPL71, 2019-07-11:
@blakadder Please can you you help me find the gpio for external switch? Thank you

blakadder, 2019-07-11:
and how do you imagine i would be able to do that without the device in front of me?

joba-1, 2019-07-11:
no guarantees, but similar to a tip I got from Jason2866, you can try to configure it as wemos, configure each gpio as a switch and look which gpio is triggered in weblog when the external switch is pushed

GPL71, 2019-07-13:
@joba-1 thank you tris is a good idea. I konw which gpio are  R G B W.....

mientki, 2019-08-05:
@GPL71  Can you tell which GPIO correspond to particular RGBW colors?

Fibrino, 2019-08-06:
> @joba-1 thank you tris is a good idea. I konw which gpio are R G B W.....

Hi GPL71. Could you please share the information you've collected currently?
I would also like to install Tasmota on Shelly RGBW2...

amdnox, 2019-09-03:
Hey I´m not CPL71 but I guess I have the answer to that question.

GPIO
0 = FLASH
1 = Tx
2 = LED Status
3 = Rx
4 = PWM White
5 = AC Switch
6-11 I´m not sure
12 = PWM RED
13 = I´m not sure
14 = PWM Blue
15 = PWM Green
16 = I´m not sure

A0 is also unclear because it output a value and it changes depending on the RGBW stripe Color but its alway 326-372

farconada, 2020-10-30:
Any plan to support Shelly RGBW2 ?

blakadder, 2020-10-30:
https://templates.blakadder.com/shelly_RGBW2.html

kugelkopf123, 2020-10-30:
And here some more infos:

https://tasmota.github.io/docs/devices/Shelly-RGBW2/