arendst/Tasmota #5662: List with IR Codes

helmar74, 2019-04-16:
Hi everbody,

as  the [YTF IR Bridge](https://github.com/arendst/Sonoff-Tasmota/wiki/YTF-IR-Bridge) is supported since version 6.4.1.8, 
I ordered one and i started to collect  IR Remote Codes:
[IR Code List](https://github.com/arendst/Sonoff-Tasmota/wiki/Codes-for-IR-Remotes-(for-YTF-IR-Bridge))

Please be aware that you should use Tasmota > 6.5.0.3 as there's a bug for IR Codes > 32Bit: #5523

Everbody is invited to also collect IR Codes to get a shared information for everyone.

Regards

Helmar

helmar74, 2019-04-16:
Closing issue

closed 2019-04-16 by helmar74

ascillato2, 2019-04-19:
@helmar74 

Hi, 

Please, could you add this to the wiki? 

Thanks

helmar74, 2019-04-19:
It' s in there already.....
Look at the device site

ascillato, 2019-04-19:
Great!

Thank you very much :+1: