arendst/Tasmota #5663: PowerOnState with LedMask Issue

meingraham, 2019-04-16:
### BUG DESCRIPTION
I finally installed one of the Kesen KS-604S in the house (I was just testing before). I hadn't noticed this before, but it appears as though the `PowerOnState 0` setting is not taking effect. The relays that are ON when I restart the plug are coming back in the ON state after a restart.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Kesen KS-604S
- [x] Tasmota binary firmware version number used : 6.5.0.8 / pre-compiled
- [x] Development IDE - Compiler / Upload tools used : N/A / Tuya-Convert
- [x] Provide the output of command ``status 0`` :
```
STATUS = {"Status":{"Module":0,"FriendlyName":["Plug","Lights","USB","LED2"],"Topic":"myTopic","ButtonTopic":"","Power":15,"PowerOnState":0,"LedState":1,"LedMask":"0001","SaveData":1,"SaveState":0,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020300/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:28","StartupUTC":"2019-04-16T13:19:25","Sleep":50,"CfgHolder":4617,"BootCount":21,"SaveCount":48,"SaveAddress":"FA000"}}
STATUS2 = {"StatusFWR":{"Version":"6.5.0.8(4fa4397-sonoff)","BuildDateTime":"2019-04-13T16:01:33","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008108","280500000100000000000000000000000000","000000C8"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":506,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
STATUS5 = {"StatusNET":{"Hostname":"myTopic-<deviceID>","IPAddress":"192.168.1.nnn","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"myMAC","Webserver":2,"WifiConfig":4}}
STATUS6 = {"StatusMQT":{"MqttHost":"myHost","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_myMACid","MqttUser":"myUserID","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 16 13:21:53 2019","Local":"Tue Apr 16 09:21:53 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"06:53","Sunset":"19:59"}}
STATUS10 = {"StatusSNS":{"Time":"2019-04-16T09:21:53"}}
STATUS11 = {"StatusSTS":{"Time":"2019-04-16T09:21:53","Uptime":"0T00:02:28","Vcc":3.086,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":59,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","Wifi":{"AP":1,"SSId":"mySSID","BSSId":"E4:F0:42:DE:76:F4","Channel":6,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [x] Provide the output of console when you experience your issue if apply :
```
09:19:21 HTP: Restart
09:19:22 SRC: WebGui from 0.0.0.0
00:00:00 CFG: Loaded from flash at F4, Count 46
00:00:00 Project sonoff Plug Version 6.5.0.8(4fa4397-sonoff)-2_3_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP-, SSId Rich's Network, Channel 1, BSSId 54:E4:3A:EC:AA:D8, RSSI -92, Encryption 1
00:00:04 WIF: Network 1, AP-, SSId ATT8FfL25W, Channel 1, BSSId F8:18:97:75:98:15, RSSI -91, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId Acosta, Channel 5, BSSId 24:F5:A2:0F:34:09, RSSI -89, Encryption 1
00:00:04 WIF: Network 3, AP-, SSId Acosta-guest, Channel 5, BSSId 2A:F5:A2:0F:34:09, RSSI -88, Encryption 0
00:00:04 WIF: Network 4, AP-, SSId DIRECT-89-HP OfficeJet Pro 8710, Channel 6, BSSId 84:A9:3E:FE:8F:8A, RSSI -58, Encryption 1
00:00:04 WIF: Network 5, AP1, SSId mySSID, Channel 6, BSSId E4:F0:42:DE:76:F4, RSSI -50, Encryption 1
00:00:04 WIF: Network 6, AP-, SSId MySpectrumWiFib0-2G, Channel 6, BSSId E8:AD:A6:47:06:B6, RSSI -90, Encryption 1
00:00:04 WIF: Network 7, AP-, SSId CLC, Channel 6, BSSId 02:27:22:E9:49:DB, RSSI -91, Encryption 1
00:00:04 WIF: Network 8, AP-, SSId Gaston Dr 2G, Channel 9, BSSId B0:BE:76:08:D1:D3, RSSI -88, Encryption 1
00:00:04 WIF: Network 9, AP-, SSId Avalon, Channel 10, BSSId 94:10:3E:9F:D0:94, RSSI -82, Encryption 1
00:00:04 WIF: Network 10, AP-, SSId Avalon_EXT, Channel 10, BSSId C0:FF:D4:B6:67:10, RSSI -88, Encryption 1
00:00:04 WIF: Network 11, AP-, SSId Avalon-guest, Channel 10, BSSId 96:10:3E:9F:D0:96, RSSI -83, Encryption 0
00:00:04 WIF: Network 12, AP-, SSId HP-Print-C5-Photosmart 6520, Channel 11, BSSId 38:EA:A7:39:4E:C5, RSSI -71, Encryption 1
00:00:04 WIF: Network 13, AP-, SSId benz, Channel 11, BSSId 58:6D:8F:13:62:0F, RSSI -89, Encryption 1
00:00:04 WIF: Network 14, AP1, SSId mySSID, Channel 11, BSSId myBSSID, RSSI -61, Encryption 1
00:00:04 WIF: Network 15, AP-, SSId ATTuc57EFi, Channel 11, BSSId 78:96:84:6E:B0:40, RSSI -84, Encryption 1
00:00:04 WIF: Network 16, AP-, SSId ATTQczrS22, Channel 11, BSSId BC:2E:48:BB:5F:E0, RSSI -89, Encryption 1
00:00:04 WIF: Patch issue 2186
00:00:04 WIF: Connecting to AP1 mySSID in mode 11N as myTopic-<deviceID>...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on myTopic-<deviceID> with IP address 192.168.1.nnn
[... 47 lines cut ...]
**Rule**
```
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":438,"Rules":"ON System#Boot DO Power1 On ENDON ON Power2#State DO Power4 %value% ENDON"}
```

I have set `PowerRetain 0`.  I checked and I don't believe this behavior is due to retained messages from my MQTT broker. I used MQTT-Explorer (the only retained message it showed was the LWT message) and also issued mosquitto_sub to record the messages going to the plug.

**mosquitto_sub**
```
09:19:38 tele/myTopic/LWT Online
09:19:38 cmnd/myTopic/POWER (null)
09:19:38 tele/myTopic/INFO1 {"Module":"KS-604S","Version":"6.5.0.8(4fa4397-sonoff)","FallbackTopic":"cmnd/DVES_myMACid_fb/","GroupTopic":"sonoffs"}
09:19:38 tele/myTopic/INFO2 {"WebServerMode":"Admin","Hostname":"myTopic-<deviceID>","IPAddress":"192.168.1.nnn"}
09:19:38 tele/myTopic/INFO3 {"RestartReason":"Software/System restart"}
09:19:38 stat/myTopic/RESULT {"POWER1":"ON"}
09:19:38 stat/myTopic/POWER1 ON
09:19:38 stat/myTopic/RESULT {"POWER2":"ON"}
09:19:38 stat/myTopic/POWER2 ON
09:19:38 stat/myTopic/RESULT {"POWER3":"ON"}
09:19:38 stat/myTopic/POWER3 ON
09:19:38 stat/myTopic/RESULT {"POWER4":"ON"}
09:19:38 stat/myTopic/POWER4 ON
09:19:40 stat/myTopic/RESULT {"POWER1":"ON"}
09:19:40 stat/myTopic/POWER1 ON
```

ascillato, 2019-04-16:
Is the same issue as https://github.com/arendst/Sonoff-Tasmota/issues/5594 ?

meingraham, 2019-04-16:
@ascillato,

Perhaps. They are both related to PowerOnState and both with Templates. But the operational behavior for my device is as expected after startup. Relays toggle, LEDs work, etc. The issue is only that the relays' state on reboot is not OFF.

Based on the Log and the Mosquitto Log which show no `POWER` `cmnd` MQTT messages, do you still suggest that I disable MQTT and try it again?

FYI - I can only restart the device (i.e., I cannot cycle power without flipping a circuit breaker for the entire circuit in the house).

Mike

meingraham, 2019-04-17:
Perhaps the common element, based on @BostonMikeGetHub's testing, is PowerOnState 0 using web GUI restart. As I don't have a means to cycle power on my in wall receptacle, my testing has been exclusively done by web GUI restart.

BostonMikeGetHub, 2019-04-17:
@meingraham, I think the issues are related to PowerOnState.  Issue #5594 occurs when PowerOnState = 3. Issue #5663 occurs when PowerOnState = 0.

meingraham, 2019-04-17:
Alrighty... hopefully Theo has enough information with the two scenarios to find the root cause.

ascillato2, 2019-04-18:
Let's gather both issues into one.

If the issues result not to be related, we will reopen this one.

closed 2019-04-18 by ascillato2