arendst/Tasmota #5664: Missing font in xdrv_01_webserver.ino

bakman2, 2019-04-17:
### BUG DESCRIPTION
xdrv_01_webserver.ino - line 248 states: `font-family:verdana;`

Verdana is not available on my system and as there is no fallback it becomes Times New Roman and looks bad.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : RF brige
- [x] Tasmota binary firmware version number used : precompiled 6.5.0
- [x] Development IDE - Compiler / Upload tools used : N/A
- [x] Provide the output of command ``status 0`` :
```
11:10:42 MQT: stat/sonoff-rf-bridge/STATUS11 = {"StatusSTS":{"Time":"2019-04-17T11:10:42","Uptime":"0T00:18:22","Vcc":3.152,"SleepMode":"Normal","Sleep":0,"LoadAvg":999,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"xx:D9:E7:xx:E1:29","Channel":12,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Provide the output of console when you experience your issue if apply :N/A

### TO REPRODUCE

Use a system without `verdana` and access the webgui.

### EXPECTED BEHAVIOUR

Pretty font rendering.
Suggestion: change it to `font-family:verdana;sans-serif` so that there is a fallback available.

### SCREENSHOTS

[image]

[image]

Jason2866, 2019-04-17:
As designed, install Font Verdana or use it as it is.
Since the main goal of Tasmota is small code size there will be no Design optimization for Webfrontend for different Fonts which increase code size.

bakman2, 2019-04-17:
- verdana is a licensed font
- verdana falls under the sans-serif family, so change verdana to sans-serif to make it work for everyone without sacrificing bits.

blakadder, 2019-04-17:
Verdana shouldn't be a requirement since its is not an open font and is not available on all systems by default. The suggestion to change to entire 'sans-serif' family is valid

ascillato2, 2019-04-18:
Closing this issue as it has been added by Theo. Thanks for sharing your ideas

closed 2019-04-18 by ascillato2