arendst/Tasmota #5665: Let's Encrypt switching root certificates which will impact USE_MQTT_TLS_CA_CERT

tssva, 2019-04-17:
Let's Encrypt has [announced](https://letsencrypt.org/2019/04/15/transitioning-to-isrg-root.html) that beginning on July 8, 2019 by default they will begin issuing a certificate chain which leads to their own ISRG Root X1 certificate instead of the IdenTrust root certificate which has previously been used.

The USE_MQTT_TLS_CA_CERT feature currently uses the IdenTrust certificate. Beginning July 8th MQTT using TLS this will fail for newly issued certificates. Just replacing the IdenTrust certificate in the code with the ISRG Root X1 certificate as of July 8th isn't a solution. Previously issued certificates will be valid for up to 90 days after July 7th which means through October 5th. Also users may wish to override the default and continue to receive cross-signed certificates after July 8th, so they can schedule their migration to a more appropriate time for them.

This is an issue which is going to need to be addressed through a combination of code, documentation and an awareness effort.

A couple of possible code options:

1. Use this as an opportunity to turn USE_MQTT_TLS_CA_CERT into a more generic option not tied to Let's Encrypt by adding a config option to specify which header file to include. Default to the existing header until some time after July 8th in order to prevent a surprise change in behavior. Ship header files for both Let's Encrypt certificates. Add documentation detailing how to create header files for other CAs.

2. Keep USE_MQTT_TLS_CA_CERT focused on Let's Encrypt. Include both certificates in the sonoff_letsencrypt.h file and add a config item to select which to use. Again default to the current certificate until sometime after July 8th.

As far as documentation a wiki page describing the issues and choices going forward should be considered. Links to resources on the Let's Encrypt website regarding how to configure Let's Encrypt ACME clients to continue to request cross-signed certificates and  to manually request certificate renewal to get certificates signed with the new root prior to the normal expiration of existing certificates to facilitate a timed migration should probably be included.

As far as awareness the issue and a link to the wiki page might be included in release notes and possibly use the #warning to issue a warning during compile if USE_MQTT_TLS_CA_CERT is enabled.

I'm willing to contribute to PRs and wiki pages to address these but I just installed Tasmota for the first time yesterday and since I don't yet know the community thought trying to get some consensus first would be more appropriate.

Jason2866, 2019-04-17:
Thanks for your input. Could be so kind and add / edit wiki?

blakadder, 2019-04-18:
You have my blessing

ascillato2, 2019-04-23:
Thanks for sharing this information. Very appreciated :+1:

closed 2019-04-23 by ascillato2

timstanley1985, 2019-05-08:
Would either of you be able to point me in the direction of some guidance on how to create a header file please?. I would like to use my own self signed CA cert as TLS fingerprint isn't working with 2.5.0