arendst/Tasmota #5668: Hvac panasonic ac

riddik14, 2019-04-17:
I noticed that tasmota does not support hvac ir command, can you implement something about it? I tried to sniff the ir code but with the ir send command it doesn't work, does anyone use irsend / irhvac to control panasonic air conditioners?

Jason2866, 2019-04-17:
Panasonic AC isnt supported in the used library IRremoteESP8266
Controlling AC with IR isnt just a simple sended command it uses a Protocol.
So somebody has to decode the protocol and expand the library with.
IR basic work and protocol implementation isnt the focus of Tasmota.
Maybe somebody does it someday...

ascillato2, 2019-04-18:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-18 by ascillato2

hpirila, 2019-05-13:
Panasonic AC is now supported in latest version IRremoteESP8266 2.6. It would be nice to get that into upcoming tasmota release.