arendst/Tasmota #5670: EtekCity Voltson Template

tj31moll, 2019-04-18:
**Have you look for this feature in other issues and in the wiki?**

Yes

**Is your feature request related to a problem? Please describe.**

Need template/pin layout for Voltson plug

**Describe the solution you'd like**

Simple Template

ascillato, 2019-04-18:
Hi, 

As you already have this device, please help us checking the following thread and please make and submit a template for this device

https://community.smartthings.com/t/etekcity-voltson-esw01-usa/114744/14

ascillato2, 2019-04-18:
If you need further assistant on this, please, do not hesitate on asking here or in the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-18 by ascillato2