arendst/Tasmota #5671: POLL: Disable USE_ADC_VCC for permanent Analog user input

arendst, 2019-04-18:
The ESP8266 provides an Analog input to be used as either Analog input OR as a voltage monitor. **Switching between the two is only possible at compile time**.

Currently the ESP8266 Analog input is enabled only in the sonoff-sensor firmware version. All other versions use it for voltage monitoring.

As recent devices start using the Analog input for either overtemperature detection (Shelly 2.5) or as Button (Supernight smart plug) I think it's a good time to decide to default enable the Analog input for the task it was designed for and make the voltage monitoring optional using the define.

I'm in favor of the switch as I never saw real use for the voltage monitoring. Pls comment if you think otherwise.

ascillato, 2019-04-18:
Agree

meingraham, 2019-04-18:
"Pls comment if you think otherwise." - I do not think otherwise, but thought I'd comment anyway :wink:

Jason2866, 2019-04-18:
Change to analog input. 
The measured voltage is useless. Never helped to find a power issue...

wongnam, 2019-04-18:
Agree, I found that VCC monitoring is unreliable because its value is different between core version or  between NodeMCU / Sonoff devices module.

digiblur, 2019-04-18:
Agree... change to allow the use of A0.  If there's some odd chance where someone needs it the other way it is way too easy for them to compile their own.

blakadder, 2019-04-18:
Voltage monitoring is more of a novelty anyway and definitely not useful to average users. Only fear is that users will be confused if the A0 reading appears on the gui by default

meingraham, 2019-04-18:
Will A0 still only be available if compiled with sensors option (e.g., sonoff-sensors.bin)  or available like any input, even in base version (e.g., sonoff.bin)?

joba-1, 2019-04-18:
my 2c:
* Internal voltage is important mostly for battery operation only. This is not feasible with tasmota (no deep sleep, permanent wifi ->too power hungry): A0 for external sensors makes more sense as default.
* A0 is not used often. Readings should not be displayed per default, only when activated.
* A configurable mapping function like display=a*A0^2+b*A0+c and a configurable unit string would be nice.

arendst, 2019-04-19:
I plan to implement the ADC0 as any other GPIO so in the GUI a dropdown box with possible functionality. Doing it that way will allow for not seeing the Analog input if no functionality is selected.

arendst, 2019-04-19:
In branch **development-adc0** you'll find the implementation of enabling ADC0 for users. It is compatible with the current templates where FLAG = 1 enables the ADC0 as analog input and FLAG = 2 enables the analog input as temperature sensor.

As I have very little coding time the next weeks I released it as another branch to free the current development branch from possible bugs implemented by the default ADC0 option.

Give it a try an I might be able to fix things the next day.

Frogmore42, 2019-04-20:
The new implementation looks much nicer.

arendst, 2019-04-20:
I agree. I'm off for a few weeks giving you time to play with. Once I'm back I will integrate.

arendst, 2019-05-13:
Thx for all your input.

Implementing now.

effelle, 2019-05-13:
@arendst, I've completed my tests right now. 
I think is a good implementation!
What I didn't see, for now, is a calibration tool: depending of the sensor how we can do fine tuning? 
Temperature for example, or even lux (if light analog was meant for that). 
For me should be enabled just for sensors .bin.

arendst, 2019-05-13:
Implemented in dev release 6.5.0.10

closed 2019-05-13 by arendst

ptrooms, 2019-12-03:
Note: this VCC monitor option could BE helpful for battery powered devices, like Tuya.

ascillato, 2019-12-03:
If you need it, you can compile it by yourself (with the steps explained in the wiki). The only required change is in my_user_config.h file to uncomment the line #define USE_VCC_ADC.

As now the ADC is used by several devices like shelly for internal temperature protection, it will remain this option as default. If your device don't use the A0 pin, you can do as explained for using VCC. Take into account that internal VCC was not precise, was core dependant and it is measured internally inside the ESP8266 Chip. So, in the case of battery powered devices, it is better to attach the A0 pin to the battery directly (with the correspondent adaptation electronics) rather than measuring the internal VCC of the ESP chip.

ptrooms, 2019-12-04:
@ascillato  Good suggestion, thanks. Will do/research that on my todo list. The idea behind voltage measuring is to get a message that the battery is draining. 
I Was not aware that the internal voltage measurement was  (that) inaccurate.